renode/renode #1: Improving EFM32 support

basilfx, 2017-11-09:
I'm porting RIOT-OS to the Silicon Labs EFM32 microcontrollers, by [generating the required files](https://github.com/basilfx/EFM2Riot) based on parsing data from their SDK. I've been experimenting with generating Renode platform files, and it turns out that this works great. 

My current setup involves parsing SVD files for the peripheral and augment that with SDK/header file information (flash/sram sizes etc).

At this point, I have +/- 400 CPU files, and I'd like to contribute them. I have a few questions:

* Is there an interest of supporting more EFM32 microcontrollers? I'd like to help!
* If so, what would be the desired structure? Adding 400 CPU's to the platforms/cpu/ folder won't work out I guess. Putting them remotely?

A few more questions about the platform files:
* What is the benefit of using ApplySVD over separate Tag instructions? The latter doesn't require an additional 2+MB file.
* Single-files or inheritance (e.g EFM32GG990F256 inherits from EFM32GG)? So far, only the CPU and NVIC could be inherited, but that's only a few lines per file extra.

Thanks!

basilfx, 2017-11-12:
I currently have some generated files available over [here](https://github.com/basilfx/EFM2RIOT/tree/master/dist-development/contrib). I am able to run the default example of RIOT-OS on all but two boards (SLSTK3401a and SLTB001a lack SVD files for generation).

Note that `EFM32DeviceInformation.cs` cannot be generalized for all the different families of the EFM32 MCUs. I therefore generate one per family.

mgielda, 2017-11-24:
Hi Bas! Thank you for contributing. We have some ideas how to do this even better with Renode 1.2 based on your contribution, bear with us a little and we'll propose how we see this.

basilfx, 2017-11-26:
I will. Please let me know if I can be of any help :-)

PiotrZierhoffer, 2018-04-12:
With the new set of SiLabs platforms merged, I think we can close this issue.

closed 2018-04-12 by PiotrZierhoffer