renode/renode #5: Bluetooth Low Energy

AhmedHussein534, 2018-02-26:
does renode supports BLE projects under zephyr?

mgielda, 2018-04-11:
Hi Ahmed, this is a matter of supporting a specific radio model for a given Bluetooth-enabled platform. Which Bluetooth platform (board/SoC) were you interested in specifically, and what sample would you like to run - could you describe the use case in more detail, this would help us with setting our priorities.

mgielda, 2018-05-09:
Hi Ahmed, we do not at this point support any Bluetooth peripherals, but it's definitely something we'd like to do if there is enough need for it. I'll mark this as a feature request.

mchodzikiewicz, 2019-09-23:
Hi, I am interested in testing nrf52 as well. Since the bluetooth stack is a binary blob, perhaps a way of work is to mock supervisor calls instead of actual peripheral beneath it? http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dai0179b/ar01s02s07.html

Citrullin, 2020-11-12:
I am interested in this as well. Especially with the NimBLE stack.

jain0nikhil, 2021-02-17:
Is Bluetooth peripheral support enable for nRF52840DK if yes, can you share some sample code?

PiotrZierhoffer, 2021-02-17:
@jain0nikhil while we do have the nRF52840 chip supported, it does not yet have radio connectivity.

jain0nikhil, 2021-02-17:
@PiotrZierhoffer is there any plan for supporting Bluetooth Peripheral. 
One more question: I am trying to run one custom FW on emulator in which FLASH changed to 0x27000 from 0x00 but nothing happen in emulator after loading binary. Do I need to specify some extra value for that ?

jain0nikhil, 2021-02-19:
is there any board which support radio connectivity in renode which I can refer for adding support for nRF?

PiotrZierhoffer, 2021-02-19:
We have some 802.15.4 radios you cal look at, for example CC2538 or CC2520. The communication layer will be similar, but we have no bt radios at the moment.

nRF52840 BT is not in the nearest scope, so I can't give you any specifics.

Regarding your other question: you need to adjust the flash location in the repl file and in your linker scripts. That's not enough information to say anything specific though.

grifcj, 2021-11-05:
Was able to get a softdevice-based peripheral advertising. Used the Nordic SDK 17.1.0 and SD 7.2.0. Based on the ble_app_hrs demo with some modifications to disable the security related features as peripheral models don't currently support. Quickly tried connection with the central from same example set, but that unfortunately didn't work.

Requires a few simple fixes to peripheral models, which I did on top of existing bluetooth PR here in branch [bluetooth-softdevice](https://github.com/grifcj/renode-infrastructure/tree/bluetooth-softdevice). Wasn't sure if changes should stack on top of existing PR, as it might redundantly contain the existing PR's commits.

I attach the modified ble_app_hrs main.c and *.resc for reference. Can go in a bit more detail on building with that SDK or just supply the binaries if desired.
[example.zip](https://github.com/renode/renode/files/7481631/example.zip)

One other important caveat is that simulation seems to need to run at 1 us quantum. Anything more, and the peripheral will advertise once, but then effectively halts with WFE as scheduling with timer0 gets messed up. Compare values to drive timer events seem to be somewhat aware of time passing.

Overall, trying to workout issues with no source code (i.e. just the soft device binary) is unsurprisingly a bit tricky. Given Nordic seems to be moving on to open source Zephyr based solutions for the future anyways, probably turn my attention to enabling more features (e.g. security) with that SDK. 

[image]

Citrullin, 2021-11-07:
> Overall, trying to workout issues with no source code (i.e. just the soft device binary) is unsurprisingly a bit tricky. Given Nordic seems to be moving on to open source Zephyr based solutions for [...]

If you have time you can give NimBLE a try. It is used as the BLE stack in RIOT and MyNewt. Espressif also uses a fork for their ESP32 with FreeRTOS. So, I guess you can also use it in FreeRTOS.

grifcj, 2021-11-07:
Would like to cross-check models against another stack. So far, basics seem to work with zephyr and at least advertising with softdevice. If NimBLE hardware implementation is similar enough, it might already function as well.

Spent an hour or so with MyNewt, unfortunately wasn't able to get it to build an acceptable elf of the blehr sample in nimble repo targeting the nrf52840. Have any direction there? Didn't try RIOT. Perhaps have better luck with their getting started docs. Doesn't matter if RIOT or MyNewt, whichever one is easiest, but need some sample apps, preferably a central and peripheral that implements hr, like the [sample apps for Zephyr](https://docs.zephyrproject.org/latest/samples/index.html).

With that, it could be assessed whether or not NimBLE functions on renode with the [PR branch](https://github.com/grifcj/renode-infrastructure/tree/bluetooth-work)

Unfortunately, my time is limited for platform ramp up / education in near term. Would probably prefer spending effort in near term towards expanding out zephyr-based solutions since I already have a foothold there.

Citrullin, 2021-11-08:
> Spent an hour or so with MyNewt, unfortunately wasn't able to get it to build an acceptable elf of the blehr sample in nimble repo targeting the nrf52840. Have any direction there? Didn't try RIOT. [...]

I also don't have any experience with MyNewt. I can help you with RIOT though, if you want to. If you give me your contact, I can probably make some time in order to help you with this. 
I only used the IPv6 over BLE stack though. But I guess if that works fine, the rest shouldn't be an issue. 
I would recommend a Linux though. I tried to make it work on Windows, but it makes it very complicated to get it running due to the not proper USB support in WSL2. Mac also makes some issues in my experience. Linux (at least Debian based) is very straight forward, since the build system of RIOT is just a couple of Makefiles.
The RIOT documentation though... Let's say developer don't really like to write a lot of documentation, I guess :D

grifcj, 2021-11-09:
Attempts with RIOT went a little better. Setting up was just cloning the repo, putting arm toolchain on path and calling `make BOARD=nrf52840dk -C examples/nimble_heart_rate_sensor`.

[image]

There was at least one issue in the RTC model. IRQ was being gated on event enable, and that's not how the hardware works as can be seen in the nrf52840 product spec, so that needs to be fixed and pushed to PR.

[image]

So advertising seems to work, but connection with central not quite working yet. Didn't see a central example in riot repo, so was just using the zephyr central to try to connect. Central sees advertising from riot peripheral, and sends a connect request, but things don't get established. Will have to investigate more.

If you'd like to try hacking some yourself, the build docs are [here](https://renode.readthedocs.io/en/latest/advanced/building_from_sources.html#building-from-source). Which is basically just install pre-reqs and run build.sh from cloned renode repo. The current bluetooth capabilities require build from source, since it's ongoing PR. You'll need the bluetooth-work branch from my forks of renode and renode-infrastructure. 

Attaching script file I used. You would need to update paths to bins to where they are on your system.
[bluetooth.zip](https://github.com/renode/renode/files/7503461/bluetooth.zip)

If that's way too brief to be of much use, you can try dropping me a line at grifcj@gmail.com.

Citrullin, 2021-11-09:
> If you'd like to try hacking some yourself, the build docs are [here](https://renode.readthedocs.io/en/latest/advanced/building_from_sources.html#building-from-source). Which is basically just [...]

Great that the build process works. I am not so much into the low level Bluetooth stack. I will give it eventually a try. Thanks for the pointer to the resources.

PiotrZierhoffer, 2022-05-17:
With the BLE support for nRF52840 we released (with enormous help from @grifcj) I believe I can close this issue. If any other issues related to BLE are encountered, let's track them separately

https://antmicro.com/blog/2022/04/developing-and-testing-ble-on-nrf52840-with-renode-and-zephyr/

closed 2022-05-17 by PiotrZierhoffer