renode/renode #6: UART interrupt handling

sandeep2208, 2018-03-29:
I am currently using the renode framework for some of my unit testing and i had a question about the UART Interrupt handling. I am using a STM32f4 series controller and from my testing i saw that the UART can only be configured to POLL for data rather than interrupt based. I see that there is a  GPIOInterruptManager function which is used in the MIV_CoreGPIO.cs file. I believe this is used by the UART in the example project that is defined in your user guide. Can this same function be used for the STM32 project as well for interrupt handling or are you guys adding any new code to support interrupt handling for STM32 based controller??

PiotrZierhoffer, 2018-04-13:
STM32F4 platforms use `STM32_UART.cs` model, which does have basic interrupt handling implemented, with the `IRQ` property.

I can see that not all interrupts are connected to the controller. Please consult `platforms/cpus/stm32f4.repl` - you will see that we have interrupts connected for `uart1`, `uart4` and `uart5`. Which UART do you want to use?

I will try to add the connections for other UARTs as well.

MiV_CoreGPIO is a GPIO controller to which we connect things like LEDs, buttons etc. Actually, in the real MiV platform, UART does not have any interrupts routed to the interrupt controller.

PiotrZierhoffer, 2018-05-25:
This is fixed by f26f533

closed 2018-05-25 by PiotrZierhoffer