renode/renode #7: Gtk# is required to run the application in a windows environment

Apacz87, 2018-05-04:
Running application on Windows 10 system, using version 1.3 from zip file, results in program crash if Gtk# is not instaled. In the documentation from which I read, I did not find any information about this requirement. Gtk# it is only listed as required when compiling.

PiotrZierhoffer, 2018-05-24:
Thank you for the report. We are most likely missing a dll, will investigate it soon.

PiotrZierhoffer, 2018-06-14:
We are migrating from GTK# to WPF on Windows - so while this report is valid, we will be fixing in another way, completely removing the dependency.

When it's done, I'll close the issue.

PiotrZierhoffer, 2018-06-23:
With the tag v1.4 this is no longer relevant, so I close the issue.

closed 2018-06-23 by PiotrZierhoffer