renode/renode #12: Crash when emulating riscv floating point example

AntonKrug, 2018-05-25:
When the first floating point arithemtical C code is going to execute, it will cause the emulation to crash, the line is:

`for (float percentage = 0.0f; percentage <= 1.3f; percentage += 0.04f) {`

And resulting crash:

```
Unhandled Exception: System.AccessViolationException: Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
   at Antmicro.Renode.Peripherals.CPU.TranslationCPU.CpuLoop()
   at System.Threading.ThreadHelper.ThreadStart_Context(Object state)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart()
```

The code works fine in riscv hardware.

When single stepping the code the crash occurs at the 800018b4 instruction:

```
147           for (float percentage = 0.0f; percentage <= 1.3f; percentage += 0.04f) {
800018a0:   flw     fa4,-24(s0)
800018a4:   lui     a5,0x8000b
800018a8:   flw     fa5,-1980(a5) # 0x8000a844
800018ac:   fadd.s  fa5,fa4,fa5
800018b0:   fsw     fa5,-24(s0)
800018b4:   flw     fa4,-24(s0)
800018b8:   lui     a5,0x8000b
800018bc:   flw     fa5,-1976(a5) # 0x8000a848
800018c0:   fle.s   a5,fa4,fa5
800018c4:   bnez    a5,0x800016a0 <main+64>
146         for (int i = 0 ; i < (NELEMS(sets) - 1); i++) {
800018c8:   lw      a5,-20(s0)
800018cc:   addi    a5,a5,1
800018d0:   sw      a5,-20(s0)
800018d4:   lw      a4,-20(s0)
800018d8:   li      a5,3
800018dc:   bleu    a4,a5,0x80001698 <main+56>
162         screenClear();
800018e0:   jal     ra,0x80001600 <screenClear>
```

Content of the registers just before the crash is:
```
zero	0x0 (Hex)	
ra	0x80001690 (Hex)	
sp	0x8000d030 (Hex)	
gp	0x8000b500 (Hex)	
tp	0x0 (Hex)	
t0	0x80004548 (Hex)	
t1	0xf (Hex)	
t2	0x0 (Hex)	
s0	0x8000d070 (Hex)	
s1	0x0 (Hex)	
a0	0x4 (Hex)	
a1	0x4a (Hex)	
a2	0x8000b10c (Hex)	
a3	0x4 (Hex)	
a4	0x0 (Hex)	
a5	0x3 (Hex)	
a6	0x0 (Hex)	
a7	0x0 (Hex)	
s2	0x0 (Hex)	
s3	0x0 (Hex)	
s4	0x0 (Hex)	
s5	0x0 (Hex)	
s6	0x0 (Hex)	
s7	0x0 (Hex)	
s8	0x0 (Hex)	
s9	0x0 (Hex)	
s10	0x0 (Hex)	
s11	0x0 (Hex)	
t3	0x0 (Hex)	
t4	0x0 (Hex)	
t5	0x0 (Hex)	
t6	0x0 (Hex)	
pc	0x800018b4 (Hex)	
f0	0x0 (Hex)	
f1	0x0 (Hex)	
f2	0x0 (Hex)	
f3	0x0 (Hex)	
f4	0x0 (Hex)	
f5	0x0 (Hex)	
f6	0x0 (Hex)	
f7	0x0 (Hex)	
f8	0x0 (Hex)	
f9	0x0 (Hex)	
f10	0x0 (Hex)	
f11	0x0 (Hex)	
f12	0x0 (Hex)	
f13	0x0 (Hex)	
f14	0x0 (Hex)	
f15	0x0 (Hex)	
f16	0x0 (Hex)	
f17	0x0 (Hex)	
f18	0x0 (Hex)	
f19	0x0 (Hex)	
f20	0x0 (Hex)	
f21	0x0 (Hex)	
f22	0x0 (Hex)	
f23	0x0 (Hex)	
f24	0x0 (Hex)	
f25	0x0 (Hex)	
f26	0x0 (Hex)	
f27	0x0 (Hex)	
f28	0x0 (Hex)	
f29	0x0 (Hex)	
f30	0x0 (Hex)	
f31	0x0 (Hex)	
```

PiotrZierhoffer, 2018-06-14:
I verified it to work on master. The issue was there in the previous release.

AntonKrug, 2018-06-25:
I'm testing the Renode 1.4 with a raytracer example and encountered some more errors:

```
12:20:28.1758 [WARNING] cpu: Unsupported GDB command: qL1200000000000000000
12:22:39.6777 [WARNING] cpu: Unsupported GDB command: qL1200000000000000000
12:22:39.6917 [WARNING] cpu: Unsupported GDB command: qL1200000000000000000
12:22:39.7861 [WARNING] cpu: Unsupported GDB command: qL1200000000000000000
12:26:19.6336 [WARNING] cpu: Unsupported GDB command: vKill;a410
12:26:19.6727 [ERROR] cpu: CPU abort [PC=0x80002DA4]: Illegal instruction exception!.
12:26:19.7138 [WARNING] uart: Unhandled write to offset 0xC. Unhandled bits: [0] when writing value 0x1. Tags: BIT8 (0x1).
12:26:19.7514 [ERROR] cpu: CPU abort [PC=0x80003224]: Illegal instruction exception!.
12:26:22.2069 [ERROR] cpu: CPU abort [PC=0x80003068]: Illegal instruction exception!.
12:26:23.2206 [WARNING] sysbus: [CPU0: 0x0] ReadByte from non existing peripheral at 0x0.
12:26:23.2206 [ERROR] cpu: CPU abort [PC=0x0]: Trying to execute code outside RAM or ROM at 0x00000000.
12:26:23.2211 [WARNING] sysbus: [CPU0: 0x0] ReadByte from non existing peripheral at 0x0.
12:26:23.2211 [ERROR] cpu: CPU abort [PC=0x0]: Trying to execute code outside RAM or ROM at 0x00000000.
12:26:23.2306 [WARNING] sysbus: [CPU0: 0x0] ReadByte from non existing peripheral at 0x0.
```

For example this is the location around 80002da4

```
80002d9c auipc a5,0x6
80002da0 flw ft2,1332(a5)
80002da4 lui a3, 0xffcf6
```

And here are the build binaries:

[Debug.zip](https://github.com/renode/renode/files/2132902/Debug.zip)

And sources are here:

https://github.com/AntonKrug/miv-rv32imaf-raytracer-uart-cpp

The example continues to run and on a surface it looks like correctly, but when it's done it's loops the errors caused by the PC being at 0x0. I will have to double check if that was caused by our hal (not catching the return correctly and not using the _fini loop correctly there), or it might be related to the previous errors.

AntonKrug, 2018-06-25:
The PC being at 0x0 is problem of our HAL, for some reason the _fini doesn't do infinite loop anymore.

But the problematic instruction is I think generated by the __ieee754_logf() from newlib.
As the logf is used only for esthetics in that example (it's misbehaviour will not affect rendering).

AntonKrug, 2018-06-25:
Another example has a problem as well. Sources are here:
https://github.com/truhlikfredy/miv-rv32imaf-dct

Build project is here:
[Debug.zip](https://github.com/renode/renode/files/2133391/Debug.zip)

Here the implementation of cosine function __kernel_cosf() in the newlib causing these:

```
14:52:48.3673 [ERROR] cpu: CPU abort [PC=0x80001CD0]: Illegal instruction exception!.
```

And it's after the flw instruction again:

```
80001cc4:   flw     ft0,288(a3) # 0x8000ade0
80001cc8:   auipc   a3,0x9
80001ccc:   flw     fa0,284(a3) # 0x8000ade4
80001cd0:   fmadd.s fa5,fa4,fa5,fa3
80001cd4:   auipc   a3,0x9
80001cd8:   flw     fa2,276(a3) # 0x8000ade8
```

PiotrZierhoffer, 2018-06-25:
I reopen this then

AntonKrug, 2018-06-25:
But now it doesn't crash the emulation, and even the code continues.

PiotrZierhoffer, 2018-06-26:
We think this is actually not related to floating point operations. There is some other corruption, we are able to cause these fails on different instructions, but only on Windows.

When we are able to describe the issue with more details, I'll close this one, but for now let's leave it open.

AntonKrug, 2018-06-26:
Yes you are right, it might be something else and yes I can't see it on Linux.

PiotrZierhoffer, 2018-06-27:
It is confirmed that is was related to handling of the generated code - and floating point operations were tricky enough to cause the bug. We think we have the fix.

PiotrZierhoffer, 2018-06-28:
Fixed with 8dce5268d615c9a697f055ed42e58fb60c2b1abd

closed 2018-06-28 by PiotrZierhoffer