renode/renode #13: Cosmetic error when copy is used

AntonKrug, 2018-05-25:
Cosmetic problem cause when COPY/PASTE feature is used anywhere while Renode is running in the background, it will cause the following error to be displayed:

```
(Renode:23972): Gdk-CRITICAL **: inner_clipboard_window_procedure: assertion 'success' failed
```

It could be ignored and that would be fine, but the problem is that when Renode is used from within other UI, such as Eclipse, then it will cause that given tab to catch attention/focus. When there are multiple tools switching between one console tab it will forcefully switch to the Renode's console on any of the Copy events.

PiotrZierhoffer, 2018-06-23:
This is no longer visible on Windows, as we migrated to WPF. On other OSes it depends on the GTK version. On newer installations we don't observe any GTK warnings anymore. Unfortunately, we cannot control these logs. Redirecting stderr to /dev/null helps, but can also hide stack traces in case of errors.

@AntonKrug does this solve the issue for you?

AntonKrug, 2018-06-25:
Yes, I can't replicate these warnings anymore. Great work.

mgielda, 2018-06-25:
I assume we should close it then. Will only apply to people with old GTK on Linux/MacOS, and even if so, it's a common problem, nothing easily fixable.

closed 2018-06-25 by mgielda

AntonKrug, 2018-06-25:
Yes, if I will experience it again, I will update it here. And I will retest it with few distributions on Linux.