renode/renode #8: Windows - Renode crashes when closed in certain way.

AntonKrug, 2018-05-25:
When renode is closed with the "x" button it creates crash popup window.

[image]

This behaviour is not noticed when it's killed by some other mechanism, for example if the renode is launched with eclipse then if eclipse is killing it, it willl shutdown without any popup.

AntonKrug, 2018-06-25:
Great job with the 1.4 release.
Quickly tested with 64bit Windows 10 and it the crash seems to be resolved. Probably the issue can be closed and if it will resurface and proper testing I will update the issue.

PiotrZierhoffer, 2018-06-25:
Yes, it was related to GTK releasing its resources. Now, with no GTK on Windows, it is resolved.

closed 2018-06-25 by PiotrZierhoffer