renode/renode #9: Windows - Missing font message

AntonKrug, 2018-05-25:
```
Font 'Roboto Mono' not available in the system. Using 'Sans' instead
12:39:59.1273 [WARNING] The font 'Roboto Mono' defined in the config file cannot be loaded.
```

This is a cosmetic issue, but probably should be addressed, I got few ideas how to address these:
- bundling the required font with the renode (if it's royalty free and renode depends on it)
- make the font failure and switching fallback lower log level (I don't like this idea)
- chose a different mono font which will be universally present on all platform (if there exists any)
- have predefined default fonts for each platform and only complain if these will not be found (sounds to me as the most reasonable choice)

I have mixed feelings about this issue, because it doesn't really have a proper solution and leaving it as is will probably cause some user to be confusion at some point. And creating log of verbose output can cause catching attention and devaluating the more important warning messages. Lovering the log levels is not solution as well. So i'm not sure how this could be addressed properly.

PiotrZierhoffer, 2018-06-23:
At the moment we provide the font. Unfortunately, loading it for the XWT framework (the GUI library we use) is not trivial and OS-dependent. It was impossible on GTK on Windows, but it seems to be possible with WPF. So this is the approach we'll take.

AntonKrug, 2018-06-25:
On Debian I could quiet it down by installing the fonts (go figure :-) )
So now I wonder if SoftConsole installer should probably bundle a font (or at least have it listed as requirements/dependencies) to make Renode quieter. But still a cosmetic issue.

PiotrZierhoffer, 2018-06-28:
Fixed with 64c3bf3d166b2b3c81388f7aece93427d602f1cf

@AntonKrug are you sure you had to install the font package on Debian? We never observe these warnings on Linux systems, the font loading should work just fine

AntonKrug, 2018-06-28:
I was testing it absolutely bare minimum Debian, probably installing a browser or any other app could have the fonts as dependency and any normal system very likely has it installed already. If you want I could re-run the tests again from scratch and let you know what versions of Debian do need it and what doesn't. It's in the "contrib" and not the main "chanel" so I think Debian is trying to be very conservative and install only very vanilla things in their base install (the older version didn't even had support to play mp3 because they were concerned about it's license)

PiotrZierhoffer, 2018-06-28:
Anyway, we should not rely on the presence of the font in the system. Nevertheless, on Windows we load it automatically with no additional action from the user or installer required, so I'm closing this issue.

We will verify the plain Debian separately.

closed 2018-06-28 by PiotrZierhoffer