renode/renode #15: Feature request: sifive_e300 board

arunthomas, 2018-06-13:
It would be great if Renode provided emulation support for the sifive_e300 platform. QEMU has good support for the e300, so I expect it wouldn't be too much effort to add support to Renode.

mgielda, 2018-06-13:
Hi Arun, we do have FE310 support (based on the E300 platform) - there's even an example: https://github.com/renode/renode/blob/master/scripts/single-node/sifive_fe310.resc

As far as I understand, it's only FE310 that's available on a 'board' (the HiFive1 or our RISC-V FE310 SoM), E300 is the platform name but not a chip/board.

Note that since recently we also support HiFive Unleashed.

mgielda, 2018-06-14:
No worries. Glad to be of help.