renode/renode #20: Value was either too large or too small for an Int32

zzzh, 2018-07-19:
it happened when I try the command "include @scripts/single-node/miv.repl" following https://renode.readthedocs.io/en/latest/tutorials/miv-example.html

any ideas?

zzzh, 2018-07-19:
BTW: I was running renode on 64bit ubuntu 16.04

PiotrZierhoffer, 2018-07-19:
I will reopen this, as it sounds to be something we need to fix.

Do I get it right, that you did the following actions?
1. Run Renode from the console
2. pin it the the side bar
3. start Renode from the sidebad
4. `include @scritps/...`
5. Crash.

Is that correct? Can you please paste the full stack trace of the error that pops up?

AntonKrug, 2018-07-19:
Did you mean "include @scripts/single-node/miv.repl" or "include @scripts/single-node/miv.resc". I tried to install Renode from dep package just right now on 64bit Ubuntu 16.04 and can't replicate the issue as I don't even know what steps I should do exactly.

One thing which crosses my mind is that you could have multiple mono installed and when you run from the sidebar it has different environment variables than from command line/bash/shell.

The /usr/bin/renode invokes mono wherever it will be found this might be different places depending between running shortcut and using it from the console. From the console probably the bashrc is run as first which will setup custom paths and then mono executed from there will be different.

If you make script like this:
`mono --version && read`
And then you make a shortcut for it, will display different mono version than when you are running the script from command line?

zzzh, 2018-07-19:
Hi, PiotrZierhoffer,

Yeah, it's happened exactly like what you described.

Here's the detail:
Value was either too large or too small for an Int32.
  at System.Convert.ToInt32 (System.Double value) [0x00084] in <71d8ad678db34313b7f718a414dfcb25>:0 
  at Antmicro.Renode.Logging.ConsoleBackend.GetMessageLinesCountForRow (System.String row) [0x0001d] in <73acd1120ee04249a578de8dc4a3fbc4>:0 
  at Antmicro.Renode.Logging.ConsoleBackend+<>c.<GetMessageLinesCount>b__25_0 (System.String x) [0x00000] in <73acd1120ee04249a578de8dc4a3fbc4>:0 
  at System.Linq.Enumerable.Sum[TSource] (System.Collections.Generic.IEnumerable`1[T] source, System.Func`2[T,TResult] selector) [0x0002e] in <c0502b9f220448589f6c0bfcd98a6c9b>:0 
  at Antmicro.Renode.Logging.ConsoleBackend.GetMessageLinesCount (System.String message) [0x00015] in <73acd1120ee04249a578de8dc4a3fbc4>:0 
  at Antmicro.Renode.Logging.ConsoleBackend.WriteNewLine (System.String line) [0x00028] in <73acd1120ee04249a578de8dc4a3fbc4>:0 
  at Antmicro.Renode.Logging.ConsoleBackend.Log (Antmicro.Renode.Logging.LogEntry entry) [0x001f5] in <73acd1120ee04249a578de8dc4a3fbc4>:0 
  at Antmicro.Renode.Logging.Logger+ActualLogger.WriteLogEntryToBackends (Antmicro.Renode.Logging.LogEntry entry) [0x0000f] in <73acd1120ee04249a578de8dc4a3fbc4>:0 
  at Antmicro.Renode.Logging.Logger+ActualLogger.LoggingThreadBody () [0x00033] in <73acd1120ee04249a578de8dc4a3fbc4>:0 
  at System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) [0x00014] in <71d8ad678db34313b7f718a414dfcb25>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <71d8ad678db34313b7f718a414dfcb25>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <71d8ad678db34313b7f718a414dfcb25>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) [0x0002b] in <71d8ad678db34313b7f718a414dfcb25>:0 
  at System.Threading.ThreadHelper.ThreadStart () [0x00008] in <71d8ad678db34313b7f718a414dfcb25>:0 

Here's a picture:
[image]

zzzh, 2018-07-19:
Hi, AntonKrug,

I mean "include @scripts/single-node/miv.resc" because repl would not work.

I think there's only 1 mono installed, below is what I got for "whereis mono":
mono: /usr/bin/mono /usr/lib/mono /etc/mono /usr/share/mono /usr/share/man/man1/mono.1.gz

BTW: I did make a script as you said, but I don't know how to run it's shortcut by a click, I can't put it in sidebar and there's no terminal or bash option in "open with" list.

AntonKrug, 2018-07-19:
Oh I see, just noticed Ubuntu UI tries to make unpleasant and doesn't recognize sh script as executable. Let's just check if what version it is, run the `mono --version` from a terminal.

Started clean VM, went to mono-project to add the mono repository, installed mono-complete instead of recommended mono-develop. Then I installed by hand screen,gksu, gtksharp2 with apt-get. Then downloaded DEB file of Renode 1.4.1 from the github and installed it with dpkg -i ...

Just wild guess, could it this issues linked with the file-failing-to-download issue? I attached you the elf file in the previous thread to see if that will have an effect. I cleared my renode cache/index and disabled my networking but can't trigger the issue from sidebar nor from the terminal.

Once I encountered an issue when I installed the packages in slightly odd order. So I'm still curious about your installed packages, not sure, but maybe the `cat /var/log/apt/history.log` could tell something?

Sorry for not being very helpful :(

zzzh, 2018-07-20:
Hi, AntonKrug,

you are very helpful :), thank you.

here's what you want:
mono --version
Mono JIT compiler version 5.12.0.226 (tarball Thu May  3 09:48:32 UTC 2018)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
	TLS:           __thread
	SIGSEGV:       altstack
	Notifications: epoll
	Architecture:  amd64
	Disabled:      none
	Misc:          softdebug 
	Interpreter:   yes
	LLVM:          supported, not enabled.

[apt_install.log](https://github.com/renode/renode/files/2212143/apt_install.log)

	GC:            sgen (concurrent by default)

PiotrZierhoffer, 2018-07-20:
@zzzh is this error repeatable for you? Or was it a one time event?

The error was most likely caused by `Console.WindowWitdh` reported as 0. Why did that happen? I don't really know, as it is a standard .NET thing. But if you can repeat it a deterministic way, then maybe we'd be able to debug it.

PiotrZierhoffer, 2018-07-20:
Ok, we are able to repeat this error. I don't really know how to fix it, though.

We add a Renode.desktop entry to `/usr/share/applications`, which gives you a Renode shortcut in the Unity Dash. When you start it, everything works fine.

But when you create the shortcut manually, it does not open the terminal window at all - that's why we crash. We will try to prevent the crash, but one of the ways to do that is to prevent Renode from starting without the terminal - we will think about the correct approach internally.

What you can do to alleviate your problem is to run `grep -ri renode ~/.local/`, open the `*.desktop` file it finds and add `Terminal=true` to the file. It will cause the shortcut to open the window required for our logger.

zzzh, 2018-07-20:
@PiotrZierhoffer , Great, good to know you had found the cause of this issue.

I'll try your way to alleviate it later, since it's running in vmware, because of a samba issue, I'd return it to a snapshot before install renode.

PiotrZierhoffer, 2020-09-01:
We actually had a hard time getting a reproducible situation where this happened, but we stumbled upon a similar issue in a different situation. This is now resolved on master

closed 2020-09-01 by PiotrZierhoffer