renode/renode #21: can't download the elf file

zzzh, 2018-07-19:
[ERROR] Failed to download from http://antmicro.com/projects/renode/riscv-liteos-port.elf-s_689820-e68d3bcf0a12c25daa66fc51e474281bcbed2fc7, reason: Timeout.

it happened when I try the command "include @scripts/single-node/miv.repl" following https://renode.readthedocs.io/en/latest/tutorials/miv-example.html

any ideas?

zzzh, 2018-07-19:
Actually, it's not a renode issue, but seems the site holding the elf files can't be reached.
Please provide a way to download the elf files so that people who are interested  can see how renode works.

PiotrZierhoffer, 2018-07-19:
It could have been a temporary issue. Could you please verify once again?

Once the binary is downloaded, it is cached - so you have to download them only once on each machine.

If you have your own software for this platform, you can also use it by setting the $bin variable (`$bin=@path/to/file.elf`).

AntonKrug, 2018-07-19:
Probably it was temporary as I can download it without any issue.

zzzh, 2018-07-19:
Hi, PiotrZierhoffer and AntonKrug, 

I think it's because some internet issues that make it impossible to access that site from where I am, and it may also happen to others.

So, would you please to put the elf files at github too?

AntonKrug, 2018-07-19:
Would this help?

[riscv-liteos-port.zip](https://github.com/renode/renode/files/2210573/riscv-liteos-port.zip)

(unzip it after download)

AntonKrug, 2018-07-19:
That's up to the Antmicro guys how they would prefer to structure it. I just wanted to give you something to get you going in the meantime.

AmeyaVS, 2018-07-25:
Hello,

I have similar issue while running the `test.sh` to run the sanity tests.

Here is a short list:
- Should Generate Interrupts On Gpio High Level:  
    ```text
    TargetInvocationException: Exception has been thrown by the target of an invocation. KeywordException: Could not execute command 'sysbus LoadELF @http://antmicro.com/projects/renode/riscv-interrupt-blinky-gpio_interrupts-level_high.elf-s_134928-e5184e20f1458b9784c10c48ad8fde0a738cfb65': Failed to download http://antmicro.com/projects/renode/riscv-interrupt-blinky-gpio_interrupts-level_high.elf-s_134928-e5184e20f1458b9784c10c48ad8fde0a738cfb65, see log for details. 
    ```

- Should Generate Interrupts On Gpio Low Level:  
    ```text
    TargetInvocationException: Exception has been thrown by the target of an invocation. KeywordException: Could not execute command 'sysbus LoadELF @http://antmicro.com/projects/renode/riscv-interrupt-blinky-gpio_interrupts-level_low.elf-s_134928-3d7a09bb3cf434fde9bf9d2c29888ffb11861c0b': Failed to download http://antmicro.com/projects/renode/riscv-interrupt-blinky-gpio_interrupts-level_low.elf-s_134928-3d7a09bb3cf434fde9bf9d2c29888ffb11861c0b, see log for details. 
    ```

I have setup the same development environment on two different systems(Ubuntu 16.04 LTS).
And in only one of the system I am observing the same failure repeatedly, also I am able to download the binaries on the system browser and command-line.

Maybe the issue is related.

-Ameya

zzzh, 2018-07-26:
@AmeyaVS    I would say the system you observed same failure may have internet issue, just like what happened on my system.

AmeyaVS, 2018-07-26:
@zzzh I tried today, somehow it is now working.
The thing is I was able to download the files using the browser or command-line yesterday, when I faced this issue.

zzzh, 2018-07-27:
@AmeyaVS good to know it's working now, the reason of your issue should be in the log.

closed 2018-07-27 by zzzh