renode/renode #22: libmonoboehm-2.0-1 install issue

zzzh, 2018-07-20:
I noticed that there's a "-p  build binary packages" option in building renode from source, that might be what I need.

So I did it following this page : http://renode.readthedocs.io/en/latest/advanced/building_from_sources.html#building-from-source

When I run this command: 
"sudo apt-get install git mono-complete automake autoconf libtool g++ realpath gksu libgtk2.0-dev screen uml-utilities gtk-sharp2 python2.7"  

This issue appeared, it says :
"The following packages have unmet dependencies:
 mono-utils : Depends: libmonoboehm-2.0-1 (= 5.12.0.226-0xamarin3+ubuntu1604b1) but it is not going to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution)."

I tried 'apt-get -f install' and it looks like works, but have no effect to solve the issue.

Any ideas?

PiotrZierhoffer, 2018-07-20:
What distribution did you run it on? Was it also Ubuntu 16.04?

PiotrZierhoffer, 2018-07-20:
Also, packaging requires some more dependencies: `rpm`, `bsdtar`, `gem` and `fpm` gem package. This part is not yet documented.

zzzh, 2018-07-20:
Yeah, it's still on 64bit ubuntu 16.04.

So, that page is not ready to be a guide for installing renode by source? 
I appreciate renode work but it would be better if there's some words in that page like "not ready yet"

@PiotrZierhoffer

PiotrZierhoffer, 2018-07-20:
It's just -p option is not usually required for users - and the script tells you step by step which dependencies you lack in the process.

To build from source it's enough to follow the docs.

I am surprised by the libmonoboehm error though, and we will investigate it.

zzzh, 2018-07-20:
Got it, thank you! @PiotrZierhoffer

AntonKrug, 2018-07-20:
When I removed the renode temporary with `atp-get remove renode` then it was easier for me to install any other dependencies and resolved similar issues as you are describing.  And then when I was done I installed it back with `dpkg -i renode.deb`

mgielda, 2018-07-20:
Deleted duplicate comment.

Dear @zzzh, thanks for the feedback. Note that as mentioned by @PiotrZierhoffer, the source installation is documented (of course, there's always some issues we can't predict that people stumble upon, the best thing to do is to let us know). It's only the packaging - which I assume is not a feature 95% of users need to use - that isn't documented yet, but we never claim anywhere that it is. Obviously we'll want to document that too, just haven't gotten to that yet.

PiotrZierhoffer, 2018-07-20:
We just created a clean Ubuntu 16.04 and executed our instructions step by step. Everything worked fine.

I don't know where to point you exactly. Maybe you have some nonstandard package sources for apt? Or have a package installed, that is a blocker for the Boehm package, but that is not very probable.

AntonKrug, 2018-07-20:
For the Antmicro guys or for anybody else, it's hard to cover 100.0% cases as there is a lot of distributions and everybody could have them set up in different ways (using the same ubuntu without unity but with xfce manager will cause to have very different preinstalled packages etc...), sometimes the installations even might be bit broken (using backport repositories a lot, forcing a specific version of a package by accident, installing some binaries behind the packager).  Bigger companies like Google on purpose target below 100.0% as it's just too resource demanding to cover all the last few edge cases.

I tried to install git mono-complete automake autoconf libtool g++ realpath gksu libgtk2.0-dev screen uml-utilities gtk-sharp2 python2.7 on my VM and it worked fine.

Still for you I would recommend to uninstall the renode for the moment and then try installing all the dependencies and then reinstall renode. If that will not work then maybe you have multiple different sources which are conflicting with each other. I noticed in my setup that I already had the libmonoboehm probably as I was installing the mono the first time. So do you have the same apt sources as us? When do you run `apt-get update` what sources are listed?

[image]

From the file you attached on the other issue it states that it was trying to install the libmonoboehm-2.0-1 already :/ 

Yet another idea if the uninstall of renode will not work is to do update and upgrade of all the packages, sometimes when there are newer packages released and part of your system depends on older part on newer then the dependancy resolved can get confused in the way you described it above, and do distr-upgrade. But be careful as something else might break, if your source repository point to wrong repository or if they are setup awkwardly this might cause you even more problems.

https://askubuntu.com/questions/215267/will-apt-get-dist-upgrade-upgrade-my-system-to-newer-version

zzzh, 2018-07-20:
@mgielda , sure,  there's always some issues no one can't predict, I understand that.
I appreciate renode, it's great work, thanks for all of you guys' effort to make it happen.

AntonKrug, 2018-07-20:
You mentioned it's a VM, if you don't have a sensitive information on it and you have bandwidth to upload it somewhere then I could have a look and try to fix it.

zzzh, 2018-07-20:
@PiotrZierhoffer , actually, I noticed renode because of zyphyr, so I installed zephyr sdk before renode, here's the page I was following: 
http://docs.zephyrproject.org/1.12.0/getting_started/installation_linux.html

I guess there's somthing conflict between zephyr sdk and renode installation

zzzh, 2018-07-20:
@AntonKrug It's above 20GB in size, so may take very long time to upload..

I only installed zephyr thing for it, that's cmake, zephyr sdk and renode , in time order, so I think you should can repeat it by installing them in same order.

I did change apt source, it's called aliyun source because it's quicker for me.
here's the source.list file for apt:
# deb cdrom:[Ubuntu 16.04 LTS _Xenial Xerus_ - Release amd64 (20160420.1)]/ xenial main restricted
deb-src http://archive.ubuntu.com/ubuntu xenial main restricted #Added by software-properties
deb http://mirrors.aliyun.com/ubuntu/ xenial main restricted
deb-src http://mirrors.aliyun.com/ubuntu/ xenial main restricted multiverse universe #Added by software-properties
deb http://mirrors.aliyun.com/ubuntu/ xenial-updates main restricted
deb-src http://mirrors.aliyun.com/ubuntu/ xenial-updates main restricted multiverse universe #Added by software-properties
deb http://mirrors.aliyun.com/ubuntu/ xenial universe
deb http://mirrors.aliyun.com/ubuntu/ xenial-updates universe
deb http://mirrors.aliyun.com/ubuntu/ xenial multiverse
deb http://mirrors.aliyun.com/ubuntu/ xenial-updates multiverse
deb http://mirrors.aliyun.com/ubuntu/ xenial-backports main restricted universe multiverse
deb-src http://mirrors.aliyun.com/ubuntu/ xenial-backports main restricted universe multiverse #Added by software-properties
deb http://archive.canonical.com/ubuntu xenial partner
deb-src http://archive.canonical.com/ubuntu xenial partner
deb http://mirrors.aliyun.com/ubuntu/ xenial-security main restricted
deb-src http://mirrors.aliyun.com/ubuntu/ xenial-security main restricted multiverse universe #Added by software-properties
deb http://mirrors.aliyun.com/ubuntu/ xenial-security universe
deb http://mirrors.aliyun.com/ubuntu/ xenial-security multiverse

AntonKrug, 2018-07-20:
I might try to replicate it next week.
Noticed you have xenial-backports there as well, backports can confuse sometimes the dependancies.

zzzh, 2018-07-20:
I understand it's complicate to get it good for all the situations people may have on their computers, so why don't you guys put it in a docker? that would make things much easier.

AntonKrug, 2018-07-20:
Ok I will rerun them exactly on my VM, will start from scratch to make sure. But that will be next week.

AntonKrug, 2018-07-23:
So I followed your steps, the tujikawa/ppa showed me some errors (I personaly do not like the PPAs, they often break stuff). Then I installed the mono and the rest of dependencies as you show (as it's in the http://renode.readthedocs.io/en/latest/advanced/building_from_sources.html#building-from-source)

Then I wanted quickly see if I could run renode, so I did:

```
wget https://github.com/renode/renode/releases/download/v1.4.1/renode_1.4.1_amd64.deb
sudo dpkg -i renode_1.4.1_amd64.deb
```

And realized that for some reason I had only mono 4.x so then I used the mono project website 
http://www.mono-project.com/download/stable/#download-lin
As their steps are slightly different (xenial vs stable-xenial)
This cause a error similar to yours so I had to `sudo apg-get remove renode` first to install their mono, then rerun the `sudo apt-get install git mono-complete automake autoconf libtool g++ realpath gksu libgtk2.0-dev screen uml-utilities gtk-sharp2 python2.7` step and only then installed back the renode `sudo dpkg -i renode_1.4.1_amd64.deb` and that worked well, I didn't try to compile from the source today, but the binary works and stuff is installed.

Could you maybe start again and for the mono step follow the mono-project steps?

```
sudo apt update
sudo apt upgrade
sudo apt-get install --no-install-recommends git cmake ninja-build gperf ccache doxygen dfu-util device-tree-compiler python3-ply python3-pip python3-setuptools python3-wheel xz-utils file make gcc-multilib autoconf automake libtool

sudo add-apt-repository ppa:plushuang-tw/uget-stable
sudo add-apt-repository ppa:t-tujikawa/ppa
sudo apt update
sudo apt-get install uget
sudo apt-get install aria2

sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF
sudo apt install apt-transport-https
echo "deb https://download.mono-project.com/repo/ubuntu stable-xenial main" | sudo tee /etc/apt/sources.list.d/mono-official-stable.list
sudo apt update

sudo apt-get install git mono-complete automake autoconf libtool g++ realpath gksu libgtk2.0-dev screen uml-utilities gtk-sharp2 python2.7

# optional check
wget https://github.com/renode/renode/releases/download/v1.4.1/renode_1.4.1_amd64.deb
sudo dpkg -i renode_1.4.1_amd64.deb
```

zzzh, 2018-07-23:
@AntonKrug, thank you so much for the help, it must took you a lot of time.

I'll try following your steps.

AntonKrug, 2018-07-23:
Not really :-) it was fine, when it was busy, I was doing some other things in between.

AntonKrug, 2018-07-23:
I don't know the internals well enough to say for 100%.
It could cause problems, but it shouldn't the applications nowadays are more careful and specify exactly what python they want and do not call `python` with fingers crossed that it will give them the python they wish for. In past some programs were like that, now the software is more specific and careful. 

Installing both pythons at the same time is fine when you are using package managers (as they will be careful not step on each other's toes).

Btw what is output of this
```
sudo apt-cache show renode
```
Curious if you have binary renode already installed or not.

zzzh, 2018-07-23:
$ sudo apt-cache show renode
N: Unable to locate package renode
E: No packages found

It's still in previous snapshot since I was busying to solve my samba issue, it kept refusing to work , my way work well in ubuntu 14.04, but not in ubuntu 16.04

AntonKrug, 2018-07-23:
Not finding renode should be fine, as you building from the sources.

zzzh, 2018-07-24:
Gotcha

zzzh, 2018-07-24:
@PiotrZierhoffer that's a good news! thanks for telling me.

zzzh, 2018-07-25:
I tried the mono project steps, provided by @AntonKrug, thank him for the effort.

here's my results:

1. building from the source way, with the -p option
libmonoboehm-2.0-1 install issue had disappeared, but stuck at "Cloning into 'lib/resources'...", error: "RPC failed; curl 18 transfer closed with outstanding read data remaining" or "RPC failed; curl 56 GnuTLS recv error (-54): Error in the pull function."

2. install from renode_1.4.1_amd64.deb
install successed, but with an issue:
(monitor) include @scripts/single-node/miv.repl
No such file scripts/single-node/miv.repl.

and I guess it still have the elf file download issue

PiotrZierhoffer, 2018-07-25:
The first issue is about the connection with GitHub - if you rerun the script it should be better.

The second issue is about a local file: the file name is miv.resc, not repl. Repl are platform description files, that are loaded by resc files - Renode scripts.

closed 2018-07-26 by zzzh

AntonKrug, 2018-07-26:
Using TAB should autocomplete the sentectes and avoid the typos.