renode/renode #23: Unaligned memory accesses on RISC-V are not causing a trap.

AntonKrug, 2018-07-25:
Memory accesses should be aligned to with 0x4 bytes, if unaligned access if generated it will cause a trap. Then the trap should in software detect the mcause=4 and then generate aligned accesses to the memory and in the end get the equivalent of the unaligned access done in software.

In Renode I can't cause the trap happening on unaligned accesses and the software happily writes to the memory even if they are not aligned, without the need of the software trap. Is it this case, or did I got something wrong?

I would say the emulation should be close to the hardware as much as possible and trap whenever the real hardware traps even if Renode can get the same final result without the trap.

If needed I could supply a small replicator of this problem.

PiotrZierhoffer, 2018-07-26:
Yes, please, it would be great to have a way to reproduce easily.

PiotrZierhoffer, 2018-08-13:
I think it is. Unfortunately we did not yet manage to take care of it, as we're swamped with other things. We'll take a look as soon as we can.

AntonKrug, 2018-08-13:
That is ok, take your time.

PiotrZierhoffer, 2018-08-31:
With fcab5b12a38ae25678d8dc2004c3fc38e6f22ee7 it should be fixed now. We are about to release unit tests for stores/loads as well.

This is done unconditionally, but will be changed in future to optionally support misaligned accesses (as the ISA makes it implementation-dependent).

Instruction fetches are not verified at the moment, but we'll treat it as a separate case.

closed 2018-08-31 by PiotrZierhoffer

PiotrZierhoffer, 2018-08-31:
We also generate an mcause=4 exception on AMO instructions, when it should be 6. This is a known limitation we will be working on soon.

AntonKrug, 2018-09-05:
I tested quickly with a store instruction and it works as expected.