renode/renode #26: Bug when parsing qRcmd monitor commands

AntonKrug, 2018-10-31:
When connecting to renode with gdb the monitor commands who do not respond anything are recognized as an unsupported command.

Getters works because they return values, but setter which stays quiet will be seen as unsupported command even when it's working. In the example below I change PC with setter and it gave unsupported command error. But reading it back it clearly shows that the value got updated properly.

This might look like a cosmetic issue but causes havoc on other tools which use gdb (for example Eclipse) which interprets this as a failure and stop executing their flow even if they could happily continue.

```
(gdb) load example.elf
Loading section .text, size 0x2680 lma 0x8000000
Loading section .data, size 0x920 lma 0x8002680
Start address 0x8000000, load size 12192
Transfer rate: 440 KB/sec, 3048 bytes/write.
(gdb) monitor cpu PC
0x8000000
(gdb) monitor cpu PC 0x12345
Target does not support this command.
(gdb) monitor cpu PC
0x12345
```

I think I found the source of the problem and a quick test showed the fix is working, I will format the description of the problem and submit pull request shortly.

AntonKrug, 2018-10-31:
Just noticed the bug is actually caused inside the renode-infrastructure, so not sure if the issue should be moved there, or if it's fine to keep the problems together in the main top repository.

Pull request to fix this bug is here:
https://github.com/renode/renode-infrastructure/pull/3

PiotrZierhoffer, 2018-10-31:
The issue here is fine! Thanks for the report and the pull request, we'll test it and let you know.

I think it makes sense in general, but will verify it

AntonKrug, 2018-10-31:
Yes please do as I'm very raracely using C#

mateusz-holenko, 2018-11-07:
Your code (together with another one of mine) is merged to master. Thanks for the fix!

closed 2018-11-07 by mateusz-holenko

AntonKrug, 2018-11-07:
Thanks for accepting it. And you are welcome. Funny how much impact can one line have down the pipeline of tools.