renode/renode #29: disable-xwt option breaks port option

asutherlandus, 2018-12-06:
In the older version of renode I needed this to suppress the console:
./renode --disable-xwt --port=3320

In the new version this results in:

socat - tcp:localhost:3320
2018/12/06 09:08:23 socat[6815] E connect(5, AF=2 127.0.0.1:3320, 16): Connection refused

Removing --disable-xwt solves this problem as the port option now disables the console pop-up.
./renode --port=3320
Works as expected

For backwards compatibility it would be nice if the --disable-xwt didn't interfere with the port option.

PiotrZierhoffer, 2018-12-06:
Thanks, a good catch. It should definitely work as expected

PiotrZierhoffer, 2018-12-14:
Fixed on master, you can use port with --disable-xwt option or without it. In the latter case the UART windows will open as normal.

closed 2018-12-14 by PiotrZierhoffer