renode/renode #31: Awkward macros can crash the platform

AntonKrug, 2019-01-15:
When trying to make generic CSR fetcher and avoid code duplication for all the 5 cores:
```
$core=sysbus.e51
macro CSR
"""
$core MSTATUS
"""
runmacro $CSR
```

I get to crash the platform consistently: 
```
Fatal error:
Object reference not set to an instance of an object.
   at Antmicro.Renode.UserInterface.Monitor.RunCommand(ICommandInteraction writer, Command command, IList`1 parameters)
   at Antmicro.Renode.UserInterface.Monitor.ExecuteCommand(Token[] com, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.ParseTokens(IEnumerable`1 tokensToParse, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.Parse(String cmd, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.HandleCommand(String cmd, ICommandInteraction ci)
   at AntShell.Shell.HandleCommand(String cmd, ICommandInteraction ic)
   at AntShell.CommandLine.HandleControlSequence(ControlSequence seq)
   at AntShell.Terminal.NavigableTerminalEmulator.HandleInput(Object input)
   at AntShell.Terminal.NavigableTerminalEmulator.Run(Boolean stopOnError)
   at AntShell.Shell.Start(Boolean stopOnError)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart(Object obj)

Unhandled Exception: System.NullReferenceException: Object reference not set to an instance of an object.
   at Antmicro.Renode.UserInterface.Monitor.RunCommand(ICommandInteraction writer, Command command, IList`1 parameters)
   at Antmicro.Renode.UserInterface.Monitor.ExecuteCommand(Token[] com, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.ParseTokens(IEnumerable`1 tokensToParse, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.Parse(String cmd, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.HandleCommand(String cmd, ICommandInteraction ci)
   at AntShell.Shell.HandleCommand(String cmd, ICommandInteraction ic)
   at AntShell.CommandLine.HandleControlSequence(ControlSequence seq)
   at AntShell.Terminal.NavigableTerminalEmulator.HandleInput(Object input)
   at AntShell.Terminal.NavigableTerminalEmulator.Run(Boolean stopOnError)
   at AntShell.Shell.Start(Boolean stopOnError)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart(Object obj)
```

I would say the macros with errors should be reject, but not cause whole platform to crash.

PiotrZierhoffer, 2019-01-21:
You're definitely right, it should not crash. I am not sure if this specific code should work or not (need to verify), but crashing is not acceptable.

AntonKrug, 2019-01-21:
I wish it would, but probably it will not and that's fine (I'm experimenting with how far I can get with macros). But platform shouldn't crash either way.

PiotrZierhoffer, 2019-01-22:
I can't seem to reproduce this error. I get a parsing error, but not a crash. Could you provide more details? What commands were executed before you issued these lines?

PiotrZierhoffer, 2019-01-22:
What's more, I get it to work as intended. The grammar here is not pretty, but it is doable:

```
(monitor) mach create
(machine-0) machine LoadPlatformDescription @platforms/cpus/polarfire.repl 
(machine-0) $core=sysbus.e51
(machine-0) macro CSR
> """
> `echo $core` MSTATUS
> """
```

What is your exact Renode version?

AntonKrug, 2019-01-22:
1. When making the macro there was a parsing error
2. Crash only happened after trying to execute such macro.

I will give the workaround a try, clearly what I do is slightly abusing the macros.

AntonKrug, 2019-01-22:
Now you are referencing the original macro, or your proposed workaround? The `echo $core` workaround works for me too. Thanks.

PiotrZierhoffer, 2019-01-23:
I did not update my code, sorry for that. Updated.

I meant my workaround. Still, I cannot reproduce the crash. I will try a version bundled with SC.

PiotrZierhoffer, 2019-01-31:
@AntonKrug I never managed to replicate the crash. Can you produce a full log of commands executed by Renode to get there? Or a SC snapshot doing this? 

If it's not replicable I think we could close the ticket

AntonKrug, 2019-01-31:
At some version it doesn't happen, will have to look at what exact conditions it happened. If you want,  close it in the meantime and reopen it when I find proper conditions/steps.

PiotrZierhoffer, 2019-01-31:
Let's do it like that then

closed 2019-01-31 by PiotrZierhoffer