renode/renode #32: Closing the platform directly after it started outputs odd error

AntonKrug, 2019-01-16:
I'm wondering if I do something strange with my platform, when I start it I can connect with the GDB and use it as usually. But when I launch, do not do anything (not even connecting with GDB) and I close it straight after I get this error:

```
11:35:49.7209 [DEBUG] machine-0: Disposing sysbus.e51.
11:35:49.7289 [DEBUG] machine-0: Disposing sysbus.u54_1.
11:35:49.7309 [DEBUG] machine-0: Disposing sysbus.u54_2.
11:35:49.7350 [DEBUG] machine-0: Disposing sysbus.u54_3.
11:35:49.7375 [DEBUG] machine-0: Disposing sysbus.u54_4.
11:35:49.7405 [DEBUG] machine-0: Disposing sysbus.
11:35:49.7415 [DEBUG] machine-0: Disposing sysbus.mmc.
11:35:49.7420 [DEBUG] machine-0: Disposing sysbus.envmData.
11:35:49.7425 [DEBUG] machine-0: Disposing sysbus.lim.
11:35:49.7425 [DEBUG] machine-0: Disposing sysbus.ram.
11:35:49.7425 [DEBUG] machine-0: Disposing sysbus.ddr.
11:35:49.7435 [INFO] machine-0: Disposed.
Fatal error:
An action requiring GUI environment detected, but the XWT provider is not started.
   at Antmicro.Renode.UI.ApplicationExtensions.CheckXwtStatus()
   at Antmicro.Renode.UI.ApplicationExtensions.InvokeInUIThread(Action action)
   at TermSharp.Vt100.ByteUtf8Decoder.Feed(Byte b)
   at AntShell.Terminal.BasicTerminalEmulator.SendControlSequence(Byte[] seq)
   at AntShell.Terminal.BasicTerminalEmulator.SendCSI(Byte[] seq)
   at AntShell.Terminal.NavigableTerminalEmulator.Stop()
   at System.Action.Invoke()
   at Antmicro.Renode.Emulator.Exit()
```

AntonKrug, 2019-01-22:
Screenshots for visual reference:

[image]
[image]

truhlikfredy, 2019-01-28:
I found it while using SoftConsole on Windows 10, but looks like it has nothing to do with SoftConsole or Windows10. I just downloaded Renode 1.6.2, extracted zip and run it on Windows 7 and I can replicate the issue as well.
Starting Renode:
[image]

Crash:
[image]

AntonKrug, 2019-01-29:
@tm1234 Could replicate it on his Windows 10 host and on a clean Win10 VM as well.

TM1234, 2019-01-29:
Yes - I did a clean install of Windows 10 on a new VirtualBox VM and was able to reproduce it using just Renode 1.6 (no SoftConsole or any Microchip related stuff) and a non RISC-V platform description:

1. Close Renode console window using the "x" window frame control = the error mentioned above
2. Close Renode by issuing the Renode "quit" command = error
3. Close Renode by typing Ctrl-C in the command shell/console that launched Renode = no error

Here is a screenshot of a run of Renode which was terminated using the quit command.

Hope this helps.
[image]

TM1234, 2019-01-29:
Actually even if I just run renode.exe with no script at all and run the "quit" command in the Renode console when it appears I get the same error in the command shell from which renode.exe was run.

[image]
[image]

PiotrZierhoffer, 2019-01-29:
It seems to be Windows-specific. Thanks for the info, we'll verify it!

PiotrZierhoffer, 2019-01-31:
It turns out to be a difference in .NET and Mono internals. Quite interesting, and showed some bad design on our side. Should be relatively easy to fix, most likely next week.

PiotrZierhoffer, 2020-05-14:
Apparently this was a much more complex problem than anticipated at first.

We have not fixed the underlying cause, as it is quite complex and would involve redoing a lot of Renode's flow. We had an attempt on that, but the cost outweighed the benefit.

We decided to hush the warning for now, fixed by @ajelinski  in ea1c2e7

closed 2020-05-14 by PiotrZierhoffer