renode/renode #35: Which robot module should be added?

licunhao, 2019-04-24:
I found module robot it needed by renode.
I have tried to use pip install robot to fix it.
But it have another problem.

- 1. [ ] 
- 2. [ ] Traceback (most recent call last):
- 3. [ ] File "/opt/renode/tests/run_tests.py", line 10, in <module>
- 4. [ ] tests_engine.run()
- 5. [ ] File "/opt/renode/tests/tests_engine.py", line 96, in run
- 6. [ ] if not suite.run(options):
- 7. [ ] File "/opt/renode/tests/robot_tests_provider.py", line 118, in run
- 8. [ ] _suite = robot.parsing.model.TestData(source=self.path)
- 9. [ ] AttributeError: 'module' object has no attribute 'parsing'
- 10. [ ] 

Could you please help me to solve this problem?

mateusz-holenko, 2019-04-24:
Have you tried installing packages listed [in the documentation](https://renode.readthedocs.io/en/latest/advanced/building_from_sources.html#additional-prerequisites-for-robot-framework-testing)?

AntonKrug, 2019-04-24:
You might need netifaces (and the other 2 dependencies), for which you need python-dev on Linux. And on Windows you have to follow the link in the error you will get when installing pip install netifaces

licunhao, 2019-04-25:
@AntonKrug @mateusz-holenko 
Thanks for your reply.
I solved the problem as yours.
I close the issue.

closed 2019-04-25 by licunhao