renode/renode #39: Website should describe what hardware is supported

mithro, 2019-08-02:
It is impossible to figure out what hardware is emulated out of the box.

There are a lots of boards @ https://github.com/renode/renode/tree/master/platforms/boards

There are lots of CPUs @ https://github.com/renode/renode/tree/master/platforms/cpus
 * ARM SoCs
   * stm32
   * silabs happy geko
   * Tegra?
 * RISC-V
   * SiFive RISC-V parts
   * PolarFire SoC?
   * LiteX

mithro, 2019-08-02:
@mgielda - Was trying to show people at PyCon AU and the first question was "does it support XYZ hardware".

mithro, 2019-08-02:
There is a bunch of useful information at https://antmicro.com/products/renode/ which doesn't appear on your renode.io website...

AntonKrug, 2019-08-02:
On similar topic the polarfire.repl should be renamed to polarfire-soc.repl (polarfire is the previous generation without hard CPUs)

PiotrZierhoffer, 2019-08-08:
@AntonKrug this will be addressed shortly, thanks for pointing that out

whoopsmith, 2019-09-01:
Is there any support for a M4F using hardfloat?

PiotrZierhoffer, 2019-10-09:
@whoopsmith yes, the core itself supports hardfloat. I believe some of our EFR platforms use it, but we don't have a demo for it

mgielda, 2019-10-09:
@AntonKrug polarfire was renamed to polarfire-soc. @mithro thanks for the help - the list is now live at https://renode.readthedocs.io/en/latest/introduction/supported-boards.html, there is now a link from the website leading there (we will be rebuilding the website in the more general sense soon-ish, so foregoing adding the entire list there as it's hard to fit it to the theme easily in this iteration)

We will be adding to the supported platforms list once we have demos for some more platforms that are supported, but I think we have a good start. Closing.

closed 2019-10-09 by mgielda