renode/renode #40: Best practice to add a board/CPU?

yuhui-zheng, 2019-08-11:
Hello,

I would like to see if there's additional documentation that I can follow to add a "customized" board/CPU. 

At this point, I'm specifically interested in WiFi interface, where the underlying WiFi component can be either on-chip or onboard. (WiFi driver is likely vended to us in binary, and data sheet is likely incomplete..) With Renode, the hope is to have as little modification in kernel/OS code as possible, so that the emulation result resembles result running on a real piece of hardware. (In a slightly different approach, I guess we could "mock" WiFi in kernel/OS whenever running on emulator... ) Appreciate any tip!

mgielda, 2019-08-19:
Hello Yuhui!

First, happy to see you using Renode. Basically what we're aiming for is 0 modification to the emulated code - so it's indeed a good fit to what you want to achieve. Of course, simulating WiFi is not trivial, as it's typically it's own CPU running it's own software, but that would be the "Renode" way to do things. Incomplete datasheets is something we deal with, of course unless it's completely unworkable.

Take a look at https://github.com/renode/renode/tree/master/platforms/boards for various boards that we have, you'd need an analogous `.repl` (Renode platform) file to create a new board - the file puts together existing models into a platform. Of course, for the WiFi to work, you'd need a model. Happy to help you do that, but for this we would need more details. Will reach our privately as well.

yuhui-zheng, 2019-08-20:
Thank you @mgielda. Taking discussion offline, and closing this issue.

closed 2019-08-20 by yuhui-zheng