renode/renode #42: Zephyr DTS and Renode LiteEth address mismatch

kaolpr, 2019-09-21:
I recently started my adventure with Renode and one of my first attempts was to get working `samples/net/echo` from Zephyr on `litex_vexriscv` CPU with Renode LiteEth simulation. Unfortunately that was not as straightforward as one could expect.

It turns out that there are different addresses hardcoded for LiteEth in [Zephyr DTS](https://github.com/zephyrproject-rtos/zephyr/blob/5e0bfafcf889d9787665e25b5a271265ddf5678a/dts/riscv/riscv32-litex-vexriscv.dtsi#L56-L60) and here:

https://github.com/renode/renode/blob/d95794c1582df18d8e505814c47f547fe917b699/platforms/cpus/litex_common.repl#L26

It's hard to say it's a bug, but for me, as a newcomer, it was challenging (yet very educational) to figure out what's wrong. Maybe you could make it compatible out-of-the-box?

mithro, 2019-09-23:
@kaolpr There is a tool for generating a renode configuration from a LiteX CSV file here -> https://github.com/litex-hub/litex-renode

kaolpr, 2019-09-23:
I know and it's very handy, already checked it out.

The only think I wanted to point out with this issue is that Renode + Litex + Zephyr simulation looks like a great match and would be nice if it could work out-of-the-box. 

But I fully understand it's not a bug and one can not consider it a reason for any modifications (yet, this discussion will be here if anyone was looking for Zephyr + LiteEth + Renode).

mateusz-holenko, 2019-09-24:
Thanks for the comment, @kaolpr!

As you probably already know, LiteX is in fact a SoC builder which means it can generate many different configurations.
Those configurations can differ in many aspects, including base addresses of internal components.
That flexibility is great, but as a result you don't have *one* LiteX platform.

Renode, due to its modularity, is great to simulate such situation as platform configuration is described in a simple text repl file.
All you need to do to simulate another instance of LiteX is to prepare a custom version of repl platform.
As mithro has already mentioned, there is a generator that can help you with that.

You are totally right though, that it might be surprising at first.
Renode provides LiteX platform definition, but it cannot run mainline Zephyr out-of-the-box.
I want to make it clear (also for others that might stumble upon this thread in the future) that this is not a bug, what Renode comes with is just one-of-many possible LiteX platforms.

There is, however, a solution to your problem that should satisfy everyone.
I will simply add a new repl and a script that explicitly targets Zephyr OS, so that it's easier for newcomers to play with Renode + Zephyr.

mateusz-holenko, 2019-09-24:
I added the Zephyr-specific platform and script in https://github.com/renode/renode/commit/8383ccfcd404cc56399a07f94a44595233e9fb45.

Hope it solves the problem!

closed 2019-09-24 by mateusz-holenko