renode/renode #43: Bad IL Format

Jesse-Millwood, 2019-10-04:
I tried using the renode 1.8 deb package on Ubuntu 18.04. Any of the example scripts I try to run give this error: 

```
Bad IL format.
  at Antmicro.Renode.UserInterface.Monitor.RunCommand (AntShell.Commands.ICommandInteraction writer, Antmicro.Renode.UserInterface.Commands.Command command, System.Collections.Generic.IList`1[T] parameters) [0x00429] in <0960209a5fcc4ab78b093e288fdc0e38>:0 
  at Antmicro.Renode.UserInterface.Monitor.ExecuteCommand (Antmicro.Renode.UserInterface.Tokenizer.Token[] com, AntShell.Commands.ICommandInteraction writer) [0x001f3] in <0960209a5fcc4ab78b093e288fdc0e38>:0 
  at Antmicro.Renode.UserInterface.Monitor.ParseTokens (System.Collections.Generic.IEnumerable`1[T] tokensToParse, AntShell.Commands.ICommandInteraction writer) [0x001d5] in <0960209a5fcc4ab78b093e288fdc0e38>:0 
  at Antmicro.Renode.UserInterface.Monitor.Parse (System.String cmd, AntShell.Commands.ICommandInteraction writer) [0x001a7] in <0960209a5fcc4ab78b093e288fdc0e38>:0 
  at Antmicro.Renode.UserInterface.Monitor.ProcessCommand (System.String command, AntShell.Commands.ICommandInteraction writer) [0x00000] in <0960209a5fcc4ab78b093e288fdc0e38>:0 
  at Antmicro.Renode.UserInterface.Monitor.HandleCommand (System.String cmd, AntShell.Commands.ICommandInteraction ci) [0x00000] in <0960209a5fcc4ab78b093e288fdc0e38>:0 
  at AntShell.Shell.HandleCommand (System.String cmd, AntShell.Commands.ICommandInteraction ic) [0x000bd] in <a6fe5939364043ceac8ab1d449585e49>:0 
  at AntShell.CommandLine.HandleControlSequence (AntShell.Helpers.ControlSequence seq) [0x00a45] in <a6fe5939364043ceac8ab1d449585e49>:0 
  at AntShell.Terminal.NavigableTerminalEmulator.HandleInput (System.Object input) [0x0002e] in <a6fe5939364043ceac8ab1d449585e49>:0 
  at AntShell.Terminal.NavigableTerminalEmulator.Run (System.Boolean stopOnError) [0x00017] in <a6fe5939364043ceac8ab1d449585e49>:0 
  at AntShell.Shell.Start (System.Boolean stopOnError) [0x000f9] in <a6fe5939364043ceac8ab1d449585e49>:0 
  at Antmicro.Renode.UI.CommandLineInterface+<>c__DisplayClass0_1.<Run>b__3 (System.Object x) [0x00000] in <13cc38bdc7d14607a8d87be2bcf0d6ab>:0 
  at System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) <0x7f6743898110 + 0x0008f> in <285579f54af44a2ca048dad6be20e190>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) <0x7f6743895a80 + 0x0014d> in <285579f54af44a2ca048dad6be20e190>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) <0x7f67438959c0 + 0x00041> in <285579f54af44a2ca048dad6be20e190>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) <0x7f6743895930 + 0x0004d> in <285579f54af44a2ca048dad6be20e190>:0 
  at System.Threading.ThreadHelper.ThreadStart (System.Object obj) <0x7f67438981f0 + 0x0005b> in <285579f54af44a2ca048dad6be20e190>:0 
```

My system: 
Mono JIT compiler version 6.4.0.198 (tarball Tue Sep 24 01:21:28 UTC 2019)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
	TLS:           __thread
	SIGSEGV:       altstack
	Notifications: epoll
	Architecture:  amd64
	Disabled:      none
	Misc:          softdebug 
	Interpreter:   yes
	LLVM:          yes(610)
	Suspend:       hybrid
	GC:            sgen (concurrent by default)

PiotrZierhoffer, 2019-10-04:
Hi @Jesse-Millwood 

Thanks for your report. This is a known problem that is already fixed on Renode master. We will probably release a 1.8.1 version with this fix early next week.

I hope this helps, but I will keep the issue open until we release new packages.

Jesse-Millwood, 2019-10-04:
Sounds good

Thanks Piotr!

For what it's worth, it seems that it happened right after my package manager updated everything including a bunch of mono packages. It seems to be working on my Gentoo machine with this mono install: 
```Mono JIT compiler version 6.0.0.334 (tarball Fri Sep 20 21:38:29 EDT 2019)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
	TLS:           __thread
	SIGSEGV:       altstack
	Notifications: epoll
	Architecture:  amd64
	Disabled:      none
	Misc:          softdebug 
	Interpreter:   yes
	LLVM:          supported, not enabled.
	Suspend:       hybrid
	GC:            sgen (concurrent by default)
```

PiotrZierhoffer, 2019-10-09:
Yes, it seems the bug was introduced around mono 6.4.

Anyway, the package with v1.8.1 and the relevant bugfix is released, so I'm closing this one.

closed 2019-10-09 by PiotrZierhoffer