renode/renode #44: Renode locks up when running CPU

xobs, 2019-10-29:
I'm looking at using Renode to develop for RISC-V, however it seems to be locking up.

renode-init.resc
```
using sysbus
mach create
machine LoadPlatformDescription $ORIGIN/base.repl
showAnalyzer uart
sysbus LoadELF "C:/Users/smc/Code/renode-test/target/riscv32i-unknown-none-elf/debug/renode-test"
start
```

base.repl:
```
rom: Memory.MappedMemory @ sysbus 0x20000000
    size: 0x00800000

sram: Memory.MappedMemory @ sysbus 0x10000000
    size: 0x00008000

cpu: CPU.VexRiscv @ sysbus
    timeProvider: cpu_timer
    cpuType: "rv32ima"
    privilegeArchitecture: PrivilegeArchitecture.Priv1_10

timer0: Timers.LiteX_Timer @ sysbus 0xe0002800
    frequency: 100000000
    -> cpu@1

uart: UART.LiteX_UART @ sysbus 0xe0001800
    -> cpu@0

cpu_timer: Timers.LiteX_CPUTimer @ sysbus 0xe0000800
    // Run the clock at 12 MHz
    frequency: 12000000
    // IRQ #100 is Machine Timer Interrupt
    -> cpu@100
```

[renode-test.zip](https://github.com/renode/renode/files/3782501/renode-test.zip) -- compressed version of the elf file

This is an experiment to get Rust working with this system, so I'm unsure of where the problem lies.  However, I do know for a fact that renode is locking up, and Windows marks it as "Not Responding".

Windows 10, version 1.8.1.30148 (db38fa84-201910091635)

mithro, 2019-10-29:
FYI - @mateusz-holenko

PiotrZierhoffer, 2019-10-29:
I can confirm it on my machine. We will investigate this. A hack that will reduce the performance but will also allow you to move forward is running `cpu LogFunctionNames true` before running `start`.

Somehow this prevents the lock up.

PiotrZierhoffer, 2019-10-29:
The same problem can be reproduced on Linux. Can you provide some ways for us to reproduce your elf?

xobs, 2019-10-30:
The elf is the result of experimentation in Rust.  I've uploaded the project to https://github.com/xobs/untitled-rust-crate

To use:

1. Install rust (go to https://rustup.rs/) v1.38 or newer
2. Install riscv support: `rustup target add riscv32i-unknown-none-elf`
3. Run `cargo build`

The resulting elf file will be `target/riscv32i-unknown-none-elf/debug/renode-test`

mateusz-holenko, 2019-10-30:
It seems to be a problem with handling looong lines in our terminal emulalor.
The software writes a lot of text, but doesn't generate any line break.

On Linux there is an option to switch to other terminals (XTerm, GnomeTerminal) and the binary works fine on them.

xobs, 2019-10-30:
Alright, I'll do that for now.

This was more of a sanity test to check whether everything was working, so it's no problem to work around for now.