renode/renode #46: Add Wishbone bus support to VerilatorPlugin

xobs, 2019-12-02:
Currently the `VerilatorPlugin` only supports the AXI4-Lite bus.  However, many open source cores are actually Wishbone devices.

It would be nice to create a wishbone bus master interface to allow us to simulate verilog-based wishbone buses prior to creating the C# models of them.

mateusz-holenko, 2020-01-29:
It’s indeed a good idea - in fact we have already internally planned to add this support in the future. We don’t have, however, any specific roadmap planned for it at the moment.

As a workaround you can always include a Wishbone-to-AXI bridge in your Verilator simulation so that you talk to Renode over the currently supported AXI4-Lite.

mithro, 2020-01-29:
I would say that wishbone is a high priority for me.

mateusz-holenko, 2020-02-19:
https://github.com/renode/renode/commit/70ce00f346d3038e86e483e2b460c4a16ff054b6 adds Wishbone bus support to Verilator Plugin.

We are currently working on a demo for this feature.

mateusz-holenko, 2020-02-26:
We have published a demo of LiteX/VexRiscv platform with verilated LiteUART (using wishbone bus) in Renode: https://github.com/renode/renode/commit/a711027e6cd34ed613ecab1d799d629f7027f7da .

There are also commits adding support for interrupts from the verilated peripheral.

mateusz-holenko, 2020-03-04:
We published sources of the demo of LiteX/VexRiscv platform with verilated LiteUART using wishbone bus at https://github.com/antmicro/renode-verilator-integration/tree/master/samples/liteuart.

mateusz-holenko, 2020-03-04:
@xobs, @mithro: If you don't have any additional insights/comments, I guess we can close this issue?

xobs, 2020-03-05:
@mateusz-holenko Sorry to be obtuse, but how was `top.v` generated?  Was it built using the `Sim` platform?  What did you do to expose the Wishbone bus out the top?  Similar to what I did with spibone (https://github.com/xobs/spibone/blob/master/sim/generate-verilog.py)?

mateusz-holenko, 2020-03-05:
@xobs We generated the `top.v` file using the script from LiteX: https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_gen.py with `./tools/litex_gen.py --with-uart`.
It uses the `wishbone` bus by default, but can be changed to `axi` as well.

Exposing the bus is done in https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_gen.py#L99, which uses additional functions we've recently added: https://github.com/enjoy-digital/litex/blob/master/litex/soc/interconnect/wishbone.py#L73 and https://github.com/enjoy-digital/litex/blob/master/litex/soc/interconnect/wishbone.py#L80.

PiotrZierhoffer, 2020-05-18:
I think we can close this one. @xobs if you have any more questions/inquiries about this, please let us know

closed 2020-05-18 by PiotrZierhoffer