renode/renode #47: Incorrect keyboard mapping in Renode terminal

jeandudey, 2019-12-11:
Currently I assume the Renode terminal has a default keymap of `en_US` (these are the keys that match that keymap when I type), for example, trying to type the `"` character (it's `Shift + 2`, on `es_ES`) results in the `@` character. That happens with other characters too.

Example: should be `"`

[image]

**Platform:** Windows 10
**Renode version:** 1.8.2

adamsanta, 2022-08-17:
Hello,

I did not find any way to change the keymap directly in the Renode terminal, but have found a workaround on Windows:

1. Launch renode in telnet mode in a PowerShel with l: renode -P 1234 (https://renode.readthedocs.io/en/latest/basic/running.html)

2. Run Putty in telnet mode to connect to your Renode monitor with IP=127.0.0.1 and port=1234 (connection type is: other)

3. Your Putty terminal now shows the Renode monitor and you can enjoy the original keymap of your laptop to interact with it ;)

Bktero, 2022-09-14:
I have the same issue. I have just installed Renode (version 1.13.1.21690 (a133ada0-202207251201)) using the msi installer for Windows. My system uses French (keymap, language) and Renode terminal uses an EN keymap.

Bktero, 2022-09-14:
I have a similar issue. I have just installed Renode (version 1.13.1.21690 (a133ada0-202207251201)) using the msi installer for Windows. My system uses French (keymap, language) and Renode terminal uses an ES keymap.