renode/renode #51: GDB: CPU locks up when resetting $pc

xobs, 2019-12-23:
When using gdb, the CPU appears to lock up when attempting a reset by setting `$pc`.

Steps:

1. Start a server by running `machine StartGdbServer 3333`
2. Attach a debugger.  This can be done with e.g. `riscv64-unknown-elf-gdb -ex 'target remote localhost:3333' target.elf`
3. Start emulation `mon start`
4. Break emulation and reset `$pc`: `set $pc=0x20500000`
5. Attempt to continue execution

From this point onward, the emulated device will be completely unresponsive.

```
Remote debugging using localhost:3333
_start () at asm.S:27
27      asm.S: No such file or directory.
(gdb) mon start
Starting emulation...
(gdb) c
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0x20504268 in __wfi ()
(gdb) set $pc=0x20500000
(gdb) stepi
27      in asm.S
(gdb)
27      in asm.S
(gdb)
27      in asm.S
(gdb) c
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
_start () at asm.S:27
27      in asm.S
(gdb) stepi
27      in asm.S
(gdb)
27      in asm.S
(gdb)
```

pgielda, 2019-12-24:
Can you provide more info? What architecture is it?
Also -  just to make sure I understand correctly, in your binary, is 0x20500000 == asm.S:27 == _start (= you're stack at the PC you've set) ?

xobs, 2019-12-24:
This is an ideal `RISC-V` architecture -- https://git.xobs.io/xobs/xous-platonic

Yes, my binary begins at `0x20500000` which is why I'm setting that as the `$pc`.  Unfortunately the `_start` symbol doesn't appear to work for whatever reason.

I'm trying to inspect something during the startup of the program, and since Renode takes ~30 seconds to load I'm trying to come up with a way to restart emulation without reloading the whole program.

pgielda, 2019-12-24:
Renode definitely does not take 30 seconds to load normally. You probably mean more things than just starting Renode.

Also there is multiple ways to handle "re-loading":

(1) You do not have to restart Renode to start from scratch - you can do "emulation Clear" and load your script again.

(2) You can also define a "reset" macro and reboot your board. Usually this macro will contain things like reloading your binaries, setting pc, etc.

I could potentially help you more thoroughly if you're able to share your scripts with me, with an example binary (can be stripped of all the meaningful stuff).

We should definitely debug 30 second startup time, this is not normal. Can you do:
time renode -e "q"
and send me the results?
What machine/os?

xobs, 2019-12-24:
You're right, I'm taking into account time to re-attach gdb (5-10 seconds) and the time to re-define things like breakpoints.

```powershell
> measure-command {D:\Code\Renode\renode\output\bin\Release\Renode -e q}                                                        Fatal error:
An action requiring GUI environment detected, but the XWT provider is not started.
   at Antmicro.Renode.UI.ApplicationExtensions.CheckXwtStatus()
   at Antmicro.Renode.UI.ApplicationExtensions.InvokeInUIThread(Action action)
   at TermSharp.Vt100.ByteUtf8Decoder.Feed(Byte b)
   at AntShell.Terminal.BasicTerminalEmulator.SendControlSequence(Byte[] seq)
   at AntShell.Terminal.BasicTerminalEmulator.SendCSI(Byte[] seq)
   at AntShell.Terminal.NavigableTerminalEmulator.Stop()
   at System.Action.Invoke()
   at Antmicro.Renode.Emulator.Exit()

Days              : 0
Hours             : 0
Minutes           : 0
Seconds           : 5
Milliseconds      : 399
Ticks             : 53993590
TotalDays         : 6.24925810185185E-05
TotalHours        : 0.00149982194444444
TotalMinutes      : 0.0899893166666667
TotalSeconds      : 5.399359
TotalMilliseconds : 5399.359
>
```

This is running on Windows 10 x64.

xobs, 2019-12-24:
The command `emulation Clear` doesn't look like it works:

```
(machine-0) emulation clear
There was an error executing command 'emulation clear'
emulation does not provide a field, method or property clear.
(machine-0) emulation Clear
There was an error executing command 'emulation Clear'
emulation does not provide a field, method or property Clear.
(machine-0) 
```

I think the easiest thing to do would be to do option (2) and define a `reset` command that sets `$pc` to the correct value, and/or reloads the elf file.  How would I do that?

PiotrZierhoffer, 2019-12-24:
Actually it's just `Clear` (methods from EmulationManager are available directly, and Clear is not a part of the Emulation type).

To define a macro, in your script:

```
macro reset
"""
   sysbus LoadELF ...
   cpu PC 0x001234
"""

runMacro $reset
```

Then, whenever you run ` machine Reset`, this macro will be executed.

PiotrZierhoffer, 2019-12-24:
The `runMacro` command is needed so that the script actually executes the macro when loaded.

xobs, 2019-12-24:
Hmm... for whatever reason, the macro works once, but never again.  My setup script contains this:

```
macro reset
"""
    sysbus LoadELF @../kernel/target/riscv32i-unknown-none-elf/debug/xous-kernel
    cpu PC 0x20500000
"""
runMacro $reset
```

This works.

When I try to run it via gdb, it does not:

```
(gdb) b xous_kernel::mem::MemoryManager::new
Breakpoint 1 at 0x20503058: file src/mem.rs, line 66.
(gdb) mon start
Starting emulation...
(gdb) c
Continuing.

Breakpoint 1, xous_kernel::mem::MemoryManager::new () at src/mem.rs:66
warning: Source file is more recent than executable.
66                  flash: [0; FLASH_PAGE_COUNT],
(gdb) c
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
0x205067e4 in __wfi ()
(gdb) mon runMacro $reset
There was an error executing command 'sysbus LoadELF D:\Code\Xous\kernel\target\riscv32i-unknown-none-elf\debug\xous-kernel'
Cannot load ELF on an unpaused machine.
There was an error executing command 'runMacro $reset'
' failed.ine '    sysbus LoadELF @../kernel/target/riscv32i-unknown-none-elf/debug/xous-kernel
(gdb)
```

xobs, 2019-12-24:
Is the machine not paused when I break into gdb?  The CPU doesn't look like it's advancing, at least.

PiotrZierhoffer, 2019-12-24:
I will need to investigate the precise wording of this warning, but the point is that you need to run `mon machine Reset`.

Running the macro just executes the commands you provided, without changing the state of the CPU, memory etc

xobs, 2019-12-24:
I added `machine Reset` to the macro:

```
macro reset
"""
    machine Reset
    sysbus LoadELF @../kernel/target/riscv32i-unknown-none-elf/debug/xous-kernel
    cpu PC 0x20500000
"""
```

Now it's giving me packet errors:

```
(gdb) mon runMacro $reset
(gdb) b xous_kernel::mem::MemoryManager::new
Breakpoint 1 at 0x20503058: file src/mem.rs, line 66.
(gdb) mon start
Starting emulation...
(gdb) c
Continuing.

Breakpoint 1, xous_kernel::mem::MemoryManager::new () at src/mem.rs:66
warning: Source file is more recent than executable.
66                  flash: [0; FLASH_PAGE_COUNT],
(gdb) c
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
0x205067e4 in __wfi ()
(gdb) mon runMacro $reset
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
```

Eventually Renode crashes:

```
20:10:26.4771 [DEBUG] machine-0: Reset request: doing nothing, because system is not started.
20:10:26.4771 [DEBUG] sysbus: Loading ELF D:\Code\Xous\kernel\target\riscv32i-unknown-none-elf\debug\xous-kernel.
20:10:26.4888 [INFO] sysbus: Loading segment of 36172 bytes length at 0x20500000.
20:10:26.4898 [DEBUG] sysbus: Segment loaded.
20:10:27.4067 [INFO] cpu: Setting PC value to 0x20500000.
20:11:19.6001 [INFO] machine-0: Machine started.
20:11:20.3096 [DEBUG] cpu: Added hook @ 0x20503058
20:11:20.5088 [DEBUG] cpu: Executing hooks registered at address 0x20503058
20:11:22.4951 [DEBUG] cpu: Added hook @ 0x20503058
20:11:28.9461 [INFO] machine-0: Machine paused.
20:11:35.1608 [WARNING] spi.flash: Transmission finished in the unexpected state: RecognizeOperation (2259)

Process is terminated due to StackOverflowException.
```

pgielda, 2019-12-24:
That is because you call "machine Reset" within the reset macro. "machine Reset" calls this macro, so you called it recursively until you run out of stack.
Its a good idea to check for that and display an error instead, but in any case what @PiotrZierhoffer meant is you can call this macro using "machine Reset" not to put it into the macro itself.

xobs, 2019-12-24:
Ah, so does `machine Reset` implicitly call the `$reset` macro?

PiotrZierhoffer, 2019-12-24:
That's correct, if it's defined. It also clears the CPU state, the memory and calls the Reset method from each peropheral

PiotrZierhoffer, 2019-12-27:
Getting back to the original issue, I can't reproduce your problem. I use standard scripts and binaries provided with Renode, could you try the same setup as well?

In Renode's Monitor I run:
```
(monitor) i @scripts/single-node/sifive_fe310.resc 
(SiFive-FE310) machine StartGdbServer 3333
```

In GDB (I use Zephyr's toolchain, but it should not matter):
```
(gdb) tar rem :3333
Remote debugging using :3333
0x20400000 in ?? ()
(gdb) mon s
Starting emulation...
(gdb) c
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0x204018d4 in ?? ()
(gdb) set $pc=0x20400000
(gdb) stepi
0x20400004 in ?? ()
(gdb) 
0x20400008 in ?? ()
(gdb) 
0x2040000c in ?? ()
(gdb) c
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0x204000e4 in ?? ()
```

I get two prompts on the UART window as well.
```
***** BOOTING ZEPHYR OS v1.10.0-rc1 *****
shell> ***** BOOTING ZEPHYR OS v1.10.0-rc1 *****
shell> 
```

Do I understand correctly that your CPU is stuck in WFI when you execute this procedure? It might be the reason why it's stuck. If it is, then it's a bug we will fix.

One way to test it would be to call `mon cpu OnGPIO 0 true` and `mon cpu OnGPIO 0 false` (or the same commands from the Monitor, but without the `mon` part) **after** you set the new PC, but **before** your first `stepi`. If it works - it's probably the WFI. If not, then it proves nothing and needs to be investigated further (as proper interrupts need a cause etc).

xobs, 2020-01-02:
Yes, it does appear as though the actual issue is that it's stuck inside `wfi` -- if I send a key through the serial console it resumes execution.  I've updated the title of the bug to reflect the issue.