renode/renode #53: Machine learning - renode - EFR32

KarimelQ, 2019-12-26:
Hello evone,

i'm trying to use renode on a EFR32 MCU for a machine learning project, apparently renode offers the possibility to do this, unfortunatly there is no documentation about this. 
Did anyone manages to do this with renode/ or have any suggestions about this ?

Many thank's for help!

mgielda, 2019-12-27:
Hello Karim!

Renode comes with quite a bit of documentation, please take a look at Read The Docs - https://renode.readthedocs.io/

We also have a TF lite + Zephyr demo for a RISC-V platform which specifically tests a ML application in Renode: https://github.com/antmicro/litex-vexriscv-tensorflow-lite-demo

So it's hard to say Renode comes without documentation or examples. Specifically we also provide a (selected) list of supported boards: https://renode.readthedocs.io/en/latest/introduction/supported-boards.html that points to a set of precompiled demos you can easily run and then replace the binary with your own.

But to assist you further, you'd need to be more specific as to what you want to achieve - since you also reached out via email, closing here and continuing the discussion there, as this is neither a bug report nor feature request.

closed 2019-12-27 by mgielda

KarimelQ, 2019-12-27:
Hello and thank you for your reply,

by no documentation i meant docs about specific object which is TF for a
silabs's EFR32 micro-controller or micro-controler in general.

Actually, I want to test a ML application in Renode, but for a
microcontroler (cortex M4 - EFR32 for example),  TF lite + Zephyr could be
a solution but there is no example projects about this.

Thank you in advance for your time.

Karim,

-- 
EL QOUNS Abdelkarim
Élève Ingénieur en Electronique - 3ème année -  Option : Systèmes de
réseaux et de télécommunications
Enseirb-Matmeca
1 avenue du Docteur Schweitzer
BP99
33402 Talence
elqounss.karim@gmail.com <laghroubi@gmail.com>
06 33 63 44 21

mgielda, 2019-12-28:
Hi Karim - note that Renode is supposed to run any arbitrary software (just like the hardware platform) on any of the hardware we support, which means endless possibilities. We can't hope to provide examples of everything with everything :)

Anyhow, a ML binary should behave in a similar fashion as any other binary in the general sense - Renode offers a set of functionalities that do not depend on *what* you are running. So please explore the docs to see what you can do. As to an example of how you could run some TF Lite software in Renode and what would be the expected result (and these should not vary in substance between platforms, they way TF Lite is made), you have the VexRiscv example I linked to.

If you don't have the time or capabilities to go through this and combine this knowledge to build your custom setup / use case, we offer commercial support services to do that for you (including embedded systems / edge AI development itself, as we do many other things besides Renode), but you probably already know that since you reached out to us. Let's continue the thread over email.