renode/renode #54: GDB: Line length is short

xobs, 2020-01-02:
It appears as though the line length for GDB commands is rather short -- about 48 characters -- resulting in commands getting cut off:

```
(gdb) mon sysbus LoadELF @../kernel/target/riscv32i-unknown-none-elf/debug/xous-kernel
There was an error executing command 'sysbus LoadELF @../kernel/target/riscv32i-unkno'
Error while loading ELF: Could not find file '../kernel/target/riscv32i-unkno'..
        Could not find file '../kernel/target/riscv32i-unkno'.
(gdb)
```

mateusz-holenko, 2020-01-29:
This is related to a little innocent “feature” we introduced in order to nicely log the content of GDB packets (https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Main/Utilities/GDB/PacketData.cs#L106-L109).

Since `DataAsString` is used not only for logging, but also when parsing all GDB commands - the “feature” breaks handling long packets.

This should be fairly easy to fix - use limited version in logs, but the full one everywhere else.

mateusz-holenko, 2020-02-11:
@xobs - the newest master (https://github.com/renode/renode/commit/3a27c5bc23f5309c94a6f7ea14e70b917c591301) should solve the problem with the truncated long GDB commands.

truhlikfredy, 2020-02-11:
@mateusz-holenko thanks for the fix

xobs, 2020-02-13:
I verify that it is now fixed.  Would you like me to close this issue?

mateusz-holenko, 2020-02-13:
That's great it works. I'm closing this one.

closed 2020-02-13 by mateusz-holenko