renode/renode #55: Feature request: Video Recording

xobs, 2020-01-02:
Renode already has the ability to take screenshots.  It would be very handy for showing demonstrations of hardware if Renode also had the ability to record short videos to `.gif` or `.mp4` format, without needing to resort to screen scrapers such as `gifine` (Linux) or `Xbox Center` (Windows).

truhlikfredy, 2020-01-06:
I'm not Renode developer, but it looks to me as an edge case, re-implementing something which can be achieved very well with existing tools and the existing tools do a very good job.

The simple screen recorder is a pretty good one for Linux:
https://www.maartenbaert.be/simplescreenrecorder/
And win10 is now bundled with the 'gaming' recorder.

mgielda, 2020-01-29:
Hi @xobs, I will have to agree with @truhlikfredy here - we actually have some infrastructure to record videos (see those here - https://renode.io/tutorials/) but it's external to Renode and uses standard tools. We can't do everything in the framework lest we become an operating system ;-) also, since Renode has multiple consoles, many modes of usage, integrates with external tools it has limited control of etc, I would argue that a video functionality would be limited at best, for to really make a good video for a complex case you often need to refer to the filesystem, or a different e.g. Wireshark window etc. I'd rather focus on fixing the bugs you found.

I have it in my backlog to actually release the way we record videos with Renode (on Linux), but this keeps getting pushed back since there are always other thing to do. I assume this would be something helpful for you, right?

xobs, 2020-01-29:
It would!  I was trying to convince a colleague of mine to use Renode, and they came across this issue.  I didn't know if there was an "officially-sanctioned" method of doing it.

I'll let them know that it's not on the roadmap.

mithro, 2020-01-29:
@xobs - Maybe we should leave this open to remind @mgielda to publish his video recording workflow? I'm very interested in seeing that too!

xobs, 2020-01-30:
Sure.  Inquiring minds want to know.