renode/renode #56: Copy feature creates exception on Windows platforms

truhlikfredy, 2020-01-06:
How to replicate:

1. Downloaded windows ZIP package of the renode_1.8.2 release and extracted it.
2. Running it on Windows 7 64-bit OS
3. In the Renode's monitor window, selecting something (for example the "Renode, version 1.8.2.24846 (dc338ac5-201911131318)" text)
4. Right-click and then left-click on the "Copy" context menu will cause an unhandled exception, this will show in the Renode.exe window and in a separate popup as well. Note: The selected content actually gets copied into the clipboard.

```
OpenClipboard Failed (Exception from HRESULT: 0x800401D0 (CLIPBRD_E_CANT_OPEN))
   at System.Runtime.InteropServices.Marshal.ThrowExceptionForHRInternal(Int32 errorCode, IntPtr errorInfo)
   at System.Windows.Clipboard.Flush()
   at System.Windows.Clipboard.CriticalSetDataObject(Object data, Boolean copy)
   at System.Windows.Clipboard.SetDataInternal(String format, Object data)
   at Xwt.WPFBackend.WpfClipboardBackend.SetData(TransferDataType type, Func`1 dataSource)
   at Xwt.Button.OnClicked(EventArgs e)
   at Xwt.Toolkit.Invoke(Action a)
```

PiotrZierhoffer, 2020-01-09:
Does it happen every time you try to copy things? This might be related to some additional software listening to clipboard events, we'll look into it. Apparently, [Windows does that](https://stackoverflow.com/questions/68666/clipbrd-e-cant-open-error-when-setting-the-clipboard-from-net).

truhlikfredy, 2020-01-09:
Yes, in renode, but it doesn't happen in other applications

truhlikfredy, 2020-01-09:
When trying to replicate on different Windows 10 machine, it doesn't happen, yet the Windows 10 has a clipboard monitoring application.

xobs, 2020-02-13:
I get this occasionally too.  Like @truhlikfredy it successfully is copied to the clipboard:

```
Fatal error:
OpenClipboard Failed (Exception from HRESULT: 0x800401D0 (CLIPBRD_E_CANT_OPEN))
   at System.Runtime.InteropServices.Marshal.ThrowExceptionForHRInternal(Int32 errorCode, IntPtr errorInfo)
   at System.Windows.Clipboard.Flush()
   at System.Windows.Clipboard.CriticalSetDataObject(Object data, Boolean copy)
   at System.Windows.Clipboard.SetDataInternal(String format, Object data)
   at Xwt.WPFBackend.WpfClipboardBackend.SetData(TransferDataType type, Func`1 dataSource) in D:\Code\Renode\renode\lib\xwt\Xwt.WPF\Xwt.WPFBackend\ClipboardBackend.cs:line 52
   at Xwt.Clipboard.SetText(String text) in D:\Code\Renode\renode\lib\xwt\Xwt\Xwt\Clipboard.cs:line 127
   at Antmicro.Renode.UI.TerminalWidget.CopyMarkedField() in D:\Code\Renode\renode\src\Renode\UI\TerminalActions.cs:line 17
   at Antmicro.Renode.UI.TerminalWidget.<>c__DisplayClass11_0.<CreatePopupMenu>b__0(Object <p0>, EventArgs <p1>) in D:\Code\Renode\renode\src\Renode\UI\TerminalWidget.cs:line 174
   at Xwt.MenuItem.OnClicked(EventArgs e) in D:\Code\Renode\renode\lib\xwt\Xwt\Xwt\MenuItem.cs:line 191
   at Xwt.MenuItem.DoClick() in D:\Code\Renode\renode\lib\xwt\Xwt\Xwt\MenuItem.cs:line 185
   at Xwt.MenuItem.MenuItemBackendHost.OnClicked() in D:\Code\Renode\renode\lib\xwt\Xwt\Xwt\MenuItem.cs:line 53
   at Xwt.Toolkit.Invoke(Action a) in D:\Code\Renode\renode\lib\xwt\Xwt\Xwt\Toolkit.cs:line 350
```

truhlikfredy, 2020-02-19:
I get the content to the clipboard as well, should maybe that exception be silenced? I experienced some application where clipboard was not working in the past, but none of them crashed with exception, so I think it would be ok here to silently ignore the exception.

PiotrZierhoffer, 2020-02-19:
It's not very easy to catch it, unfortunately. And the biggest problem is that we never managed to actually replicate it. I hope to be able to look into it more after the Embedded World.

truhlikfredy, 2020-02-26:
Would making and sharing a VM where it can happen help? But despite replicating it there is the bigger picture, there is code which is handing the clipboard and is not handled by try/catch so any exception will throw unhandled exception, is it intended to throw unhandled exceptions when for any reason the clipboard copy will fail? I'm not saying there needs to be 'catch' and fallback implementeion or that the clipboard has to work on all odd machines, but throwing unhandled exceptions is bit atypical (and on top the clipboard works anyway)

PiotrZierhoffer, 2020-02-26:
If you can prepare a VM that actually replicates that, it would be great - we were never able to replicate it. And that's why this got a bit stale.

PiotrZierhoffer, 2020-03-09:
Thanks to @truhlikfredy we managed to replicate it.

As the API is really nonsensical, our approach is to silence the exception. This might not be 100% successful in copying, but, apparently, there is no fully certain way of doing this. We could repeat the attempt to copy stuff X times, but then it could fail just as well.

Our tests showed that indeed the data is copied whether we catch the exception or not.

@xobs could you please verify if it works for you as well? It will be on master later today

mateusz-holenko, 2021-05-05:
I guess we can close this one as the exception catching is on master for some time now and there were no more reports recently.

closed 2021-05-05 by mateusz-holenko