renode/renode #57: GDB: Server occasionally locks up

xobs, 2020-01-07:
Similar to #52, the gdb server sometimes locks up.  However, unlike #52 this is not related to the `wfi` instruction, as typing into the terminal does not unstick the server.

Steps to reproduce:
1. Download the attached `xous-kernel.zip`.  This is not a zipfile, but is an elf binary.
2. Run `renode -e '$bin=@xous-kernel.zip; i @scripts/single-node/litex_vexriscv.resc; machine StartGdbServer 3333'`
3. Run: `riscv64-unknown-elf-gdb xous-kernel.zip -ex 'tar rem :3333' -ex 'b xous_kernel::uart::Uart::putc' -ex 'mon start' -ex 'c' -ex 'finish'`

Renode will start to print out a lot of information on the terminal:

```
11:52:41.1740 [DEBUG] cpu: Added hook @ 0x20505080
11:52:41.1823 [DEBUG] cpu: Executing hooks registered at address 0x20505080
11:52:42.6393 [DEBUG] cpu: Added hook @ 0x20505088
11:52:42.6403 [DEBUG] cpu: Executing hooks registered at address 0x20505088
11:52:45.9932 [DEBUG] cpu: Added hook @ 0x20505080
11:52:45.9942 [DEBUG] cpu: Executing hooks registered at address 0x20505080
11:52:46.0332 [DEBUG] cpu: Added hook @ 0x20505088
11:52:46.1275 [DEBUG] cpu: Executing hooks registered at address 0x20505088
11:52:47.9990 [DEBUG] cpu: Added hook @ 0x20505080
11:52:47.9995 [DEBUG] cpu: Executing hooks registered at address 0x20505080
11:52:48.0141 [DEBUG] cpu: Added hook @ 0x20505088
```

The gdb server will now be frozen:

```
❯ riscv64-unknown-elf-gdb .\target\riscv32i-unknown-none-elf\debug\xous-kernel -ex 'tar rem :3333' -ex 'b xous_kernel::uart::Uart::putc' -ex 'mon start' -ex 'c' -ex 'finish'
GNU gdb (SiFive GDB 8.3.0-2019.08.0) 8.3
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-w64-mingw32 --target=riscv64-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://github.com/sifive/freedom-tools/issues>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from .\target\riscv32i-unknown-none-elf\debug\xous-kernel...
Remote debugging using :3333
_start () at asm.S:27
27      asm.S: No such file or directory.
Breakpoint 1 at 0x20505064: file src/uart.rs, line 11.
Starting emulation...
Continuing.

Breakpoint 1, xous_kernel::uart::Uart::putc (self=0x0, c=0) at src/uart.rs:11
11              let ptr = 0xE000_1800 as *mut u32;
Run till exit from #0  xous_kernel::uart::Uart::putc (self=0x0, c=0) at src/uart.rs:11
The target is not responding to GDB commands.
Stop debugging it? (y or n)
```

Running `cpu PC` shows that the program counter is still running.

Elf binary:
[xous-kernel.zip](https://github.com/renode/renode/files/4028812/xous-kernel.zip)

xobs, 2020-01-07:
I suspect the issue is that it's in an infinite loop:

```rust
    pub fn putc(&self, c: u8) {
        let ptr = 0xE000_1800 as *mut u32;
        unsafe {
            loop {
                if ptr.add(1).read_volatile() != 0 {
                    ptr.add(0).write_volatile(c as u32);
                    return;
                }
            }
        }
    }
```

And so the breakpoint is never actually getting hit.  However, `^C` should always break into the debugger.

xobs, 2020-01-07:
Interestingly, I'm trying to debug this on my system and when I do the above, an "Assertion Failed" box pops up:

---------------------------
Assertion Failed: Abort=Quit, Retry=Debug, Ignore=Continue
---------------------------
This handle should be blocking

   at Antmicro.Renode.Time.TimeHandle.UnblockHandle() in D:\Code\Renode\renode\src\Infrastructure\src\Emulator\Main\Time\TimeHandle.cs:line 175

   at Antmicro.Renode.Time.TimeSourceBase.ExecuteUnblockPhase(LinkedListNode`1 handle) in D:\Code\Renode\renode\src\Infrastructure\src\Emulator\Main\Time\TimeSourceBase.cs:line 546

   at Antmicro.Renode.Time.TimeSourceBase.PhaseExecutor`1.ExecuteInParallel(IEnumerable`1 targets) in D:\Code\Renode\renode\src\Infrastructure\src\Emulator\Main\Time\TimeSourceBase.cs:line 833

   at Antmicro.Renode.Time.TimeSourceBase.InnerExecute(TimeInterval& virtualTimeElapsed, Nullable`1 timeLimit) in D:\Code\Renode\renode\src\Infrastructure\src\Emulator\Main\Time\TimeSourceBase.cs:line 406

   at Antmicro.Renode.Time.SlaveTimeSource.DispatchInner() in D:\Code\Renode\renode\src\Infrastructure\src\Emulator\Main\Time\SlaveTimeSource.cs:line 237

   at Antmicro.Renode.Time.SlaveTimeSource.Dispatch() in D:\Code\Renode\renode\src\Infrastructure\src\Emulator\Main\Time\SlaveTimeSource.cs:line 168

   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)

   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)

   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)

   at System.Threading.ThreadHelper.ThreadStart()

---------------------------
Abort   Retry   Ignore   
---------------------------

PiotrZierhoffer, 2020-01-15:
It seems that the error you're getting is not the correct one - it only hides the true cause of the issue. So we need to get rid of the invalid error message, and then work on the actual issue you're observing

xobs, 2020-01-15:
Do you get the same behavior on your end?  I must run with asserts enabled.

mateusz-holenko, 2020-01-29:
Is the `putc` function implementation correct in the first place? In the loop you read `ptr + 1` (`0xE000_1800` - liteuart base address + `0x4` == `TxFull` register) and only if it’s not `0` (it means tx is full, right?) you put a byte to the tx queue. Shouldn’t it be the other way around - you don’t put anything into the queue as long as it’s full?

In Renode `TxFull` is always `0`, so you will never see anything printed on UART. 
BTW, I forced `TxFull` to return `1` for testing purposes and noticed the binary printed “Xous Starting Up” on UART.

Anyway GDB should not block - I was able to reproduce the original problem, but didn’t get the assertion problem. I’ll look deeper into it.

xobs, 2020-01-30:
> Is the `putc` function implementation correct in the first place? In the loop you read `ptr + 1` (`0xE000_1800` - liteuart base address + `0x4` == `TxFull` register) and only if it’s not `0` (it [...]

You're right, my mistake.  Too many double-negatives.  The Rust code is now correct.

> In Renode `TxFull` is always `0`, so you will never see anything printed on UART.
> BTW, I forced `TxFull` to return `1` for testing purposes and noticed the binary printed “Xous Starting Up” on [...]

I only got the assertion when running the Debug build.  I'm also running under Windows, in case that matters.

mateusz-holenko, 2020-02-03:
@xobs - https://github.com/renode/renode/commit/3b4800130dd1e3c4061f7bc1fb56dc7c555c0a17 should fix the hidden exception problem - could you run the test again on the newest Renode?

xobs, 2020-02-04:
I'm running a1ab5323b3dc7034050f0a55493fcf686d8815ed (with src/Infrastructure at c1616066167899ed85f8d75f2343706186c8d586) and I still get the error:

```
---------------------------
Assertion Failed: Abort=Quit, Retry=Debug, Ignore=Continue
---------------------------
This handle should be blocking

   at Antmicro.Renode.Time.TimeHandle.UnblockHandle() in D:\Code\Renode\renode\src\Infrastructure\src\Emulator\Main\Time\TimeHandle.cs:line 175

   at Antmicro.Renode.Time.TimeSourceBase.ExecuteUnblockPhase(LinkedListNode`1 handle) in D:\Code\Renode\renode\src\Infrastructure\src\Emulator\Main\Time\TimeSourceBase.cs:line 546

   at Antmicro.Renode.Time.TimeSourceBase.PhaseExecutor`1.ExecuteInParallel(IEnumerable`1 targets) in D:\Code\Renode\renode\src\Infrastructure\src\Emulator\Main\Time\TimeSourceBase.cs:line 833

   at Antmicro.Renode.Time.TimeSourceBase.InnerExecute(TimeInterval& virtualTimeElapsed, Nullable`1 timeLimit) in D:\Code\Renode\renode\src\Infrastructure\src\Emulator\Main\Time\TimeSourceBase.cs:line 406

   at Antmicro.Renode.Time.SlaveTimeSource.DispatchInner() in D:\Code\Renode\renode\src\Infrastructure\src\Emulator\Main\Time\SlaveTimeSource.cs:line 237

   at Antmicro.Renode.Time.SlaveTimeSource.Dispatch() in D:\Code\Renode\renode\src\Infrastructure\src\Emulator\Main\Time\SlaveTimeSource.cs:line 168

   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)

   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)

   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)

   at System.Threading.ThreadHelper.ThreadStart()

---------------------------
Abort   Retry   Ignore   
---------------------------
```

xobs, 2020-02-05:
It definitely feels like a race condition.

It happens when issuing a `stepi` (gdb `s`) instruction.  There's about a 20% chance of hitting the exception.  It happens randomly and regardless of what instruction is being executed.

For example, the first instruction in my program is `j`, which sometimes triggers the exception.  Then that is followed by a series of stores into offsets of `$sp`, all of which are basically identical opcodes.  Sometimes these will trigger the exception, sometimes not.

Additionally, I've had it throw an exception on instructions such as `addi`, which don't touch memory at all, indicating it's not a problem with memory access but is a problem with the CPU itself.

How can I troubleshoot this issue?

xobs, 2020-03-10:
Speaking of v1.9.0, any chance of fixing https://github.com/renode/renode/issues/57? It makes it really unusable.

I've been running this patch for quite some time that prevents the issue.  Maybe it only shows up in debug builds, though?

```patch
diff --git a/src/Emulator/Main/Time/TimeHandle.cs b/src/Emulator/Main/Time/TimeHandle.cs
index ec0b4e1f..6822891f 100644
--- a/src/Emulator/Main/Time/TimeHandle.cs
+++ b/src/Emulator/Main/Time/TimeHandle.cs
@@ -172,7 +172,7 @@ namespace Antmicro.Renode.Time
             this.Trace();
             lock(innerLock)
             {
-                Debug.Assert(isBlocking, "This handle should be blocking");
+                // Debug.Assert(isBlocking, "This handle should be blocking");

                 if(!waitsToBeUnblocked)
                 {
```

If asserts get compiled out in release mode, then this issue can stay unfixed with 1.9.0, as nobody else will have this issue.

PiotrZierhoffer, 2020-03-10:
This is true - assertions are not included in Release builds.

We will try to reanalyze it in the following week, possibly asking you to include some patches with debug logs here and there, if we're not able to reproduce this.