renode/renode #58: Windows: Renode sometimes starts offscreen

xobs, 2020-01-09:
The Renode binary will occasionally start with its menubar offscreen.  Resizing the window fixes this issue, however it shouldn't get into this state in the first place.

This is a fresh install, on a machine with a single display.  This happens every time I start Renode now, though I don't know how common it is on other platforms.

[image]

mateusz-holenko, 2020-01-29:
I confirm we’ve noticed this problem locally too.

In Renode there is a code for automatic positioning of new windows so that they don’t hide one another: https://github.com/renode/renode-infrastructure/blob/master/src/UI/ConsoleBackendAnalyzers/WindowPositionProvider.cs#L38. 

The problem might be in the initial values calculated on Windows platform - I’ll need to verify this.

PiotrZierhoffer, 2020-01-29:
@xobs can you please verify if you have any screen scaling enabled?

xobs, 2020-01-30:
It's a "hidpi" screen, which has scaling set to anywhere between 150-200%.

I have verified that it happens whenever "Scaling" is set to anything other than "100%".  At "100%" the window goes in the far upper-right corner, so it makes sense that anything larger would force it offscreen.

xobs, 2020-02-17:
I've done some investigation, and it is maddeningly random.

For example, if I apply the following patch, it works just fine regardless of resolution:

```patch
diff --git a/src/Renode/UI/TermsharpProvider.cs b/src/Renode/UI/TermsharpProvider.cs
index cc696f6..490d873 100644
--- a/src/Renode/UI/TermsharpProvider.cs
+++ b/src/Renode/UI/TermsharpProvider.cs
@@ -33,11 +33,11 @@ namespace Antmicro.Renode.UI
                 window.Title = consoleName == null ? "Renode" : consoleName;
                 window.Width = 700;
                 window.Height = 400;
-                window.Location = WindowPositionProvider.Instance.GetNextPosition();
                 window.Padding = new WidgetSpacing();
                 window.Content = terminalWidget;
                 terminalWidget.Initialized += mre.Set;
                 window.Show();
+                window.Location = WindowPositionProvider.Instance.GetNextPosition();
                 window.Closed += (sender, e) =>
                 {
                     InnerOnClose();
```

This causes the window to appear at a random location initially, but then to move to the correct location.  The final location is correct regardless of the HiDPI settings.

It may be a bug in the Xwt.WPF backend?

ajelinski, 2020-03-05:
The commit targeting this issue has just been pushed to master: https://github.com/renode/renode/commit/0b59bf591f2a58c229fd1ad12595234532ee7197.

Please check if such change solves the problem in your case.

PiotrZierhoffer, 2020-03-05:
@xobs FYI we set the position twice. It will not solve the initial blink on the scaled windows displays, but will work fine in every other case.

I believe it's more of a "unsupported scenario" than a bug - some things in Xwt.WPF seem to be calculated just after the window is shown. Maybe we'll look into the details someday, but for now I think this is a good-enough solution

xobs, 2020-03-07:
Thanks, the approach looks reasonable.  I'll test it on my hidpi device.

xobs, 2020-03-07:
I have verified the fix works.  Thanks!

closed 2020-03-07 by xobs