renode/renode #59: RISC-V: Missing SATP register

xobs, 2020-01-21:
The `satp` register is missing from the Renode interface.  It is not possible to print this register via the Renode console, unlike other registers:

```
(machine-0) cpu SATP
There was an error executing command 'cpu SATP'
sysbus.cpu does not provide a field, method or property SATP.
(machine-0) cpu MTVEC
0x20500110
(machine-0) cpu STVEC
0x0
(machine-0) 
```

This CPU is configured with `privilegeArchitecture: PrivilegeArchitecture.Priv1_10`, so it ought to have this register.

Note that because of #50, it is also not possible to read this register via gdb.

mateusz-holenko, 2020-01-29:
SATP is implemented internally in Renode, but not visible publicly to the user. It shouldn’t be too hard to expose it (both in monitor and in gdb).

mateusz-holenko, 2020-02-04:
https://github.com/renode/renode-infrastructure/commit/61e0bcdd5755b894dd1b2b51365cce76808a45da exposes `SATP` and `SPTBR` registers in monitor, so now `(machine-0) cpu SATP` should work.

Both registers share internally the same `id`, so at the monitor level they are interchangeable:
* for 1.09 privilege architecture CPU both point to `SPTBR`
* for 1.10 privilege architecture CPU both point to `SATP`

xobs, 2020-02-04:
Perfect, l've verified that this works as intended.  Thanks for the fix.

closed 2020-02-04 by xobs