renode/renode #60: RISC-V: Emulator crashes when accessing invalid memory

xobs, 2020-01-21:
I'm currently trying to get an MMU working, and so I'm doing some strange things in the emulator.

The attached ELF (renamed to .zip) file crashes Renode 1.8.2 completely as soon as it attempts to issue `mret`:
[xous-kernel.zip](https://github.com/renode/renode/files/4090489/xous-kernel.zip)

```
$ renode .\boot.resc
18:14:45.5287 [INFO] Loaded monitor commands from: C:\Program Files\Renode\./scripts/monitor.py
18:14:46.3227 [INFO] Including script: C:\Users\smcro\Code\Xous\platonic\boot.resc
18:14:46.3767 [INFO] System bus created.
18:14:47.3617 [DEBUG] Segment size automatically calculated to value 64KiB
18:14:47.3627 [DEBUG] Segment size automatically calculated to value 1MiB
18:14:47.3737 [DEBUG] Segment size automatically calculated to value 4MiB
18:14:47.6667 [DEBUG] sysbus: Loading ELF C:\Users\smcro\Code\Xous\kernel\target\riscv32i-unknown-none-elf\debug\xous-kernel.
18:14:47.7057 [INFO] sysbus: Loading segment of 69192 bytes length at 0x20500000.
18:14:47.7267 [DEBUG] sysbus: Segment loaded.
18:14:47.7537 [WARNING] Could not load demangling library. Symbols will not be demangled.
18:14:47.7867 [INFO] cpu: Setting PC value to 0x20500000.
18:14:47.8417 [INFO] machine-0: GDB server with all CPUs started on port :3333
18:14:47.8977 [DEBUG] sysbus: Loading ELF C:\Users\smcro\Code\Xous\kernel\target\riscv32i-unknown-none-elf\debug\xous-kernel.
18:14:47.9217 [INFO] sysbus: Loading segment of 69192 bytes length at 0x20500000.
18:14:47.9217 [DEBUG] sysbus: Segment loaded.
18:14:48.2607 [INFO] cpu: Setting PC value to 0x20500000.
18:14:48.2767 [INFO] machine-0: Machine started.

Unhandled Exception: System.AccessViolationException: Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
   at Antmicro.Renode.Peripherals.CPU.TranslationCPU.ExecuteInstructions(UInt64 numberOfInstructionsToExecute, UInt64& numberOfExecutedInstructions)
   at Antmicro.Renode.Peripherals.CPU.TranslationCPU.CpuThreadBodyInner(Boolean singleStep)
   at Antmicro.Renode.Peripherals.CPU.TranslationCPU.CpuThreadBody()
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)    at System.Threading.ThreadHelper.ThreadStart()
$
```

This seems like it could be a security issue, since it's trying to access random memory and the entire emulator is crashing.

mateusz-holenko, 2020-01-29:
I ran the attached binary several times, but couldn’t reproduce the crash (on Linux):
```
./renode.sh -e '$bin=@/tmp/xous-kernel.zip; s @scripts/single-node/litex_vexriscv.resc'
```
The output on UART:

```
Starting up...
Creating memory manager...
Created Memory Manager: MemoryManager: 
    flash: 20000000 .. 21000000 (4096)
    ram:   40000000 .. 41000000 (4096)
    io:    e0000000 .. e0010000 (65536)
    lcd:   b0000000 .. b0008000 (8)
Creating process table...
SATP value: 00600020  Root page: 40004000  pt: 0x40fc79e8  pt: 0x40004000
Mapping phys: 40000000 -> virt: 40000000  (vpn1: 0100  vpn0: 0000    ppn1: 0400  ppn0: 0000)
l1_pt is at 0x40fc7d50  (0x40004400)
    top-level VPN1 0100: 00000000 (will allocate a new one)
    Allocated new top-level page for VPN1 0100 in process @ 40005000
    New top-level page entry: 40005001
Entries mapped: > 256:40005001 < (1024)
Mapping phys: 40000084 -> virt: 40000084  (vpn1: 0100  vpn0: 0000    ppn1: 0400  ppn0: 0000)
l1_pt is at 0x40fc7d50  (0x40004400)
Entries mapped: > 256:40005001 < (1024)
Mapping phys: 40001084 -> virt: 40001084  (vpn1: 0100  vpn0: 0004    ppn1: 0400  ppn0: 0004)
l1_pt is at 0x40fc7d50  (0x40004400)
Entries mapped: > 256:40005001 < (1024)
Mapping phys: 40002084 -> virt: 40002084  (vpn1: 0100  vpn0: 0008    ppn1: 0400  ppn0: 0008)
l1_pt is at 0x40fc7d50  (0x40004400)
Entries mapped: > 256:40005001 < (1024)
Mapping phys: 40003084 -> virt: 40003084  (vpn1: 0100  vpn0: 000c    ppn1: 0400  ppn0: 000c)
l1_pt is at 0x40fc7d50  (0x40004400)
Entries mapped: > 256:40005001 < (1024)
Mapping phys: 20500000 -> virt: 20500000  (vpn1: 0081  vpn0: 0000    ppn1: 0205  ppn0: 0000)
l1_pt is at 0x40fc7d50  (0x40004204)
    top-level VPN1 0081: 00000000 (will allocate a new one)
    Allocated new top-level page for VPN1 0081 in process @ 40006000
    New top-level page entry: 40006001
Entries mapped: > 129:40006001 256:40005001 < (1024)
Mapping phys: 20501000 -> virt: 20501000  (vpn1: 0081  vpn0: 0004    ppn1: 0205  ppn0: 0004)
l1_pt is at 0x40fc7d50  (0x40004204)
Entries mapped: > 129:40006001 256:40005001 < (1024)
Mapping phys: 20502000 -> virt: 20502000  (vpn1: 0081  vpn0: 0008    ppn1: 0205  ppn0: 0008)
l1_pt is at 0x40fc7d50  (0x40004204)
Entries mapped: > 129:40006001 256:40005001 < (1024)
Mapping phys: 20503000 -> virt: 20503000  (vpn1: 0081  vpn0: 000c    ppn1: 0205  ppn0: 000c)
l1_pt is at 0x40fc7d50  (0x40004204)
Entries mapped: > 129:40006001 256:40005001 < (1024)
Mapping phys: 20504000 -> virt: 20504000  (vpn1: 0081  vpn0: 0010    ppn1: 0205  ppn0: 0010)
l1_pt is at 0x40fc7d50  (0x40004204)
Entries mapped: > 129:40006001 256:40005001 < (1024)
Mapping phys: 20505000 -> virt: 20505000  (vpn1: 0081  vpn0: 0014    ppn1: 0205  ppn0: 0014)
l1_pt is at 0x40fc7d50  (0x40004204)
Entries mapped: > 129:40006001 256:40005001 < (1024)
Mapping phys: 20506000 -> virt: 20506000  (vpn1: 0081  vpn0: 0018    ppn1: 0205  ppn0: 0018)
l1_pt is at 0x40fc7d50  (0x40004204)
Entries mapped: > 129:40006001 256:40005001 < (1024)
Mapping phys: 20507000 -> virt: 20507000  (vpn1: 0081  vpn0: 001c    ppn1: 0205  ppn0: 001c)
l1_pt is at 0x40fc7d50  (0x40004204)
Entries mapped: > 129:40006001 256:40005001 < (1024)
Mapping phys: 20508000 -> virt: 20508000  (vpn1: 0081  vpn0: 0020    ppn1: 0205  ppn0: 0020)
l1_pt is at 0x40fc7d50  (0x40004204)
Entries mapped: > 129:40006001 256:40005001 < (1024)
Mapping phys: 20509000 -> virt: 20509000  (vpn1: 0081  vpn0: 0024    ppn1: 0205  ppn0: 0024)
l1_pt is at 0x40fc7d50  (0x40004204)
Entries mapped: > 129:40006001 256:40005001 < (1024)
Mapping phys: 2050a000 -> virt: 2050a000  (vpn1: 0081  vpn0: 0028    ppn1: 0205  ppn0: 0028)
l1_pt is at 0x40fc7d50  (0x40004204)
Entries mapped: > 129:40006001 256:40005001 < (1024)
Mapping phys: 2050b000 -> virt: 2050b000  (vpn1: 0081  vpn0: 002c    ppn1: 0205  ppn0: 002c)
l1_pt is at 0x40fc7d50  (0x40004204)
Entries mapped: > 129:40006001 256:40005001 < (1024)
Mapping phys: 2050c000 -> virt: 2050c000  (vpn1: 0081  vpn0: 0030    ppn1: 0205  ppn0: 0030)
l1_pt is at 0x40fc7d50  (0x40004204)
Entries mapped: > 129:40006001 256:40005001 < (1024)
Mapping phys: 2050d000 -> virt: 2050d000  (vpn1: 0081  vpn0: 0034    ppn1: 0205  ppn0: 0034)
l1_pt is at 0x40fc7d50  (0x40004204)
Entries mapped: > 129:40006001 256:40005001 < (1024)
Mapping phys: 2050e000 -> virt: 2050e000  (vpn1: 0081  vpn0: 0038    ppn1: 0205  ppn0: 0038)
l1_pt is at 0x40fc7d50  (0x40004204)
Entries mapped: > 129:40006001 256:40005001 < (1024)
Mapping phys: 2050f000 -> virt: 2050f000  (vpn1: 0081  vpn0: 003c    ppn1: 0205  ppn0: 003c)
l1_pt is at 0x40fc7d50  (0x40004204)
Entries mapped: > 129:40006001 256:40005001 < (1024)
Mapping phys: 20510000 -> virt: 20510000  (vpn1: 0081  vpn0: 0040    ppn1: 0205  ppn0: 0040)
l1_pt is at 0x40fc7d50  (0x40004204)
Entries mapped: > 129:40006001 256:40005001 < (1024)
PID 1: Process (satp: 0x00600020, mode: 0, ASID: 1, PPN: 40004000) root page @ 40004000
Enabling MMU...
MMU enabled, jumping to kmain
Entering main loop
```

Is it what you expect to see?

I guess you are working on Windows - I’ll compare the behaviour there.

xobs, 2020-01-30:
Perhaps you're right and it's a problem with .NET on Windows.

I do know for a fact that the manipulation of `$satp` is very, very, extremely broken in that binary, so it's trying to translate a random address.  It looks like it's escaping the VM.

PiotrZierhoffer, 2020-02-12:
I managed to reproduce the issue on Windows.

We will look into it

xobs, 2020-02-21:
It definitely has to do with giving the emulator an invalid $satp value.  I haven't managed to get a smaller reproduction case, but I _have_ hit the bug several times already when I put incorrect values into $satp.

Crashing Renode is a very quick way to indicate that I've done something wrong...