renode/renode #62: RISC-V: CSR accesses are unchecked

xobs, 2020-01-24:
Renode currently allows any CSR to be read or written regardless of the privilege level.  This means that it's possible for any User mode process to simply rewrite its SATP register and give itself full write access to the entire system.

Renode should issue an Exception Code of "2", which should generate a machine mode trap.  This appears to be how VexRiscv does it: https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/CsrPlugin.scala#L1040

# Steps to Reproduce

Assemble the following program:

```asm
.global _start
.text
_start:
        csrw satp, zero
loop:
        j loop
```

This can be accomplished with the following command, which also generates base64 output:

```sh
$ riscv64-unknown-elf-gcc -mabi=ilp32 -march=rv32ima -o test test.S -nostdlib -nostdinc
$ base64 < test
riscv64-unknown-elf-gcc -mabi=ilp32 -march=rv32ima -o test test.S -nostdlib -nostdinc; base64 < test
f0VMRgEBAQAAAAAAAAAAAAIA8wABAAAAVAABADQAAADwAQAAAAAAADQAIAABACgABgAFAAEAAAAA
AAAAAAABAAAAAQBcAAAAXAAAAAUAAAAAEAAAcxAAGG8AAABBIwAAAHJpc2N2AAEZAAAABXJ2MzJp
MnAwX20ycDBfYTJwMAAAAAAAAAAAAAAAAAAAAAAAAAAAAFQAAQAAAAAAAwABAAAAAAAAAAAAAAAA
AAMAAgABAAAAAAAAAAAAAAAEAPH/EQAAAFgAAQAAAAAAAAABABYAAABcGAEAAAAAABAA8f8oAAAA
XBABAAAAAAAQAAEASQAAAFQAAQAAAAAAEAABADgAAABcEAEAAAAAABAAAQBEAAAAXBABAAAAAAAQ
AAEAUAAAAFwQAQAAAAAAEAABAF8AAABcEAEAAAAAABAAAQBmAAAAXBABAAAAAAAQAAEAAC90bXAv
Y2Nxb1YzQ2gubwBsb29wAF9fZ2xvYmFsX3BvaW50ZXIkAF9fU0RBVEFfQkVHSU5fXwBfX0JTU19F
TkRfXwBfX2Jzc19zdGFydABfX0RBVEFfQkVHSU5fXwBfZWRhdGEAX2VuZAAALnN5bXRhYgAuc3Ry
dGFiAC5zaHN0cnRhYgAudGV4dAAucmlzY3YuYXR0cmlidXRlcwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAbAAAAAQAAAAYAAABUAAEAVAAAAAgAAAAAAAAAAAAAAAQA
AAAAAAAAIQAAAAMAAHAAAAAAAAAAAFwAAAAkAAAAAAAAAAAAAAABAAAAAAAAAAEAAAACAAAAAAAA
AAAAAACAAAAA0AAAAAQAAAAFAAAABAAAABAAAAAJAAAAAwAAAAAAAAAAAAAAUAEAAGsAAAAAAAAA
AAAAAAEAAAAAAAAAEQAAAAMAAAAAAAAAAAAAALsBAAAzAAAAAAAAAAAAAAABAAAAAAAAAA==
$
```

Then, start `litex_vexriscv_linux.resc` in Renode:

```
(monitor) i @scripts/single-node/litex_vexriscv_linux.resc 
(machine-0) start
Starting emulation...
(machine-0)
```

Finally, log in and paste the output.  Note that due to UART buffering issues, the output will be incorrect.  However, you can verify the md5sum to ensure that the binary is correct.

```sh
root@buildroot:~# cat > test.b64
f0VMRgEBAQAAAAAAAAAAAAIA8wABAAAAVAABADQAAADwAQAAAAAAADQAIAABACgABgAFAAEAAAAA
AAAAAAABAAAAAQBcAAAAXAAAAAUAAAAAEAAAcxAAGG8AAABBIwAAAHJpc2N2AAEZAAAABXJ2MzJp
MnAwX20ycDBfYTJwMAAAAAAAAAAAAAAAAAAAAAAAAAAAAFQAAQAAAAAAAwABAAAAAAAAAAAAAAAA
AAMAAgABAAAAAAAAAAAAAAAEAPH/EQAAAFgAAQAAAAAAAAABABYAAABcGAEAAAAAABAA8f8oAAAA
XBABAAAAAAAQAAEASQAAAFQAAQAAAAAAEAABADgAAABcEAEAAAAAABAAAQBEAAAAXBABAAAAAAAQ
AAEAUAAAAFwQAQAAAAAAEAABAF8AAABcEAEAAAAAABAAAQBmAAAAXBABAAAAAAAQAAEAAC90bXAv
Y2Nxb1YzQ2gubwBsb29wAF9fZ2xvYmFsX3BvaW50ZXIkAF9fU0RBVEFfQkVHSU5fXwBfX0JTU19F
TkRfXwBfX2Jzc19zdGFydABfX0RBVEFfQkVHSU5fXwBfZWRhdGEAX2VuZAAALnN5bXRhYgAuc3Ry
dGFiAC5zaHN0cnRhYgAudGV4dAAucmlzY3YuYXR0cmlidXRlcwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAbAAAAAQAAAAYAAABUAAEAVAAAAAgAAAAAAAAAAAAAAAQA
AAAAAAAAIQAAAAMAAHAAAAAAAAAAAFwAAAAkAAAAAAAAAAAAAAABAAAAAAAAAAEAAAACAAAAAAAA
AAAAAACAAAAA0AAAAAQAAAAFAAAABAAAABAAAAAJAAAAAwAAAAAAAAAAAAAAUAEAAGsAAAAAAAAA
AAAAAAEAAAAAAAAAEQAAAAMAAAAAAAAAAAAAALsBAAAzAAAAAAAAAAAAAAABAAAAAAAAAA==
^D
root@buildroot:~# base64 -d test.b64 > test
root@buildroot:~# md5sum test
54f03da8a882a9d37dd1b361a56fd1b7  test
root@buildroot:~# chmod a+x test
root@buildroot:~# ./test
```

At this point, the system will be frozen.

xobs, 2020-01-24:
Note that I will be attempting to verify that this works under a real vexriscv in simulation, to ensure that this isn't just me misreading the specs.

xobs, 2020-01-24:
If I run some code under verillator, it exits, although this is not conclusive since apparently the default handler jumps into the emulator on an illegal instruction and runs:

```
static void litex_stop(void){
#ifdef CSR_SUPERVISOR_FINISH_ADDR
    supervisor_finish_write(1);
#endif
	while(1);
}
```

However, the Renode CPU PRIV is `0x0`, indicating it's still in User mode, and the PC jumped to `0x10058` which is unmapped memory.

xobs, 2020-01-24:
More telling -- I have a different program that actually has a handler for illegal instructions.

Under verilator, it prints the following:

```
MMU enabled, jumping to kmain
SATP: 80040017
loader: MSTATUS: Mstatus { bits: 264 }
KMAIN: In User mode
CPU Exception: Illegal instruction 0x18002573 at 0x2050020c
```

Under Renode, the same thing prints out:

```
MMU enabled, jumping to kmain
SATP: 80040017
loader: MSTATUS: Mstatus { bits: 8 }
KMAIN: In User mode
kmain: SATP: 80040017
kmain: MSTATUS: Mstatus { bits: 0 }
Entering main loop
Attempting to disable the MMU (80040017):
Done!  Now: 00000000
```

The actual code that prints this is:

```rust
    sprintln!("KMAIN: In User mode");
    sprintln!("kmain: SATP: {:08x}", satp::read().bits());
    sprintln!("kmain: MSTATUS: {:?}", mstatus::read());
    sprintln!("Entering main loop");
    sprintln!("Attempting to disable the MMU ({:08x}):", satp::read().bits());
    satp::write(0);
    println!("Done!  Now: {:08x}", satp::read().bits());
```

In vexriscv under verilator, even attempting to access satp is met with an illegal argument exception, whereas with Renode it's permitted.

xobs, 2020-01-24:
Another difference: If we try to read from `satp`, it should cause an illegal instruction exception from U mode.  Changing the opcode to `csrr`, we get the following program:

```asm
.global _start
.text
_start:
        csrr a0, satp
loop:
        j loop
```

Compile this, strip it, and print out the base64:

```sh
$ riscv64-unknown-elf-gcc -mabi=ilp32 -march=rv32ima -o test test.S -nostdlib -nostdinc
$ riscv64-unknown-elf-strip test
$ base64 test
f0VMRgEBAQAAAAAAAAAAAAIA8wABAAAAVAABADQAAACkAAAAAAAAADQAIAABACgABAADAAEAAAAA
AAAAAAABAAAAAQBcAAAAXAAAAAUAAAAAEAAAcyUAGG8AAABBIwAAAHJpc2N2AAEZAAAABXJ2MzJp
MnAwX20ycDBfYTJwMAAALnNoc3RydGFiAC50ZXh0AC5yaXNjdi5hdHRyaWJ1dGVzAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACwAAAAEAAAAGAAAAVAABAFQAAAAIAAAA
AAAAAAAAAAAEAAAAAAAAABEAAAADAABwAAAAAAAAAABcAAAAJAAAAAAAAAAAAAAAAQAAAAAAAAAB
AAAAAwAAAAAAAAAAAAAAgAAAACMAAAAAAAAAAAAAAAEAAAAAAAAA
$
```

Then do the same dance to load it on both Renode and verilator.

If you run the program under Renode, you can quit the program with `^C` -- the system doesn't hang.:

```sh
root@buildroot:~# ./test
^C
root@buildroot:~# 
```

However, under Verilator, the following happens:

```sh
root@buildroot:~# cat > test.b64
f0VMRgEBAQAAAAAAAAAAAAIA8wABAAAAVAABADQAAACkAAAAAAAAADQAIAABACgABAADAAEAAAAA
AAAAAAABAAAAAQBcAAAAXAAAAAUAAAAAEAAAcyUAGG8AAABBIwAAAHJpc2N2AAEZAAAABXJ2MzJp
MnAwX20ycDBfYTJwMAAALnNoc3RydGFiAC50ZXh0AC5yaXNjdi5hdHRyaWJ1dGVzAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACwAAAAEAAAAGAAAAVAABAFQAAAAIAAAA
AAAAAAAAAAAEAAAAAAAAABEAAAADAABwAAAAAAAAAABcAAAAJAAAAAAAAAAAAAAAAQAAAAAAAAAB
AAAAAwAAAAAAAAAAAAAAgAAAACMAAAAAAAAAAAAAAAEAAAAAAAAA
^D
root@buildroot:~# base64 -d test.b64 > test
root@buildroot:~# chmod a+x test
root@buildroot:~# ./test
- /mnt/d/Code/FPGA/linux-on-litex-vexriscv/build/sim/gateware/dut.v:1243: Verilog $finish
xobs@Cuboid:~/Code/FPGA/linux-on-litex-vexriscv$
```

xobs, 2020-01-29:
I've spent the day working on this, and I now have it generating "Illegal Instruction" exceptions, however I'm still trying to figure out how to get it to go into Machine mode when the exception hits...

mateusz-holenko, 2020-01-29:
In Renode there is already a csr validation logic - take a look at: https://github.com/antmicro/tlib/blob/master/arch/riscv/op_helper.c#L630.

It is, however, disabled for VexRiscv CPU: https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Cores/RiscV/VexRiscv.cs#L19 .

We did so, because {Machine,Supervisor}IrqPending CSRs did not validate correctly - they have bits saying they are read only, but we need to write them.

As for going into M-Mode - do you have any interrupt delegation configured? By default all interrupts should be handled in M-Mode.

xobs, 2020-01-30:
I didn't realize you already had CSR validation logic, but that it was disabled.

I was having trouble going into M-mode because of the bug fixed in https://github.com/antmicro/tlib/pull/1/commits/93ea9b1f1865484f001d905898bacb9eb5fee5d4 -- it would trap, then try to update `$mstatus`, then trap again.  Moving the privilege step fixed that problem.  Maybe that's the issue you're referring to with validation?

The issue I'm running is that Renode is behaving differently from Verilator.  Would the right thing to do be to re-enable CSR validation, because that's how it is in hardware?

PiotrZierhoffer, 2020-01-30:
We'll need to redo the validation mechanism a bit to allow non-standard CSRs to have non-standard bit patterns. But otherwise you're correct, we'd want that enabled.

mateusz-holenko, 2020-02-11:
@xobs: In the current master (https://github.com/renode/renode/commit/e34fb77d77a39eed751cf3b4e8e7d13a8a42cdcf) we verify privilege level when accessing CSRs in VexRiscv and raise `Illegal Instruction` (2) exception in case the verification fails. Does it solve this issue?

xobs, 2020-02-13:
@mateusz-holenko it almost works.  The exception doesn't appear to be setting `$mepc` correctly:

Renode:
```
CPU Exception: Illegal instruction 0x00000000 at 0x2050020c
```

Verilator:
```
CPU Exception: Illegal instruction 0x18002573 at 0x2050020c
```

mateusz-holenko, 2020-02-18:
This looks like a bit more complicated issue - I'll look into this.

mateusz-holenko, 2020-02-26:
I created a separate task for tracking the MEPC value problem: #76. Closing this one.

closed 2020-02-26 by mateusz-holenko