renode/renode #63: error running litex_vexriscv_linux.resc

pdp7, 2020-02-02:
I installed renode_1.8.2_amd64.deb and I am trying to run LiteX on Linux-VexRiscV with renode.  

I saw @mateusz-holenko comment in this PR: https://github.com/litex-hub/linux-on-litex-vexriscv/pull/33#issuecomment-516329237
> You can simulate it by starting Renode and typing in monitor:
> ```start @scripts/single-node/litex_vexriscv_linux.resc```

I ran this in the monitor:
```
start @scripts/single-node/litex_vexriscv_linux.resc
```

but i got an error:
```
Object reference not set to an instance of an object
  at System.Collections.Generic.LargeArrayBuilder`1[T].GetBuffer (System.Int32 index) [0x00022] in <13c0993ff82548209b09f271bd5e6a78>:0 
  at System.Collections.Generic.LargeArrayBuilder`1[T].CopyTo (System.Collections.Generic.CopyPosition position, T[] array, System.Int32 arrayIndex, System.Int32 count) [0x00041] in <13c0993ff82548209b09f271bd5e6a78>:0 
  at System.Collections.Generic.SparseArrayBuilder`1[T].CopyTo (T[] array, System.Int32 arrayIndex, System.Int32 count) [0x0009a] in <13c0993ff82548209b09f271bd5e6a78>:0 
  at System.Collections.Generic.SparseArrayBuilder`1[T].ToArray () [0x00028] in <13c0993ff82548209b09f271bd5e6a78>:0 
  at System.Linq.Enumerable+Concat2Iterator`1[TSource].ToArray () [0x00024] in <13c0993ff82548209b09f271bd5e6a78>:0 
  at System.Linq.Enumerable.ToArray[TSource] (System.Collections.Generic.IEnumerable`1[T] source) [0x00021] in <13c0993ff82548209b09f271bd5e6a78>:0 
  at Antmicro.Renode.UserInterface.Monitor.RunCommand (AntShell.Commands.ICommandInteraction writer, Antmicro.Renode.UserInterface.Commands.Command command, System.Collections.Generic.IList`1[T] parameters) [0x00406] in <d47614c6567449c69a4ac1c1bdcf6d28>:0 
  at Antmicro.Renode.UserInterface.Monitor.ExecuteCommand (Antmicro.Renode.UserInterface.Tokenizer.Token[] com, AntShell.Commands.ICommandInteraction writer) [0x00133] in <d47614c6567449c69a4ac1c1bdcf6d28>:0 
  at Antmicro.Renode.UserInterface.Monitor.ParseTokens (System.Collections.Generic.IEnumerable`1[T] tokensToParse, AntShell.Commands.ICommandInteraction writer) [0x001d5] in <d47614c6567449c69a4ac1c1bdcf6d28>:0 
  at Antmicro.Renode.UserInterface.Monitor.Parse (System.String cmd, AntShell.Commands.ICommandInteraction writer) [0x001a7] in <d47614c6567449c69a4ac1c1bdcf6d28>:0 
  at Antmicro.Renode.UserInterface.Monitor.ProcessCommand (System.String command, AntShell.Commands.ICommandInteraction writer) [0x00000] in <d47614c6567449c69a4ac1c1bdcf6d28>:0 
  at Antmicro.Renode.UserInterface.Monitor.HandleCommand (System.String cmd, AntShell.Commands.ICommandInteraction ci) [0x00000] in <d47614c6567449c69a4ac1c1bdcf6d28>:0 
  at AntShell.Shell.HandleCommand (System.String cmd, AntShell.Commands.ICommandInteraction ic) [0x000bd] in <892b95394fdf4e18974bb957344bb8b5>:0 
  at AntShell.CommandLine.HandleControlSequence (AntShell.Helpers.ControlSequence seq) [0x00a45] in <892b95394fdf4e18974bb957344bb8b5>:0 
  at AntShell.Terminal.NavigableTerminalEmulator.HandleInput (System.Object input) [0x0002e] in <892b95394fdf4e18974bb957344bb8b5>:0 
  at AntShell.Terminal.NavigableTerminalEmulator.Run (System.Boolean stopOnError) [0x00017] in <892b95394fdf4e18974bb957344bb8b5>:0 
  at AntShell.Shell.Start (System.Boolean stopOnError) [0x000f9] in <892b95394fdf4e18974bb957344bb8b5>:0 
  at Antmicro.Renode.UI.CommandLineInterface+<>c__DisplayClass0_1.<Run>b__3 (System.Object x) [0x00000] in <cbc535240d924f9182150dcd1bce9341>:0 
  at System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) [0x0002c] in <d0e12f672b88444ab4b6d9b2ecf20142>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x0008d] in <d0e12f672b88444ab4b6d9b2ecf20142>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <d0e12f672b88444ab4b6d9b2ecf20142>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) [0x00031] in <d0e12f672b88444ab4b6d9b2ecf20142>:0 
  at System.Threading.ThreadHelper.ThreadStart (System.Object obj) [0x00012] in <d0e12f672b88444ab4b6d9b2ecf20142>:0 
```

Any idea how I could troubleshoot?

Thanks!
[image]

pdp7, 2020-02-03:
It appears that Ubuntu 19.10 has too old of a version of mono.

I installed the latest Stable Mono release is: 6.8.0 Stable (6.8.0.96):
https://www.mono-project.com/download/stable/

with:
```
Mono JIT compiler version 6.8.0.96 (tarball Wed Jan 15 10:08:18 UTC 2020)
```

Litex on Linux runs ok with Renode!
[image]

cc @mithro

closed 2020-02-03 by pdp7

mithro, 2020-02-03:
Can renode check the Mono version and return a useful error message in some way?

mateusz-holenko, 2020-02-03:
@pdp7 - Good to hear it works!

Ubuntu 19.10 comes by default with mono 5.18, whereas Renode requires at least 5.20. The `deb` package, however, have correct dependencies so installing it in your system should automatically update mono. I'll take a look into this.

I'll also add a check in the startup script that verifies Renode is run on a correct version of mono as @mithro suggested.

pgielda, 2020-02-04:
Btw can't we somehow gracefully detect this error somehow at start and if we detect that mono version is wrong just inform the user about it in a nice(r) way?

PiotrZierhoffer, 2020-02-04:
This is already in progress.

An interesting fact - when you install Renode with dpkg on an older mono, it warns you, but does not fail.