renode/renode #67: litex_video for litex_linux_vexriscv.repl

pdp7, 2020-02-11:
Would it be possible for me to add litex_video to the litex_linux_vexriscv.repl?

I want to simulate using the LCD on Hackaday badge.  It is using litex-on-linux-vexriscv, so I thought this would be a good way to test.

I'm running Renode, version 1.8.2.22285 (dc338ac5-201911131222) which
I installed using the Debian package from the GitHub releases page.

I also cloned the repo and am using the files from there:
/home/pdp7/dev/renode/tests/platforms/LiteX/litex_linux_vexriscv.repl (attached)

Here is the state of the repo:
```diff
dp7@x1:~/dev/renode$ git diff
diff --git a/lib/termsharp b/lib/termsharp
--- a/lib/termsharp
+++ b/lib/termsharp
@@ -1 +1 @@
-Subproject commit 948fb9ad98bc33535659c027dd5264b68b42bc58
+Subproject commit 948fb9ad98bc33535659c027dd5264b68b42bc58-dirty
diff --git a/tests/platforms/LiteX/litex_linux_vexriscv.repl
b/tests/platforms/LiteX/litex_linux_vexriscv.repl
index 123554b..c57b2af 100644
--- a/tests/platforms/LiteX/litex_linux_vexriscv.repl
+++ b/tests/platforms/LiteX/litex_linux_vexriscv.repl
@@ -62,3 +62,7 @@ led : Miscellaneous.LED @ gpio_out 0

 button: Miscellaneous.Button @ gpio_in 0
     -> gpio_in@0
+
+litex_video: Video.LiteX_Framebuffer @ sysbus 0xf0006800
+    format: PixelFormat.XBGR8888
+    memory: ram
```
```
pdp7@x1:~/dev/renode$ git log --oneline -1
c16eba8 (HEAD -> master, origin/master, origin/HEAD) [#17690] arm: Expose IT_STATE

pdp7@x1:~/dev/renode$ git status
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)

modified:   lib/termsharp (modified content)
modified:   tests/platforms/LiteX/litex_linux_vexriscv.repl
```
but I get this error:
```
(machine-3) showAnalyzer sysbus.litex_video
There was an error executing command 'showAnalyzer sysbus.litex_video'
Received 'Peripheral not found: sysbus.litex_video' error while initializing analyzer for: sysbus.litex_video. 
Are you missing a required plugin?
```

pdp7, 2020-02-11:
Contents of:
`/home/pdp7/dev/renode/tests/platforms/LiteX/litex_linux_vexriscv.repl`
```
emulator_ram: Memory.MappedMemory @ {
        sysbus 0x20000000
    }
    size: 0x10000000

ddr: Memory.MappedMemory @ {
        sysbus 0xc0000000
    }
    size: 0x10000000

uart: UART.LiteX_UART @ {
        sysbus 0xf0001000
    }
    -> cpu@0

eth: Network.LiteX_Ethernet @ {
        sysbus 0xF0003800;

        sysbus new Bus.BusMultiRegistration { address: 0x30000000; size: 0x2000; region: "buffer" };
        sysbus new Bus.BusMultiRegistration { address: 0xB0000000; size: 0x2000; region: "buffer" }
    }
    -> cpu@2

spi: SPI.LiteX_SPI @ {
    sysbus 0xF0006800
}

spi_loopback: SPI.SPILoopback @ spi

i2c: I2C.LiteX_I2C @ {
    sysbus 0xF0007000
}

si7021: Sensors.SI70xx @ i2c 5
    model: Model.SI7021

gpio_out: GPIOPort.LiteX_GPIO @ {
    sysbus 0xF0005800
}
    type: Type.Out
    0 -> led@0

gpio_in: GPIOPort.LiteX_GPIO @ {
    sysbus 0xF0006000
}
    type: Type.In

cpuTimer: Timers.LiteX_CPUTimer @ {
        sysbus 0xF0000800
    }
    // according to the LiteX Configuration this should be 10.000.000 Hz, but I modified it in order to obtain 10ms tick
    frequency: 40000000
    // IRQ #100 is Machine Timer Interrupt
    -> cpu@100

cpu: CPU.VexRiscv @ sysbus
    cpuType: "rv32ima"
    privilegeArchitecture: PrivilegeArchitecture.Priv1_10
    timeProvider: cpuTimer

led : Miscellaneous.LED @ gpio_out 0

button: Miscellaneous.Button @ gpio_in 0
    -> gpio_in@0

litex_video: Video.LiteX_Framebuffer @ sysbus 0xf0006800
    format: PixelFormat.XBGR8888
    memory: ram
```

pdp7, 2020-02-11:
@mateusz-holenko I thought this issue might better than the email thread I started about it.  Thanks!

mateusz-holenko, 2020-02-11:
@pdp7: `memory: ram` looks suspicious as there is no memory named `ram` in the platform. The value should probably be `ddr`.

You should get an error when running `machine LoadPlatformDescription @tests/platforms/LiteX/litex_linux_vexriscv.repl` command, though.

I'm getting one on the current master:
```
Renode, version 1.8.2.23517 (b46924b7-202002111302)
(monitor) mach create
(machine-0) machine LoadPlatformDescription @tests/platforms/LiteX/litex_linux_vexriscv.repl
There was an error executing command 'machine LoadPlatformDescription /home/houen/antmicro/hq/master/renode/tests/platforms/LiteX/litex_linux_vexriscv.rep
l'
Error E20: Undefined reference 'ram'.
At /home/houen/antmicro/hq/master/renode/tests/platforms/LiteX/litex_linux_vexriscv.repl:68:13:
    memory: ram
            ^^^
(machine-0) 
```

I'll verify it on the Debian package you use.

pdp7, 2020-02-11:
Thanks for the suggestion.

I changed `ram` to `ddr`:
```diff
pdp7@x1:~/dev/renode$ git diff
diff --git a/lib/termsharp b/lib/termsharp
--- a/lib/termsharp
+++ b/lib/termsharp
@@ -1 +1 @@
-Subproject commit 948fb9ad98bc33535659c027dd5264b68b42bc58
+Subproject commit 948fb9ad98bc33535659c027dd5264b68b42bc58-dirty
diff --git a/tests/platforms/LiteX/litex_linux_vexriscv.repl b/tests/platforms/LiteX/litex_linux_vexriscv.repl
index 123554b..4827496 100644
--- a/tests/platforms/LiteX/litex_linux_vexriscv.repl
+++ b/tests/platforms/LiteX/litex_linux_vexriscv.repl
@@ -62,3 +62,7 @@ led : Miscellaneous.LED @ gpio_out 0
 
 button: Miscellaneous.Button @ gpio_in 0
     -> gpio_in@0
+
+litex_video: Video.LiteX_Framebuffer @ sysbus 0xf0006800
+    format: PixelFormat.XBGR8888
+    memory: ddr
```
However, I still get the same error:
```
Renode, version 1.8.2.22285 (dc338ac5-201911131222)
(monitor) using sysbus
(monitor) mach create
(machine-0) machine LoadPlatformDescription @tests/platforms/LiteX/litex_linux_vexriscv.repl
(machine-0) set kernel @https://antmicro.com/projects/renode/litex_linux_vexriscv--kernel.bin-s_4578292-f63a4736100b5ff79a8d72429c1b79718ec7a446
(machine-0) set rootfs @https://antmicro.com/projects/renode/litex_linux_vexriscv--rootfs.cpio-s_4163584-c44ad487ba1f73c00430a1bb108ceef84007274f
(machine-0) set device_tree @https://antmicro.com/projects/renode/litex_linux_vexriscv--rv32.dtb-s_2609-9a915b47b8e31d0d3f268c4a297dc0b0555e8cd0
(machine-0) set emulator @https://antmicro.com/projects/renode/litex_vexriscv--emulator.bin-s_9028-796a4227b806997c6629462fdf0dcae73de06929
(machine-0) 
(machine-0) sysbus LoadBinary $emulator 0x20000000
(machine-0) sysbus LoadBinary $kernel 0xc0000000
(machine-0) sysbus LoadBinary $rootfs 0xc0800000
(machine-0) sysbus LoadBinary $device_tree 0xc1000000
(machine-0) 
(machine-0) cpu PC 0x20000000
(machine-0) showAnalyzer sysbus.uart
(machine-0) showAnalyzer sysbus.litex_video
There was an error executing command 'showAnalyzer sysbus.litex_video'
Received 'Peripheral not found: sysbus.litex_video' error while initializing analyzer for: sysbus.litex_video. Are you missing a required plugin?
(machine-0) 
```

mateusz-holenko, 2020-02-11:
Try using absolute path to `litex_linux_vexriscv.repl`. Perhaps you are not loading the file you want after all.

pdp7, 2020-02-11:
Thanks.  I am using absolute path and it is different now:
```
(machine-0) using sysbus
(machine-0) mach create
(machine-1) machine LoadPlatformDescription @/home/pdp7/dev/renode/tests/platforms/LiteX/litex_linux_vexriscv.repl
There was an error executing command 'machine LoadPlatformDescription /home/pdp7/dev/renode/tests/platforms/LiteX/litex_linux_vexriscv.repl'
Error E39: Exception was thrown during registration of 'litex_video in 'sysbus':
At /home/pdp7/dev/renode/tests/platforms/LiteX/litex_linux_vexriscv.repl:66:40:
litex_video: Video.LiteX_Framebuffer @ sysbus 0xf0006800
                                       ^
(machine-1) set kernel @https://antmicro.com/projects/renode/litex_linux_vexriscv--kernel.bin-s_4578292-f63a4736100b5ff79a8d72429c1b79718ec7a446
(machine-1) set rootfs @https://antmicro.com/projects/renode/litex_linux_vexriscv--rootfs.cpio-s_4163584-c44ad487ba1f73c00430a1bb108ceef84007274f
(machine-1) set device_tree @https://antmicro.com/projects/renode/litex_linux_vexriscv--rv32.dtb-s_2609-9a915b47b8e31d0d3f268c4a297dc0b0555e8cd0
(machine-1) set emulator @https://antmicro.com/projects/renode/litex_vexriscv--emulator.bin-s_9028-796a4227b806997c6629462fdf0dcae73de06929
(machine-1) 
(machine-1) sysbus LoadBinary $emulator 0x20000000
(machine-1) sysbus LoadBinary $kernel 0xc0000000
(machine-1) sysbus LoadBinary $rootfs 0xc0800000
(machine-1) sysbus LoadBinary $device_tree 0xc1000000
(machine-1) 
(machine-1) cpu PC 0x20000000
(machine-1) showAnalyzer sysbus.uart
(machine-1) showAnalyzer sysbus.litex_video
There was an error executing command 'showAnalyzer sysbus.litex_video'
Received 'Peripheral not found: sysbus.litex_video' error while initializing analyzer for: sysbus.litex_video. Are you missing a required plugin?
```
It appears this is the issue:
```

Could not register Given address <0xF0006800, 0xF00068FF> for peripheral Antmicro.Renode.Peripherals.Video.LiteX_Framebuffer conflicts with address <0xF0006800, 0xF000684F> of peripheral Antmicro.Renode.Peripherals.SPI.LiteX_SPI in address.
```
Any suggestions?

pdp7, 2020-02-11:
@mateusz-holenko Is there a way to view the LiteX configuration files that built the binaries like:
```
https://antmicro.com/projects/renode/litex_linux_vexriscv--kernel.bin-s_4578292-f63a4736100b5ff79a8d72429c1b79718ec7a446
```
I'd like to figure out what the correct address is for:
```
+litex_video: Video.LiteX_Framebuffer @ sysbus 0xNNNNNNNN
```

mateusz-holenko, 2020-02-11:
@pdp7: TBH I'm not even sure if this binary supports frame buffer at all. We currently does not have an easy way to trace back the LiteX configuration from the binary. It was definiately built from https://github.com/litex-hub/linux-on-litex-vexriscv, but don't know the exact commit.

You can always use the binary from https://github.com/renode/renode/blob/master/tests/platforms/LiteX/Netv_LiteX_VexRiscv.robot#L14 or build your own `linux-on-litex-vexriscv` following commands from: https://github.com/litex-hub/linux-on-litex-vexriscv#generating-the-linux-binaries-optional

pdp7, 2020-02-11:
@mateusz-holenko thanks, without rebuilding is there a way for me to determine the address I should put in:
```
litex_video: Video.LiteX_Framebuffer @ sysbus 0xNNNNNNNN
```
as the one from the NeTV example conflicts with `spi: SPI.LiteX_SPI`

mateusz-holenko, 2020-02-11:
@pdp7: The linux kernel is not a problem as it does not have any addresses hardcoded inside - all the information is dynamically read from the DTB file. You can just decompile it back to string, modify the framebuffer address to match the one in `repl` and compile it again. IOW you can put it wherever you want as long as it is consistent accress DTB and `repl` and does not conflicts with other peripherals.

The biggest problem is the VexRiscv `emulator`. It's a software that handles interrupts and does an emulation of atomic instructions, uart over SBI and (in the past) framebuffer. It has, unfortunatelly, peripheral addresses hardcoded inside. Using an `emulator` built for different version of `linux-on-litex-vexriscv` might simply not work.

That's why I strongly suggest to (a) either use a consistent set of binaries, DTB & platform definition taken from a single demo or (b) rebuilt everything from scratch. `linux-on-litex-vexriscv` will generate kernel, rootfs, DTB and the emulator for you.

mateusz-holenko, 2020-02-11:
BTW, LiteX BIOS also comes with a hardcoded set of peripheral addresses, so it must match the LiteX configuration. I didn't mention it since we don't use it in the demo (we skip the bootloader and just put binaries directly to the memory from Renode script).

pdp7, 2020-02-11:
I have built the binaries from source for the ECP5 on the Hackaday badge:
```
pdp7@x1:~/dev/enjoy/linux-on-litex-vexriscv$ ls -la buildroot/Image buildroot/rootfs.cpio buildroot/rv32.dtb emulator/emulator.bin
-rw-r--r-- 1 pdp7 pdp7 4545524 Jan 11 15:45 buildroot/Image
-rw-r--r-- 1 pdp7 pdp7 8029184 Jan 11 15:45 buildroot/rootfs.cpio
-rw-r--r-- 1 pdp7 pdp7    1808 Feb  2 15:37 buildroot/rv32.dtb
-rw-r--r-- 1 pdp7 pdp7    9584 Feb  2 15:37 emulator/emulator.bin
```
```
pdp7@x1:~/dev/enjoy/linux-on-litex-vexriscv$ cat images.json
{
	"buildroot/Image":        "0xc0000000",
	"buildroot/rootfs.cpio":  "0xc0800000",
	"buildroot/rv32.dtb":     "0xc1000000",
	"emulator/emulator.bin":  "0x20000000"
}
```
I didn't occur to me that I could use those with renode.

pdp7, 2020-02-11:
Thanks!
[image]

I can boot renode with those binaries that I built.
```
using sysbus
mach create
machine LoadPlatformDescription @/home/pdp7/dev/renode/tests/platforms/LiteX/had.repl
set kernel @/home/pdp7/dev/enjoy/linux-on-litex-vexriscv/buildroot/Image
set rootfs @/home/pdp7/dev/enjoy/linux-on-litex-vexriscv/buildroot/rootfs.cpio
set device_tree @/home/pdp7/dev/enjoy/linux-on-litex-vexriscv/buildroot/rv32.dtb
set emulator @/home/pdp7/dev/enjoy/linux-on-litex-vexriscv/emulator/emulator.bin

sysbus LoadBinary $emulator 0x20000000
sysbus LoadBinary $kernel 0xc0000000
sysbus LoadBinary $rootfs 0xc0800000
sysbus LoadBinary $device_tree 0xc1000000

cpu PC 0x20000000
showAnalyzer sysbus.uart
```

I'll take a look at how I can add the framebuffer.

mateusz-holenko, 2020-02-26:
@pdp7: How things are going? Were you able to add the framebuffer?

pdp7, 2020-02-26:
@mateusz-holenko thanks for following up.  I'm at Embedded World (great Antmicro booth!) but I will get back to trying to solution once I get back home.