renode/renode #68: build error: bsdtar is missing

pdp7, 2020-02-11:
I am building from master in this repo:
```
c16eba8 (HEAD -> master, origin/master, origin/HEAD) [#17690] arm: Expose IT_STATE
```
Error:
```
bsdtar is missing. Install it to continue.
/usr/bin/ld: /usr/lib/pkgconfig/../../lib/libmono-2.0.a(llvm-runtime.o): in function `mono_llvm_cpp_catch_exception':
(.text+0x43): undefined reference to `__cxa_begin_catch'
/usr/bin/ld: (.text+0x4f): undefined reference to `__cxa_end_catch'
/usr/bin/ld: /usr/lib/pkgconfig/../../lib/libmono-2.0.a(llvm-runtime.o):(.data.rel.local.DW.ref._ZTIPi[DW.ref._ZTIPi]+0x0): undefined reference to `typeinfo for int*'
/usr/bin/ld: /usr/lib/pkgconfig/../../lib/libmono-2.0.a(llvm-runtime.o):(.data.rel.local.DW.ref.__gxx_personality_v0[DW.ref.__gxx_personality_v0]+0x0): undefined reference to `__gxx_personality_v0'
```
I'm on Ubuntu 19.10:
```
pdp7@x1:~/dev/renode$ sudo apt install bsdtar
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package bsdtar is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libarchive-tools:i386 libarchive-tools

E: Package 'bsdtar' has no installation candidate
```

pdp7, 2020-02-11:
I solved this by installing:
```
sudo apt install libarchive-tools
```
However, build now finishes with this error:
```
Created a Debian package in output/packages/renode_1.8.2_amd64.deb
Created a Fedora package in output/packages/renode-1.8.2-1.f23.x86_64.rpm
Created an Arch package in output/packages/renode-1.8.2-1-x86_64.pkg.tar.xz
/usr/bin/ld: /usr/lib/pkgconfig/../../lib/libmono-2.0.a(llvm-runtime.o): in function `mono_llvm_cpp_catch_exception':
(.text+0x43): undefined reference to `__cxa_begin_catch'
/usr/bin/ld: (.text+0x4f): undefined reference to `__cxa_end_catch'
/usr/bin/ld: /usr/lib/pkgconfig/../../lib/libmono-2.0.a(llvm-runtime.o):(.data.rel.local.DW.ref._ZTIPi[DW.ref._ZTIPi]+0x0): undefined reference to `typeinfo for int*'
/usr/bin/ld: /usr/lib/pkgconfig/../../lib/libmono-2.0.a(llvm-runtime.o):(.data.rel.local.DW.ref.__gxx_personality_v0[DW.ref.__gxx_personality_v0]+0x0): undefined reference to `__gxx_personality_v0'
collect2: error: ld returned 1 exit status
```
Here is the [full log](https://gist.github.com/pdp7/d0a044c482104f834163c80eccd49019).

PiotrZierhoffer, 2020-02-11:
Are you building with `-p`? What mono version are you running?

pdp7, 2020-02-11:
Yes, with here is the command and mono version:
```
pdp7@x1:~/dev/renode$ ./build.sh -p
Required Renode libraries already downloaded. To repeat the process remove .renode_libs_fetched file.

>>>> xbuild tool is deprecated and will be removed in future updates, use msbuild instead <<<<

XBuild Engine Version 14.0
Mono, Version 6.8.0.105
Copyright (C) 2005-2013 Various Mono authors

Build started 2/11/2020 2:36:47 PM.
```

PiotrZierhoffer, 2020-02-11:
It seems you're building a portable linux package. This relies on some undocumented mono capabilities and proves not to be very stable across releases.

We will filter this out not to be built by default, but you will find that actually your build was successful and other packages are in `output/packages`.

To avoid this error, you can remove the reference to `make_linux_portable.sh` from `build.sh` (lines 181-184)

PiotrZierhoffer, 2020-02-11:
Btw, full requirements to building packages are listed here: https://renode.readthedocs.io/en/latest/advanced/building_from_sources.html#creating-packages

This does not, however, mitigate the problems with the mono version.

pdp7, 2020-02-11:
Thanks, it does now finish cleanly:
```
Build succeeded.
	 0 Warning(s)
	 0 Error(s)

Time Elapsed 00:00:11.7761570
Created a Debian package in output/packages/renode_1.8.2_amd64.deb
Created a Fedora package in output/packages/renode-1.8.2-1.f23.x86_64.rpm
Created an Arch package in output/packages/renode-1.8.2-1-x86_64.pkg.tar.xz
/home/pdp7/dev/renode/tools/packaging/make_linux_portable.sh
```
```diff
pdp7@x1:~/dev/renode$ git diff build.sh
diff --git a/build.sh b/build.sh
index 979f927..9e8839e 100755
--- a/build.sh
+++ b/build.sh
@@ -180,6 +180,6 @@ then
 
     if $ON_LINUX
     then
-      $ROOT_PATH/tools/packaging/make_linux_portable.sh $params
+      echo $ROOT_PATH/tools/packaging/make_linux_portable.sh $params
     fi
 fi
```
And I was able to install the deb ok:
```
pdp7@x1:~/dev/renode$ sudo dpkg -i output/packages/renode_1.8.2_amd64.deb
[sudo] password for pdp7: 
Sorry, try again.
[sudo] password for pdp7: 
(Reading database ... 354649 files and directories currently installed.)
Preparing to unpack .../renode_1.8.2_amd64.deb ...
Unpacking renode (1.8.2) over (1.8.2) ...
gtk-update-icon-cache: No theme index file.
Setting up renode (1.8.2) ...
gtk-update-icon-cache: No theme index file.
Processing triggers for desktop-file-utils (0.24-1ubuntu1) ...
Processing triggers for mime-support (3.63ubuntu1) ...
Processing triggers for gnome-menus (3.32.0-1ubuntu1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
```

closed 2020-02-11 by pdp7

PiotrZierhoffer, 2020-02-20:
@pdp7 FYI the portable package is no longer built along the regular packages, so you should no longer see the error when building with `-p`