renode/renode #69: Building on Windows without cygwin

xobs, 2020-02-13:
The Windows instructions state that cygwin is required to build, and that `build.sh` must be run first.  This is not necessarily required.

This is a script that I've started using.  It can be run in normal Powershell from a freshly-cloned renode repo:

```powershell
param (
    [Boolean]$submodules = $true,
    [switch]$debug = $false,
    [switch]$clean = $false
)

$target = "Release"
if ($debug) {
    $target = "Debug"
}

# We should only do this if "-s" is passed, or if submodules
# are not initialized
if ($submodules) {
    git submodule update --init --recursive
}

# It's unclear why this is a separate repository.
git -C lib clone https://github.com/renode/renode-resources.git resources

# Update references to Xwt
(Get-Content .\lib\termsharp\TermSharp.csproj)|
    ForEach-Object {$_ -replace """xwt\\Xwt\\Xwt.csproj", """..\xwt\Xwt\Xwt.csproj"} |
    Set-Content .\lib\termsharp\TermSharp.csproj

# Build CCTask, which is used to run subtasks
MSBuild /p:Configuration=Release /p:Platform="Any CPU" .\lib\cctask\CCTask.sln

# Copy the properties file, which contains various build environment settings
if (-not (Test-path output\properties.csproj)) {
    Copy-Item src\Infrastructure\src\Emulator\Cores\windows-properties.csproj output\properties.csproj
}

# Run the "clean" task if the user requested
if ($clean) {
    MSBuild /p:Configuration=Release /p:Platform="Any CPU" /t:Clean Renode-Windows.sln
    MSBuild /p:Configuration=Debug /p:Platform="Any CPU" /t:Clean Renode-Windows.sln
    return
}

MSBuild /p:Configuration=$target /p:Platform="Any CPU" Renode-Windows.sln

# copy llvm library
Copy-Item src/Infrastructure/src/Emulator/LLVMDisassembler/bin/$target/libLLVM.* output/bin/$target
```

The only change I need to make is specifying the complete path to the compiler, linker, and ar inside `output/properties.csproj`.

This script could be further enhanced to provide feature-parity with `build.sh` if necessary.

Is this a thing that @renode is interested in?

PiotrZierhoffer, 2020-02-13:
Thanks for this, @xobs. 

This is something we'll put in the pipeline to consider. Of course making it into a fully-featured thing will not be trivial, as other scripts are shell as well. 
One of the approaches we also tested and started the implementation of was going full Python, but this still requires some work.

Just to address your doubts about renode-resources - this repo mainly contains binaries. In case of major changes to the dependencies, we tend to squash the history, thus reducing the repo size significantly. If we kept them on the main repo, that would be impossible.

jensschroer, 2020-03-17:
I just stumbled over this myself, but for other reasons (different IDE than monodevelop..).
I was wondering if it would make sense to consider using cmake instead of scripts? 
It has native C# support since version 3.8, monodevelop and Visual Studio support cmake generators and it is also an easy choice for people outside of these IDEs.

Having said that, neither I am  a mono nor a cmake expert...

xobs, 2020-03-17:
This should work, as long as your system has msbuild.exe, which has been part of mono since maybe 2016?  Though in that case, the program may be renamed to `msbuild.exe` or `msbuild`, due to case-sensitivity on Linux.

I've used this script on a few installs now.

jensschroer, 2020-03-18:
> I just stumbled over this myself, but for other reasons (different IDE than monodevelop..).
> I was wondering if it would make sense to consider using cmake instead of scripts?
> It has native C# [...]

Never mind , I just checked the mono support in cmake a bit more, and at the time being it is restricted to Windows only. [This cmake ticket](https://gitlab.kitware.com/cmake/cmake/issues/16865) contains the details.