renode/renode #71: GDB: RISC-V: Memory addresses are untranslated

xobs, 2020-02-17:
When accessing memory via the GDB bridge, accesses are all untranslated.  This results in nonsensical output, and makes it difficult to debug.

Output from GDB:

```
(gdb) bt
#0  0x00800148 in __ebreak ()
#1  0x00806040 in vexriscv::asm::ebreak ()
    at C:\Users\smcro\.cargo\registry\src\github.com-1ecc6299db9ec823\vexriscv-0.0.2\src/asm.rs:18
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb) mon cpu SATP
0x80040fe6
(gdb) p/x $sp
$1 = 0xdffff88c
(gdb)
```

In the above trace, the stack pointer is at `0xdffff88c`, which is mapped via the MMU to somewhere in the range of `0x4xxxxxxx`.  Additionally, program data is at `0x00800000`, which is also mapped somewhere in `0x4xxxxxxx`.

The gdb bridge is not translating these values, even though SATP is active.  This prevents backtraces and variable inspection from working.

mateusz-holenko, 2020-02-26:
@xobs How does it work in the FPGA? Are you able to get correct addresses in GDB? Does it require any additional layer - like OpenOCD, support from the kernel?

I'm asking because in general it looks like a complicated problem. Consulting MMU tables in HW might not be enough, since an OS might keep more translations and replace them on the fly, isn't that right?  So to have a correct solution we would have to read the translation data from the OS memory, hence integrate tightly with it.

I'm talking about any architecture here. not strictly RISC-V; perhaps it's easier in the case of RISC-V?

xobs, 2020-02-26:
`riscv-openocd` recently added the ability to do translation in openocd, so memory can be either translated or untranslated depending on a flag: https://github.com/riscv/riscv-openocd/commit/9886f7737403d2ea378dd3349de4b70b1b62ce54

I have `wishbone-tool` that allows for inspecting memory via gdb, so I'll probably add translation to that as well, but it would be nice to have this in Renode.

xobs, 2020-10-15:
It would be really nice to get this working, particularly since I'm working on an operating system.

What about this:

* When `gdb` sets a breakpoint, check the status of memory translation. If enabled, translate the requested address to a physical page and add a breakpoint there.
* When reading memory in gdb, use the current translation tables as pointed to by `satp`. This involves one or two pointer dereferences.

I'm unaware of how other platforms do it, and you're right in that operating systems can do all sorts of odd things with page faults, e.g. on lazily-loaded libraries.

mateusz-holenko, 2021-01-06:
@xobs, in e1291cde2023cf3cac46dc7641735550dfc72613 we added the support for translating addresses accessed by GDB. Could you check if it works for you?

PiotrZierhoffer, 2021-05-05:
As the basic mechanism works, let's close this issue. The other one to be investigated is about GDB breakpoints: https://github.com/renode/renode/issues/197

closed 2021-05-05 by PiotrZierhoffer