renode/renode #72: [riscv] Whould it be possible to set the PCs with the GDB?

truhlikfredy, 2020-02-19:
On a multihart riscv target, to set the pcs from the GDB we can invoke the monitor-level command:
```monitor sysbus SetPCOnAllCores ```

Would it be possible to get the same effect from the GDB-level command directly?
```thread apply all set $pc=_start```

PiotrZierhoffer, 2020-03-10:
I do believe it actually works fine!

I started the Unleashed sample and ran:

```
(gdb) tar rem :3333
Remote debugging using :3333
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x0000000080000000 in ?? ()
(gdb) thread apply all set $pc=0xDEAD0000
(gdb) mon s
(gdb) c
```

And got this log:

```
15:06:42.9882 [INFO] hifive-unleashed: GDB server with all CPUs started on port :3333
15:09:40.5466 [INFO] hifive-unleashed: Machine started.
15:09:41.7742 [WARNING] sysbus: [e51: 0xDEAD0000] ReadByte from non existing peripheral at 0xDEAD0000.
15:09:41.7745 [WARNING] sysbus: [u54_1: 0xDEAD0000] ReadByte from non existing peripheral at 0xDEAD0000.
15:09:41.7746 [ERROR] e51: CPU abort [PC=0xDEAD0000]: Trying to execute code outside RAM or ROM at 0x00000000DEAD0000.
15:09:41.7747 [ERROR] u54_1: CPU abort [PC=0xDEAD0000]: Trying to execute code outside RAM or ROM at 0x00000000DEAD0000.
15:09:41.7751 [WARNING] sysbus: [u54_3: 0xDEAD0000] ReadByte from non existing peripheral at 0xDEAD0000.
15:09:41.7752 [ERROR] u54_3: CPU abort [PC=0xDEAD0000]: Trying to execute code outside RAM or ROM at 0x00000000DEAD0000.
15:09:41.7753 [WARNING] sysbus: [u54_4: 0xDEAD0000] ReadByte from non existing peripheral at 0xDEAD0000.
15:09:41.7753 [ERROR] u54_4: CPU abort [PC=0xDEAD0000]: Trying to execute code outside RAM or ROM at 0x00000000DEAD0000.
15:09:41.7754 [WARNING] sysbus: [u54_2: 0xDEAD0000] ReadByte from non existing peripheral at 0xDEAD0000.
15:09:41.7755 [ERROR] u54_2: CPU abort [PC=0xDEAD0000]: Trying to execute code outside RAM or ROM at 0x00000000DEAD0000.
```

truhlikfredy, 2020-03-16:
Thanks, I will convert projects to the monitor commands to the 'thread apply all set $pc=_start', hopefully there are no adverse effects to it.

closed 2020-03-16 by truhlikfredy