renode/renode #73: PC is not reset during `machine Reset` / `LoadELF`

xobs, 2020-02-21:
The current recommended method to restart a machine without reloading all of Renode is to add a reset macro to reload the ELF file and do `machine Reset`:

```
macro reset
"""
    sysbus LoadELF @test
"""
runMacro $reset
```

This works, however after the first invocation the program counter ends up at address `0x00001010`.  It becomes necessary to add a `cpu PC xxxxx` line to the `$reset` macro.

The entrypoint should be set on the CPU, or the `LoadELF()` command should return the entrypoint, or it should set the PC on the machine it's loading for.  At any rate, it shouldn't be different the first time versus subsequent loads.

mateusz-holenko, 2020-02-26:
`LoadELF` automatically sets PC to the value of ELF's entry point address. See: 

* https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Main/Peripherals/Bus/SystemBus.cs#L506,
* https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/CPU/TranslationCPU.cs#L99,
* https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/CPU/TranslationCPU.cs#L690.

I verified the behaviour is correct on the ELF used in LiteX Micropyhton test: https://github.com/renode/renode/blob/master/scripts/single-node/litex_vexriscv_micropython.resc#L10:

```
Renode, version 1.8.2.22406 (eeb83342-202002261226)
(monitor) i $CWD/test.resc
Mach created
0x1000
In reset, before LoadELF
0x1000
In reset, after LoadELF
0x40000000
(machine-0) s
Starting emulation...
(machine-0) p
Pausing emulation...
(machine-0) cpu PC
0x4002324c
(machine-0) machine Reset
In reset, before LoadELF
0x1000
In reset, after LoadELF
0x40000000
(machine-0) cpu PC
0x40000000
(machine-0) 
```

As you can see, the `PC` was `0x4002324c` before the reset and `0x40000000` after it.

Could you share scripts and binaries you are using?

xobs, 2020-02-27:
If I experience the error again I will share the results.  I suspect it has something to do with hitting breakpoints, and may be similar to an issue I'm seeing where I need to do a reset/run sequence twice in order to get breakpoints to trigger.

xobs, 2020-03-02:
I suspect this was totally user error, and for that I apologise.

The address `0x00001010` appears to be 16 bytes from the start of a binary that was built without a proper linker script.  If the contents of the binary changes during builds, and one particular version doesn't have a linker script, its origin will be at `0x00001000`.  It's also possible that the ELF loader got confused, however I will tend to believe it's working correctly.

I'll close this for now, and reopen it only if I find a reproducible case in the future.

closed 2020-03-02 by xobs