renode/renode #74: GDB: Server locks up when stepping to invalid instruction

xobs, 2020-02-21:
If an exception can't be handled, Renode's GDB server handles it by locking up.  The machine will need to be reset by going into the Renode console and issuing `machine Reset`.

# Reproducing

To reproduce, assemble the following code:

```asm
.global do_,ret
.global _start
_start:
    li		    t0, (1 << 11) | (1 << 5)
    csrw	    mstatus, t0
    li          a1, 0x80000010
    csrw        satp, a1
    add         a2, a1, a0

    li          t0, 0xffffffff
    csrw        medeleg, t0

do_mret:
    mret
loop:
    j       loop
```

This can be assembled with the command `riscv64-unknown-elf-gcc test.S -o test -O0 -nostdlib -march=rv32imac -mabi=ilp32 -ggdb3`.

Load the resulting file into Renode with `LoadELF`, and run the following:

```
using sysbus
mach create
machine LoadPlatformDescription @platform.repl
machine StartGdbServer 3333 true

macro reset
"""
    sysbus LoadELF @test
    cpu PC 0x10054
"""
runMacro $reset
```

Any platform file will do, but I'm still using the simple one:

```
cpu: CPU.VexRiscv @ sysbus
    timeProvider: cpu_timer
    cpuType: "rv32imac"
    privilegeArchitecture: PrivilegeArchitecture.Priv1_10

rom: Memory.MappedMemory @ {
        sysbus 0x00000000;
        sysbus 0x80000000 // shadow
    }
    size: 0x00100000

sram: Memory.MappedMemory @ {
        sysbus 0xffff0000
    }
    size: 0x10000

cpu_timer: Timers.LiteX_CPUTimer @ sysbus 0xf0000800
    frequency: 66000000
    -> cpu@100
```

Then attach GDB, break on `do_mret`, and step one instruction:

```
$ riscv64-unknown-elf-gdb -ex 'tar rem :3333' .\test -ex 'mon start' -ex 'b do_mret' -ex 'c'
```

# Observations

GDB can no longer function.  If you disconnect the session and create a new one, it will fail to attach.  The only way to revive it is to issue `machine Reset` in the Renode console.

mateusz-holenko, 2020-02-27:
### Code analysis

My understanding of the assembler code is as follows:

1. set `mstatus` register to 0x820 
    0x820 == MPP 0x1 and SPIE 0x1
    MPP set to 0x1 means that after `mret` the CPU will go into the S mode

2. set `satp` register to 0x8000_0010
    0x8000_0010 == PPN 0x10 and MODE 0x1 (Sv32)

3. do some addition 
      NOTE: what for? it doesn't look necessary

4. set `medeleg` to 0xffff_ffff
      this enables delegation of all exceptions to the S mode

5. return from "an interrupt" via `mret`
    this should switch to the S mode

### How is this simulated?

Since we are returning from the M mode trap handler, we should read the value
of `mepc` register (not set, by default 0x0) and jump there.

There is a rom memory registered at 0x0, so we should decode instructions from there.

The problem Renode is having is handling the MMU mapping in:
```
#0  0x00007fffdece851a in get_physical_address (env=0x7fffb8f3bc20, physical=0x7fffd4b712e8, prot=0x7fffd4b712e4, address=0, access_type=2, mmu_idx=1) at tlib/arch/riscv/helper.c:144
#1  0x00007fffdece8ddb in cpu_handle_mmu_fault (env=0x7fffb8f3bc20, address=0, access_type=2, mmu_idx=1) at tlib/arch/riscv/helper.c:334
#2  0x00007fffdeceddbe in tlb_fill (env=0x7fffb8f3bc20, addr=0, is_write=2, mmu_idx=1, retaddr=0x0) at tlib/arch/riscv/op_helper.c:858
#3  0x00007fffdecced4c in __ldb_cmmu (addr=0, mmu_idx=1) at tlib/include/softmmu_template.h:182
#4  0x00007fffdecc9f5b in ldub_code (ptr=0) at tlib/include/softmmu_header.h:91
#5  0x00007fffdecca083 in get_page_addr_code (env1=0x7fffb8f3bc20, addr=0) at tlib/include/exec-all.h:269
#6  0x00007fffdecca441 in tb_find_slow (env=0x7fffb8f3bc20, pc=0, cs_base=0, flags=0) at tlib/cpu-exec.c:102
#7  0x00007fffdecca6b8 in tb_find_fast (env=0x7fffb8f3bc20) at tlib/cpu-exec.c:166
#8  0x00007fffdeccaa20 in cpu_exec (env=0x7fffb8f3bc20) at tlib/cpu-exec.c:295
#9  0x00007fffdecc96c5 in tlib_execute (max_insns=1) at tlib/exports.c:152
#10 0x00000000404fac84 in  ()
#11 0xffffffffffffffff in  ()
#12 0x0000000000000001 in  ()
#13 0x0000000000000001 in  ()
#14 0x00007fffd8d84130 in  ()
#15 0x0000000000000000 in  ()

```

In the M mode there was no MMU translations and physical address was just equal to the virtual one so the code up to this point worked fine.

Since we are now in the S mode and translation is enabled (satp.MODE == 0x1) we should use the root page table located at (satp.PPN << 12) == 0x1_0000.

This address lies in the `rom` memory and this is the beginning of the segment read from the ELF file:

```
(machine-0) dump 0x10000 0x100
0x00010000 | 7F 45 4C 46 01 01 01 00 00 00 00 00 00 00 00 00 | ELF............
0x00010010 | 02 00 F3 00 01 00 00 00 54 00 01 00 34 00 00 00 | ........T...4...
0x00010020 | CC 03 00 00 01 00 00 00 34 00 20 00 01 00 28 00 | ........4. ...(.
0x00010030 | 0B 00 0A 00 01 00 00 00 00 00 00 00 00 00 01 00 | ................
0x00010040 | 00 00 01 00 7E 00 00 00 7E 00 00 00 05 00 00 00 | ....~...~.......
0x00010050 | 00 10 00 00 B7 12 00 00 93 82 02 82 73 90 02 30 | ............s..0
0x00010060 | B7 05 00 80 93 85 05 01 73 90 05 18 33 86 A5 00 | ........s...3...
0x00010070 | 93 02 F0 FF 73 90 22 30 73 00 20 30 01 A0 00 00 | ....s."0s. 0....
0x00010080 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
0x00010090 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
0x000100A0 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
0x000100B0 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
0x000100C0 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
0x000100D0 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
0x000100E0 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
0x000100F0 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................
```

Since we want to decode the virtual address 0x0, we read the PTE from 0x1_0000 which gives a strange value of 0x464c_457f (`<DEL>ELF<^A>` in ASCII).

At this point translation fails and Renode generates the `INST_PAGE_FAULT` exception.

Handling this exception requires jumping to the trap handler which address is read from the `stvec` register (not set, 0x0 by default). This causes a loop - we want again access address 0x0 in the S mode.

xobs, 2020-02-27:
You're correct in that the `add` is not necessary.  It's taken from some existing code.

I guess the question is: What should happen when STVEC/MTVEC are invalid? I haven't found anything in the spec that says what to do.  I suspect the answer is to hang, which is what Renode is doing.

The problem is that the gdb bridge is broken, and so we can no longer communicate with the target chip.  This is accurate, but maybe not very useful from a user standpoint.

mateusz-holenko, 2020-02-28:
Sure, you are totally right. I wanted to analyze what is going on, but agree that looping inside Renode is not the most user friendly behaviour ;)

We'll discuss this case and decide on the best way to inform user that the system configuration is "invalid/broken".