renode/renode #75: Unable to execute any mbed-os firmwares for STM32F4-Discovery board

maxgerhardt, 2020-02-21:
I have compiled the latest Renode version from this repository. 
```
Renode, version 1.8.2.28975 (82f5f4fe-202002211605)
```

Using the PlatformIO build system, I have built two mbed-os firmwares target for the STM32F4-Discovery target (STM32F407VG). One without the RTOS (Keil RTX5) and one without. The firmware code in both cases is a simple blinky and serial output on UART2. 

However the firmwares are not able to run properly when using Renode. For the non-RTOS part, it keeps spamming 

```
16:22:53.2084 [DEBUG] sysbus: Read value 0x0 from TIM5:CNT (0x40000C24).
16:22:53.2085 [WARNING] sysbus: [cpu: 0x8002662] (tag: 'RCC_CR') ReadDoubleWord from non existing peripheral at 0x40023800, returning 0xA020083.
``` 

while never reading any other timer 5 counter value than 0, it is seemingly stuck. The RTOS version just produces some output in the Renode console and then stops, with again no UART2 output. Also the state of the LED (`sysbus.gpioPortD.UserLED State`) is always `False`. 

I have setup a repository for easy reproduction at https://github.com/maxgerhardt/renode_mbedos with all the used scripts and firmwares. 

Can someone identify what problem there are in the firmware or in renode that keep mbed-os, both in the RTOS and non-RTOS version, from running?

maxgerhardt, 2020-02-28:
We have also tested this firmware on [real hardware](https://de.farnell.com/stmicroelectronics/stm32f407g-disc1/entwicklungsboard-foundation-line/dp/2506840?ost=STM32F407G-DISC1&CMP=GRHS-1000962&ddkey=https%3ASearch) and the LED and UART output functions as expected, so the firmware per se works. Were you able to reproduce our issue on renode?

PiotrZierhoffer, 2020-03-02:
Please take a look at `platforms/cpus/stm32f4.repl` - there is a `RCC_CR` that, in your case, should probably return a different value.

If you cannot hardcode the value, you can use a Python mock for this register - look at `platforms/cpus/sam_e70.repl` for the `flipflop.py` usage, you can use the same approach here.

The flipflop script will return 0x0 and 0xFFFFFF, alternately. This should be enough to help your software move forward.

maxgerhardt, 2020-03-03:
We have read out the value of the `RCC->CR`. `RCC->PLLCFGR` and `RCC_CFGR` register from the real hardware after clock initialization has been done. Using these hardcoded value it results in actual UART2 output 

```
++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x80024EB
Error Value: 0x0
For more info, visit:
https://mbed.com/s/error?error=0x80FF0100&tgt=DISCO_F407VG
-- MbedOS Error Info --
SetSysClock failed
``` 

The used values are 

```
        Tag <0x40023800, 0x40023803> "RCC_CR" 0x03037683
        Tag <0x40023804, 0x40023807> "RCC_PLLCFGR" 0x07405408
        Tag <0x40023808, 0x40023809> "RCC_CFGR" 0x0000940A
```

which corresponds to the standard setup of HSE_XTAL+PLL and LSI for the board (see [clock init](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L476xG/TARGET_NUCLEO_L476RG/system_clock.c#L73) code) 

So it seems that this isn't sufficient to make the clock init process go through :/ 

Do you have more ideas regarding this? I guess we should now write the mentioned python scripts to fix the reset clock control peripheral register in renode ourselves to get the mbed-blinky FW working?

andresR8, 2020-06-09:
I'm having the same issue. 
I've checked the Renode Logs to compare a non-mbed firmware ( stm32f746.resc file ) with the mbed-os-example-blinky-baremetal. 

Writing to this registers (**RCC:CFGR, RCC:PLLCFGR, RCC:CIR** ) is missing on Mbed example
```
[INFO] STM32F746: Machine started.
[DEBUG] nvic: Enabling FPU.
[DEBUG] sysbus: Write value 0x0 to RCC:CFGR (0x40023808).
[DEBUG] sysbus: Write value 0x24003010 to RCC:PLLCFGR (0x40023804).
[DEBUG] sysbus: Write value 0x0 to RCC:CIR (0x4002380C).
[DEBUG] nvic: Read from SHCS register. This is not yet implemented. Returning 0
[DEBUG] nvic: Write to SHCS register. This is not yet implemented. Value written was 0x0.
[WARNING] nvic: Unhandled read from offset 0xD94.
```

andresR8, 2020-07-06:
@maxgerhardt I also get this error
`Error Message: regions >= 4`
I fixed disabling the use of mpu [here](https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_lib.json#L143)

Looking in detail the RCC register you can't put it on a flipflop, because it defines the clock configurations

andresR8, 2020-08-03:
@maxgerhardt check this [PR](https://github.com/renode/renode/pull/119)

mateusz-holenko, 2021-01-14:
Fixes proposed in #119 are merged to master. We also have a script running mbed-os-based firmware: https://github.com/renode/renode/blob/master/scripts/single-node/stm32f746_mbed.resc.

closed 2021-05-05 by mateusz-holenko