renode/renode #78: error running LiteX_Linux_VexRiscv.robot

pdp7, 2020-02-28:
I encounter an error when running `robot ./tests/platforms/LiteX/LiteX_Linux_VexRiscv.robot`:
```
pdp7@x1:~/dev/renode$ robot ./tests/platforms/LiteX/LiteX_Linux_VexRiscv.robot
[ ERROR ] Error in file '/home/pdp7/dev/renode/tests/platforms/LiteX/LiteX_Linux_VexRiscv.robot': Replacing variables from setting 'Resource' failed: Variable '${RENODEKEYWORDS}' not found.
==============================================================================
LiteX Linux VexRiscv                                                          
==============================================================================
Should Boot                                                           | FAIL |
Parent suite setup failed:
No keyword with name 'Setup' found.
------------------------------------------------------------------------------
Should Control LED                                                    | FAIL |
Parent suite setup failed:
No keyword with name 'Setup' found.
------------------------------------------------------------------------------
Should Read Button                                                    | FAIL |
Parent suite setup failed:
No keyword with name 'Setup' found.
------------------------------------------------------------------------------
Should Handle SPI                                                     | FAIL |
Parent suite setup failed:
No keyword with name 'Setup' found.
------------------------------------------------------------------------------
Should Handle I2C                                                     | FAIL |
Parent suite setup failed:
No keyword with name 'Setup' found.
------------------------------------------------------------------------------
LiteX Linux VexRiscv                                                  | FAIL |
Suite setup failed:
No keyword with name 'Setup' found.

Also suite teardown failed:
No keyword with name 'Teardown' found.

5 critical tests, 0 passed, 5 failed
5 tests total, 0 passed, 5 failed
==============================================================================
Output:  /home/pdp7/dev/renode/output.xml
Log:     /home/pdp7/dev/renode/log.html
Report:  /home/pdp7/dev/renode/report.html
```

pdp7, 2020-02-28:
@mgielda this is what I had mentioned at Embedded World.  Maybe I am not invoking it correctly?

PiotrZierhoffer, 2020-02-28:
This is correct. If you're running Renode compiled from sources, and I believe you do, please run `./test.sh path/to/test.robot`

PiotrZierhoffer, 2020-02-28:
This is because we provide additional Renode-specific keywords, and test.sh sets the environment up to handle them correctly.

pdp7, 2020-02-28:
@PiotrZierhoffer thanks, it works now!
```
pdp7@x1:~/dev/renode$ ./test.sh ./tests/platforms/LiteX/LiteX_Linux_VexRiscv.robot
Preparing suites
Starting suites
Running ./tests/platforms/LiteX/LiteX_Linux_VexRiscv.robot
+++++ Starting test 'LiteX Linux VexRiscv.Should Boot'
+++++ Finished test 'LiteX Linux VexRiscv.Should Boot' in 13.38 seconds with status OK
+++++ Starting test 'LiteX Linux VexRiscv.Should Control LED'
+++++ Finished test 'LiteX Linux VexRiscv.Should Control LED' in 12.06 seconds with status OK
+++++ Starting test 'LiteX Linux VexRiscv.Should Read Button'
+++++ Finished test 'LiteX Linux VexRiscv.Should Read Button' in 14.01 seconds with status OK
+++++ Starting test 'LiteX Linux VexRiscv.Should Handle SPI'
+++++ Finished test 'LiteX Linux VexRiscv.Should Handle SPI' in 14.05 seconds with status OK
+++++ Starting test 'LiteX Linux VexRiscv.Should Handle I2C'
+++++ Finished test 'LiteX Linux VexRiscv.Should Handle I2C' in 17.22 seconds with status OK
Cleaning up suites
Aggregating all robot results
Output:  /home/pdp7/dev/renode/output/tests/robot_output.xml
Log:     /home/pdp7/dev/renode/output/tests/log.html
Report:  /home/pdp7/dev/renode/output/tests/report.html
Tests finished successfully :)
``

closed 2020-02-28 by pdp7