renode/renode #79: Mac script does not allow running "-e" with no script

carlosedp, 2020-03-04:
When I run the Mac version of Renode with -e parameters, I get the error:

```bash
> export MONO_FRAMEWORK_PATH=/Library/Frameworks/Mono.framework/Versions/Current
> export DYLD_FALLBACK_LIBRARY_PATH="$DIR:$MONO_FRAMEWORK_PATH/lib:/lib:/usr/lib"
> export PATH="$MONO_FRAMEWORK_PATH/bin:$PATH"

> /Applications/Renode.app/Contents/MacOS/macos_run.command -e "include @scripts/single-node/sifive_fe310.resc" -e "machine StartGdbServer 3333 True" -e "sysbus LoadELF @/Users/cdepaula/Documents/PlatformIO/Projects/200303-171147-zephyr-hello-world/.pio/build/hifive1/firmware.elf" -e "start"

Script path and execute command cannot be set at the same time
usage: Renode.exe [options] [script]

Options:

  -p, --plain                 BOOLEAN
                              Remove steering codes (e.g., colours) from output.
  -P, --port                  INT32
                              Instead of opening a window, listen for monitor commands on the specified port.
  -e, --execute               STRINGs separated by ';'
                              Execute command on startup (this option is exclusive with -s and startup script passed as an argument). May be used many times.
...
```

While if I call the executable directly with mono64, it works:

```bash
❯ mono64 /Applications/Renode.app/Contents/MacOS/bin/Renode.exe -e "include @scripts/single-node/sifive_fe310.resc" -e "machine StartGdbServer 3333 True" -e "sysbus LoadELF @/Users/cdepaula/Documents/PlatformIO/Projects/200303-171147-zephyr-hello-world/.pio/build/hifive1/firmware.elf" -e "start"
12:29:49.0701 [DEBUG] Segment size automatically calculated to value 16MiB (3)ntents/MacOS/./scripts/monitor.py
12:29:49.2937 [DEBUG] sysbus: Loading ELF /var/folders/dv/d24tqkhn7zg55vj4ttrqq35m0000gp/T/renode-23095/3e796033-4dd6-49d9-9d90-3d22c4ecbd1d.tmp.
12:29:49.3075 [INFO] sysbus: Loading segment of 20520 bytes length at 0x20400000.
12:29:49.3245 [DEBUG] sysbus: Segment loaded.lly calculated to value 64KiB
...
```

mateusz-holenko, 2020-03-04:
Thanks for reporting the issue @carlosedp!

I confirm it doesn't work as expected. The problem is caused by this line: https://github.com/renode/renode/blob/master/tools/packaging/macos/macos_run.command#L43.

Instead of 
<pre>
exec -a \"$PROCESS_NAME\" mono64 $MONO_OPTIONS "$EXE_PATH" $*
</pre>

it should be

<pre>
exec -a \"$PROCESS_NAME\" mono64 $MONO_OPTIONS "$EXE_PATH" "$@"
</pre>

As a temporary workaround you can just edit the file locally and apply the above change. 

I'll fix this on the repo and prepare the newer version of the package soon.

carlosedp, 2020-03-04:
Hey @mateusz-holenko , a nice addition would be replacing:
```
DIR=$(cd "$(dirname "$0")"; pwd)
```

with 

```
BIN=$(realpath "$0")
DIR=$(dirname $BIN)
```

This way one could create a symbolic link like `/usr/local/bin/renode` pointing to `/Applications/Renode.app/Contents/MacOS/macos_run.command` and just call `renode`.

mateusz-holenko, 2020-03-05:
@carlosedp: https://github.com/renode/renode/commit/2f1a151d470b0d814e928805f422931a45f3f9f2 and https://github.com/renode/renode/commit/66814df6d2d10f5eda0cad3245a1db1eb167f294 should solve the reported issue.
 
While testing the commits I noticed problems with building packages on the newest MacOS - it's fixed now fixed by https://github.com/renode/renode/commit/b0dcfe4db684b170acb86a33682c0e8c7b248e3e (hopefully it doesn't break compatibility with older versions).

We are currently preparing a new release of Renode, so the packages including all those changes should be available early next week.

carlosedp, 2020-03-11:
Fixed on v1.9.0! Thanks!

closed 2020-03-11 by carlosedp