renode/renode #82: Renode crashes when you attempt to include ""

xobs, 2020-03-12:
Using Renode 1.9.0:

```
Renode, version 1.9.0.10857 (169a3c85-202003100545)
(monitor) i @""
```

Backtrace:

```
Fatal error:
Illegal characters in path.
   at System.IO.Path.CheckInvalidPathChars(String path, Boolean checkAdditional)
   at System.IO.Path.Combine(String path1, String path2)
   at Antmicro.Renode.UserInterface.Monitor.TryGetFilenameFromAvailablePaths(String fileName, String& fullPath)           at Antmicro.Renode.UserInterface.Monitor.ParseTokens(IEnumerable`1 tokensToParse, ICommandInteraction writer)          at Antmicro.Renode.UserInterface.Monitor.Parse(String cmd, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.HandleCommand(String cmd, ICommandInteraction ci)
   at AntShell.Shell.HandleCommand(String cmd, ICommandInteraction ic)
   at AntShell.CommandLine.HandleControlSequence(ControlSequence seq)
   at AntShell.Terminal.NavigableTerminalEmulator.HandleInput(Object input)
   at AntShell.Terminal.NavigableTerminalEmulator.Run(Boolean stopOnError)
   at AntShell.Shell.Start(Boolean stopOnError)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)    at System.Threading.ThreadHelper.ThreadStart(Object obj)

Unhandled Exception: System.ArgumentException: Illegal characters in path.
   at System.IO.Path.CheckInvalidPathChars(String path, Boolean checkAdditional)
   at System.IO.Path.Combine(String path1, String path2)
   at Antmicro.Renode.UserInterface.Monitor.TryGetFilenameFromAvailablePaths(String fileName, String& fullPath)           at Antmicro.Renode.UserInterface.Monitor.ParseTokens(IEnumerable`1 tokensToParse, ICommandInteraction writer)          at Antmicro.Renode.UserInterface.Monitor.Parse(String cmd, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.HandleCommand(String cmd, ICommandInteraction ci)
   at AntShell.Shell.HandleCommand(String cmd, ICommandInteraction ic)
   at AntShell.CommandLine.HandleControlSequence(ControlSequence seq)
   at AntShell.Terminal.NavigableTerminalEmulator.HandleInput(Object input)
   at AntShell.Terminal.NavigableTerminalEmulator.Run(Boolean stopOnError)
   at AntShell.Shell.Start(Boolean stopOnError)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)    at System.Threading.ThreadHelper.ThreadStart(Object obj)
```

I discovered this when trying to figure out the syntax to include files in the "scripts" directory.

PiotrZierhoffer, 2020-03-12:
It seems to be Windows specific. Thanks for the catch

PiotrZierhoffer, 2020-03-12:
It's actually quite funny. Windows tries to use `""` as a file name, and `"` is not a legal character. We'll probably just silence the exception here.

The same happens when you `i @""/<TAB"`.

PiotrZierhoffer, 2020-03-12:
We will put this into the pipeline, but as the .NET API is not very well designed in these areas, it will not be a very trivial fix.