renode/renode #83: No Keyboard Input Events for IKeyboard Devices

xobs, 2020-03-14:
I have a class that derives from `IKeyboard, IDoubleWordPeripheral, IProvidesRegisterCollection<DoubleWordRegisterCollection>, IKnownSize`.  It shows up as a keyboard in the video input device, however `Press()` and `Release()` are not getting called.

I see an event handler attached in `src\Infrastructure\src\UI\VideoAnalyzer\Events\XWTEventSource.cs(29)`, which should call `private void HandleKeyPressed(object sender, KeyEventArgs e)` every time a key is pressed, however that function is never called.  This is because `void WidgetKeyDownHandler (object sender, System.Windows.Input.KeyEventArgs e)` is never called.

I can see `IOHandler::Attach()` is getting called correctly with `keyboard` set to my keyboard device.

Finally, I don't have a Pointer defined.  However, inside `IOHandler.cs`, if I place a breakpoint inside `public void ButtonPressed(PointerButton button)` and click inside the window, the breakpoint is hit.  However, if I place a breakpoint on `public void KeyPressed(KeyScanCode key)` or `public void KeyReleased(KeyScanCode key)`, then no breakpoint is hit.

I do see `void WidgetKeyDownHandler (object sender, System.Windows.Input.KeyEventArgs e)` getting called when I type into the renode console, and UART connections work just fine, it just appears as though IKeyboard devices with a FrameBufferDisplayWidget output aren't connected correctly.

mateusz-holenko, 2020-03-17:
I verified that on Linux on the current master (b88efd5) it seems to work fine (at least with the built-in USB keyboard).

I run the following script:

```
# load a demo with a video analyzer
include @scripts/single-node/stm32f746.resc

# add a USB keyboard
machine LoadPlatformDescriptionFromString "fake_usb: USB.PSE_USB @ sysbus 0xf0000000"
machine LoadPlatformDescriptionFromString "kb: USB.USBKeyboard @ fake_usb 1"

# enable noisy logs in the keyboard model
logLevel -1 fake_usb.kb

# open the second analyzer (the first one was opened before adding the keyboard)
showAnalyzer ltdc
```

As a result of pressing the `A` key on the keyboard on the running demo with the video analyzer focused I can see that the keyboard receives the press/release events:

```
11:23:36.3190 [WARNING] dma2d: Unhandled write to offset 0x8. Unhandled bits: [4] when writing value 0x10. (4)
11:23:39.9304 [NOISY] fake_usb.kb: Pressing A
11:23:39.9335 [NOISY] fake_usb.kb: Handling data packet of size: 8
11:23:39.9355 [NOISY] fake_usb.kb: Enqueuing end of packet marker
11:23:39.9997 [NOISY] fake_usb.kb: Releasing A
11:23:39.9999 [NOISY] fake_usb.kb: Handling data packet of size: 8
11:23:40.0002 [NOISY] fake_usb.kb: Enqueuing end of packet marker
```

Could you check if the above script works for you?

xobs, 2020-03-17:
I had to add `start` to get it running.  I don't get those NOISY messages, but I do get WARNING messages.  E.g. if I press `a` 20 times and then pause the machine:

```
18:34:59.7482 [DEBUG] sysbus: Read value 0x100000 from RCC:AHB1ENR (0x40023830).
18:34:59.7492 [DEBUG] sysbus: Write value 0x900000 to RCC:AHB1ENR (0x40023830).
18:34:59.7492 [DEBUG] sysbus: Read value 0x100000 from RCC:AHB1ENR (0x40023830).
18:35:00.0813 [WARNING] dma2d: Unhandled write to offset 0x8. Unhandled bits: [4] when writing value 0x10. (4)
18:35:05.8619 [WARNING] Unhandled keycode: a (20)
18:35:34.6239 [INFO] STM32F746: Machine paused.
18:35:34.6253 [DEBUG] STM32F746: Disposing sysbus.cpu.
18:35:34.6342 [DEBUG] STM32F746: Disposing sysbus.
18:35:34.6434 [DEBUG] sysbus:
...
```

mateusz-holenko, 2020-03-17:
You are right, `start` is necessary - sorry I didn't mention that.

This warning about `Unhandled keycode` is... the key.

Now that I look into the code of: https://github.com/renode/renode-infrastructure/blob/master/src/UI/VideoAnalyzer/Events/XWTEventSource.cs#L49-L63, it seems we lack the logic of handling keyboard events on Windows at all (`#ifdef !PLATFORM_WINDOWS`).

I'll have to verify if/why the `ifdef` is necessary there.

xobs, 2020-03-17:
I must be doing something wrong, because I can't get any events to happen with my keybord.

I instantiate the keyboard much like you've done https://github.com/betrusted-io/xous-core/blob/d10bc2e8947eed429443ba0658a3a23464d7bb22/emulation/betrusted.repl#L51-L53

```
memlcd: Video.BetrustedLCD @ sysbus 0xF0005000

kbd: Input.BetrustedKbd @ sysbus 0xF0006000
```

However, unlike your example, I cannot show the analyzer:

```
(machine-0) showAnalyzer sysbus.kbd
There was an error executing command 'showAnalyzer sysbus.kbd'
Received 'No backend found for sysbus.kbd' error while initializing analyzer for: sysbus.kbd
. Are you missing a required plugin?
(machine-0)
```

I _can_ show the analyzer for `memlcd`, which I do with `showAnalyzer memlcd`, and I _can_ set the log level of `sysbus.kbd` with logLevel -1 sysbus.kbd`, but I'm unable to see any "Unhandled keycode" when I do that.

mateusz-holenko, 2020-03-17:
We don't have any analyzers for keyboards, so `showAnalyzer sysbus.kbd` will fail - this is expected.

`ltdc` in my example is a video device - in Renode video analyzers allow to pass keyboard/mouse events from your host to the simulation.
In your case you should show the analyzer for `memlcd`.

I tried my example again but *without* the touchscreen (i.e., no pointer device) and I can confirm that now no keyboard events are detected. This must be something with the way we grab input from the host. I'll look into this.

mateusz-holenko, 2020-03-17:
It turned out to be related to the inner widget focus state.

In a setup with a touchscreen (`AbsolutePointerHandler`) each move over the canvas explicitely focus it, so all key press/release events are correctly handled. For reference see: https://github.com/renode/renode-infrastructure/blob/master/src/UI/VideoAnalyzer/Handlers/AbsolutePointerHandler.cs#L62-L63.

Without a touchscreen, the canvas is not focused - combo box/button above is instead and all key events are passed to it, so the `IOHandler` is not aware of those events.

The workaround is to add:
```
if(!widget.HasFocus)
{
    widget.CanGetFocus = true;
    widget.SetFocus();
}
```
at the beginning of `IOHandler.MouseMoved` method: https://github.com/renode/renode-infrastructure/blob/master/src/UI/VideoAnalyzer/Handlers/IOHandler.cs#L28.

xobs, 2020-03-18:
Thanks for tracking this down.  It's good to have confirmation on this problem.

Will that patch be added to the IOHandler?  Or does more testing need to be done?

xobs, 2020-03-18:
I can confirm that with that patch, I now get `Unhandled keycode` events.

mateusz-holenko, 2020-03-18:
That's great! 

Sure, this will be fixed, but I'm still analyzing if this solution is the optimal one.

xobs, 2020-03-18:
I'm looking into the keyboard scanning issue, as to why it's not implemented, and I can get a scancode at least:

```cs
        [DllImport("user32.dll")]
        static extern int MapVirtualKey(uint uCode, uint uMapType);

        [DllImport("user32.dll")]
        static extern short VkKeyScan(char ch);
```

...later...

```cs
            var vk = VkKeyScan((char)e.Key);
            var translated_key = MapVirtualKey((uint)vk, 0x00 /*MAPVK_VK_TO_VSC*/);
            Logger.LogAs(this, LogLevel.Warning, "Unhandled keycode: {0}  {1}  {2:x2}", e.Key, vk, translated_key);
```

This gets me the actual Windows scancodes.  Some examples of keys on my US keyboard in qwerty mode:

```
16:29:29.2918 [WARNING] Unhandled keycode: q  81  10
16:29:29.4490 [WARNING] Unhandled keycode: w  87  11
16:29:29.6070 [WARNING] Unhandled keycode: e  69  12
16:29:29.7830 [WARNING] Unhandled keycode: r  82  13
16:29:30.1270 [WARNING] Unhandled keycode: t  84  14
16:29:30.4590 [WARNING] Unhandled keycode: y  89  15
16:29:31.1930 [WARNING] Unhandled keycode: u  85  16
16:29:31.4640 [WARNING] Unhandled keycode: i  73  17
16:29:31.6420 [WARNING] Unhandled keycode: o  79  18
16:29:31.7430 [WARNING] Unhandled keycode: p  80  19
16:29:32.1980 [WARNING] Unhandled keycode: OpenSquareBracket  219  1a
16:29:32.6727 [WARNING] Unhandled keycode: CloseSquareBracket  221  1b
16:29:33.1157 [WARNING] Unhandled keycode: Backslash  220  2b
16:29:35.0138 [WARNING] Unhandled keycode: a  65  1e
16:29:35.1618 [WARNING] Unhandled keycode: s  83  1f
16:29:35.2888 [WARNING] Unhandled keycode: d  68  20
16:29:35.4188 [WARNING] Unhandled keycode: f  70  21
16:29:35.8038 [WARNING] Unhandled keycode: g  71  22
16:29:36.0858 [WARNING] Unhandled keycode: h  72  23
16:29:36.5218 [WARNING] Unhandled keycode: j  74  24
16:29:37.0208 [WARNING] Unhandled keycode: k  75  25
16:29:37.1918 [WARNING] Unhandled keycode: l  76  26
16:29:37.3358 [WARNING] Unhandled keycode: Semicolon  186  27
16:29:38.0425 [WARNING] Unhandled keycode: SingleQuote  222  28
16:29:38.8495 [WARNING] Unhandled keycode: Return  189  0c
16:29:40.7174 [WARNING] Unhandled keycode: z  90  2c
16:29:40.8464 [WARNING] Unhandled keycode: x  88  2d
16:29:40.9502 [WARNING] Unhandled keycode: c  67  2e
16:29:41.1182 [WARNING] Unhandled keycode: v  86  2f
16:29:41.7142 [WARNING] Unhandled keycode: b  66  30
16:29:41.9862 [WARNING] Unhandled keycode: n  78  31
16:29:42.2054 [WARNING] Unhandled keycode: m  77  32
16:29:42.3174 [WARNING] Unhandled keycode: Comma  188  33
16:29:42.4934 [WARNING] Unhandled keycode: Period  190  34
16:29:42.7644 [WARNING] Unhandled keycode: Slash  191  35
```

This is particularly useful because the scancode is the same regardless of layout.  For example, the same keys pressed in dvorak mode:

```
16:29:38.8495 [WARNING] Unhandled keycode: Return  189  0c
16:29:40.7174 [WARNING] Unhandled keycode: z  90  2c
16:29:40.8464 [WARNING] Unhandled keycode: x  88  2d
16:29:40.9502 [WARNING] Unhandled keycode: c  67  2e
16:29:41.1182 [WARNING] Unhandled keycode: v  86  2f
16:29:41.7142 [WARNING] Unhandled keycode: b  66  30
16:29:41.9862 [WARNING] Unhandled keycode: n  78  31
16:29:42.2054 [WARNING] Unhandled keycode: m  77  32
16:29:42.3174 [WARNING] Unhandled keycode: Comma  188  33
16:29:42.4934 [WARNING] Unhandled keycode: Period  190  34
16:29:42.7644 [WARNING] Unhandled keycode: Slash  191  35
16:30:52.6817 [WARNING] Unhandled keycode: SingleQuote  222  10
16:30:53.1279 [WARNING] Unhandled keycode: Comma  188  11
16:30:53.3409 [WARNING] Unhandled keycode: Period  190  12
16:30:55.0434 [WARNING] Unhandled keycode: p  80  13
16:30:55.3934 [WARNING] Unhandled keycode: y  89  14
16:30:55.6604 [WARNING] Unhandled keycode: f  70  15
16:30:55.8524 [WARNING] Unhandled keycode: g  71  16
16:30:56.0304 [WARNING] Unhandled keycode: c  67  17
16:30:56.1644 [WARNING] Unhandled keycode: r  82  18
16:30:56.4774 [WARNING] Unhandled keycode: l  76  19
16:30:57.0604 [WARNING] Unhandled keycode: Slash  191  1a
16:30:57.5974 [WARNING] Unhandled keycode: Equal  447  00
16:30:57.8564 [WARNING] Unhandled keycode: Backslash  220  2b
16:30:59.4984 [WARNING] Unhandled keycode: a  65  1e
16:30:59.6204 [WARNING] Unhandled keycode: o  79  1f
16:30:59.7504 [WARNING] Unhandled keycode: e  69  20
16:30:59.8654 [WARNING] Unhandled keycode: u  85  21
16:31:00.3104 [WARNING] Unhandled keycode: i  73  22
16:31:00.5354 [WARNING] Unhandled keycode: d  68  23
16:31:00.7393 [WARNING] Unhandled keycode: h  72  24
16:31:01.0004 [WARNING] Unhandled keycode: t  84  25
16:31:01.1784 [WARNING] Unhandled keycode: n  78  26
16:31:01.3884 [WARNING] Unhandled keycode: s  83  27
16:31:01.8764 [WARNING] Unhandled keycode: Minus  189  28
16:31:02.4624 [WARNING] Unhandled keycode: Return  189  28
```

(It's interesting how the Return key ended up with the same keycode as Minus, instead of 0xc.  I wonder if there's a bug, or if it's a special key?  Maybe it's getting cut off because the argument to VkKeyScan is just a char...)

mateusz-holenko, 2020-08-25:
@xobs: I pushed the proposed workaround to the Renode master (https://github.com/renode/renode/commit/7e4d66e4ba58471c54c3fbf601398d53becd76c3).

mateusz-holenko, 2021-06-29:
The implemented workaround seem to work fine and we are not planning to rework it in the nearest future, so I'm closing the issue for now.

closed 2021-06-29 by mateusz-holenko