renode/renode #84: Question: ARM armv7-m IRQ NVIC base address

jensschroer, 2020-03-17:
According to the ARM [ARMv7-M architecture manual](https://static.docs.arm.com/ddi0403/eb/DDI0403E_B_armv7m_arm.pdf), page 651 table B3-3 SCS, the base address for the NVIC is 0xE000E100. 
However, looking at the cpu files in renode (for example stm32f4.repl) set the base address for the NVIC to 0xE000E000 .
Why is this?

pgielda, 2020-03-27:
Its because our NVIC model actually contains also things that start at 0xE000E000 like SysTick.

closed 2020-03-27 by pgielda

Ristovski, 2021-01-26:
@pgielda It would be great if this documented somewhere.

Seeing how the NVIC *only* works when loaded at `0xE000E000`, why is it even allowed to be loaded at a different address (like 0xE000E100)?

pgielda, 2021-01-27:
I am not sure what is the question here? You can connect any peripheral model in Renode to any address on a bus. Including configurations not existing in real world. Obviously to run anything you have to have software that reflects this.

Having said that the NVIC is connected to the right place in scripts, its just that the model has more things in it so handles writes from 0xE000E000. We will revisit this and see if maybe this is confusing and should be split.