renode/renode #85: SVC and FreeRTOS

vbachler, 2020-03-27:
Hi,

I'm trying to run basic freeRTOS example on stm32f4 discovery kit, but had problems with running a scheduler, as far as I could tell SV Call was not handled correctly.

Is there a way of adding support for it?

mgielda, 2020-04-17:
Hi @vbachler the best idea to get things moving is to provide the source code with build instructions and - preferably - a precompiled binary of what you are using. Also some logs would be handy. That would allow us to reproduce your problem to see what potentially needs to be implemented.

mgielda, 2020-04-29:
Hi, actually we now have a Troubleshooting section in the docs which explains in more detail: https://renode.readthedocs.io/en/latest/introduction/troubleshooting.html

sedexo, 2020-05-27:
Hi, I have observed the same behaviour with the PendSV exception on Cortex-M based cores.

Basically all the OSes are using the PendSV exception to switch context on task switch. This is a good resource how to trigger an SV_Exception:
https://interrupt.memfault.com/blog/arm-cortex-m-exceptions-and-nvic#pendsv-example.

However on entry of the exception I can observe that the Link-Register is loaded with the invalid value of 0xFFFFFFEC. This register should contain the location from where the exception was triggered + next instruction.

[image]

This seems to be intended emulation at the moment:
https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/LLVMDisassembler/CortexMAddressTranslator.cs

PiotrZierhoffer, 2020-05-27:
@sedexo do you have a ready example that you can share and that shows this behavior? We ran tons of different software on Cortex-M cores and it would make it much easier for us to narrow the problem down if we had a failing test case ready

sedexo, 2020-05-28:
Hi Piotr,

I've prepared an example for the STM32F4Discovery from the STM32Cube IDE. I've basically ported
the ucos III os. 
`OS_CPU_PendSVHandler:
    CPSID   I                                                   @ Cortex-M7 errata notice. See Note #5
    MOVW    R2, #:lower16:OS_KA_BASEPRI_Boundary                @ Set BASEPRI priority level required for exception preemption
    MOVT    R2, #:upper16:OS_KA_BASEPRI_Boundary
    LDR     R1, [R2]
    MSR     BASEPRI, R1
    DSB
    ISB
    @CPSIE   I

    MRS     R0, PSP                                             @ PSP is process stack pointer
    STMFD   R0!, {R4-R11, R14}                                  @ Save remaining regs r4-11, R14 on process stack

    MOVW    R5, #:lower16:OSTCBCurPtr                           @ OSTCBCurPtr->StkPtr = SP;
    MOVT    R5, #:upper16:OSTCBCurPtr
    LDR     R1, [R5]
    STR     R0, [R1]                                            @ R0 is SP of process being switched out

                                                                @ At this point, entire context of process has been saved
    MOV     R4, LR                                              @ Save LR exc_return value
    BL      OSTaskSwHook                                        @ Call OSTaskSwHook() for FPU Push & Pop

    MOVW    R0, #:lower16:OSPrioCur                             @ OSPrioCur   = OSPrioHighRdy;
    MOVT    R0, #:upper16:OSPrioCur
    MOVW    R1, #:lower16:OSPrioHighRdy
    MOVT    R1, #:upper16:OSPrioHighRdy
    LDRB    R2, [R1]
    STRB    R2, [R0]

    MOVW    R1, #:lower16:OSTCBHighRdyPtr                       @ OSTCBCurPtr = OSTCBHighRdyPtr;
    MOVT    R1, #:upper16:OSTCBHighRdyPtr
    LDR     R2, [R1]
    STR     R2, [R5]

    ORR     LR,  R4, #0x04                                      @ Ensure exception return uses process stack
    LDR     R0, [R2]                                            @ R0 is new process SP; SP = OSTCBHighRdyPtr->StkPtr;
    LDMFD   R0!, {R4-R11, R14}                                  @ Restore r4-11, R14 from new process stack
    MSR     PSP, R0                                             @ Load PSP with new process SP

    MOV     R2, #0                                              @ Restore BASEPRI priority level to 0
    MSR     BASEPRI, R2
    CPSIE   I
    BX      LR                                                  @ Exception return will restore remaining context

.end`

[os_test.zip](https://github.com/renode/renode/files/4693556/os_test.zip)

While debugging on the real device I can see that the Link-Register is set to 0xffffff**ed** and then the cpu jumps back into the next context. In Renode it gets stuck with 0xffffff**ec** instead.
BR,
Michael

Viatorus, 2020-09-08:
Any updates on this?

neophob, 2020-12-03:
I have another example which needs the SV Call, take a look at https://github.com/neophob/renode-nrf52

PaulVittorino, 2022-01-31:
I am also currently experiencing this same issue with a proprietary OS while targeting the FRDM-K64F board.

It appears that @WojMel has fixed the issue in https://github.com/antmicro/tlib/commit/ff1f5828a147b6d3bf474fb5e8a15153cee5116a. The problem is bit[4] of the EXC_RETURN value is clear in @sedexo's example. This bit indicates an extended frame on the stack that includes FPU registers (See B1.5.8 of the Arm v7-M Architecture Reference Manual issue E.e). Before the change in tlib, only bits[3:0] of EXC_RETURN were considered mutable (which is correct for Cortex-M3 and Cortex-M4 without an FPU).

I plan to try disabling FPU support in my build as a workaround.

jwolski-main, 2026-10-05:
The issue should be fixed by now, especially with recent changes to add TrustZone support. Please open a followup if needed.

closed 2026-10-05 by jwolski-main